jsk-ros-pkg/jsk_common #666: [CRITICLAL] jsk_common is faling only on Saucy

k-okada, 2015-01-07:
Hi Everyone, 

It looks like Indigo is close enough for a release for Indigo. 

There are two outstanding packages. jsk_common is faling only on Saucy and concert_service_gazebo is being invalidated. My first guess for that is a circular run dependency. 

We will hold any releases besides fixes for those patches and plan to release Indigo tomorrow. 

For Hydro, it looks like rosjava_msgs is causing most of the regressions. @Daniel Stonier I think it needs to be rereleased with a fixed dependency list. 

Tully

```

mv -f .deps/testopt.Tpo .deps/testopt.Po
/bin/bash ../libtool --tag=CXX   --mode=link /usr/lib/ccache/c++  -g -O2 -fstack-protector --param=ssp-buffer-size=4 -Wformat -Werror=format-security -D_FORTIFY_SOURCE=2   -o testopt testfuncs.o testopt.o ../libnlopt_cxx.la -lm 
libtool: link: /usr/lib/ccache/c++ -g -O2 -fstack-protector --param=ssp-buffer-size=4 -Wformat -Werror=format-security -D_FORTIFY_SOURCE=2 -o .libs/testopt testfuncs.o testopt.o  ../.libs/libnlopt_cxx.so -lm -Wl,-rpath -Wl,/tmp/buildd/ros-indigo-nlopt-1.0.57-0saucy-20150105-0658/lib
../.libs/libnlopt_cxx.so: undefined reference to `std::ostream& std::ostream::_M_insert<unsigned long>(unsigned long)'
../.libs/libnlopt_cxx.so: undefined reference to `std::ctype<char>::_M_widen_init() const'
../.libs/libnlopt_cxx.so: undefined reference to `operator delete(void*)'
../.libs/libnlopt_cxx.so: undefined reference to `operator new[](unsigned long)'
../.libs/libnlopt_cxx.so: undefined reference to `__cxa_end_catch'
../.libs/libnlopt_cxx.so: undefined reference to `std::cout'
../.libs/libnlopt_cxx.so: undefined reference to `std::basic_ostream<char, std::char_traits<char> >& std::operator<< <std::char_traits<char> >(std::basic_ostream<char, std::char_traits<char> >&, char const*)'
../.libs/libnlopt_cxx.so: undefined reference to `std::ostream& std::ostream::_M_insert<double>(double)'
../.libs/libnlopt_cxx.so: undefined reference to `std::ostream::flush()'
../.libs/libnlopt_cxx.so: undefined reference to `std::ios_base::Init::~Init()'
../.libs/libnlopt_cxx.so: undefined reference to `std::ostream::put(char)'
../.libs/libnlopt_cxx.so: undefined reference to `std::__throw_bad_cast()'
../.libs/libnlopt_cxx.so: undefined reference to `std::__detail::_List_node_base::_M_unhook()'
../.libs/libnlopt_cxx.so: undefined reference to `operator delete[](void*)'
../.libs/libnlopt_cxx.so: undefined reference to `__gxx_personality_v0'
../.libs/libnlopt_cxx.so: undefined reference to `std::basic_ios<char, std::char_traits<char> >::clear(std::_Ios_Iostate)'
../.libs/libnlopt_cxx.so: undefined reference to `__cxa_begin_catch'
../.libs/libnlopt_cxx.so: undefined reference to `__cxa_rethrow'
../.libs/libnlopt_cxx.so: undefined reference to `std::ios_base::Init::Init()'
../.libs/libnlopt_cxx.so: undefined reference to `vtable for __cxxabiv1::__si_class_type_info'
../.libs/libnlopt_cxx.so: undefined reference to `operator new(unsigned long)'
../.libs/libnlopt_cxx.so: undefined reference to `std::__detail::_List_node_base::_M_transfer(std::__detail::_List_node_base*, std::__detail::_List_node_base*)'
../.libs/libnlopt_cxx.so: undefined reference to `std::__detail::_List_node_base::_M_hook(std::__detail::_List_node_base*)'
../.libs/libnlopt_cxx.so: undefined reference to `vtable for __cxxabiv1::__class_type_info'
../.libs/libnlopt_cxx.so: undefined reference to `std::ostream::operator<<(int)'
../.libs/libnlopt_cxx.so: undefined reference to `std::basic_ostream<char, std::char_traits<char> >& std::__ostream_insert<char, std::char_traits<char> >(std::basic_ostream<char, std::char_traits<char> >&, char const*, long)'
../.libs/libnlopt_cxx.so: undefined reference to `vtable for __cxxabiv1::__vmi_class_type_info'
collect2: error: ld returned 1 exit status
```

garaemon, 2015-01-07:
it looks like using gcc instead of g++ according to the error message

garaemon, 2015-01-07:
compiler?

trusty
```
/usr/lib/ccache/i686-linux-gnu-g++ -DHAVE_CONFIG_H -I. -I..  -I../api -I../util -D_FORTIFY_SOURCE=2  -g -O2 -fstack-protector --param=ssp-buffer-size=4 -Wformat -Werror=format-security -D_FORTIFY_SOURCE=2 -MT testopt.o -MD -MP -MF .deps/testopt.Tpo -c -o testopt.o testopt.cpp
```

saucy
```
libtool: link: /usr/lib/ccache/c++ -g -O2 -fstack-protector --param=ssp-buffer-size=4 -Wformat -Werror=format-security -D_FORTIFY_SOURCE=2 -o .libs/testopt testfuncs.o testopt.o  ../.libs/libnlopt_cxx.so -lm -Wl,-rpath -Wl,/tmp/buildd/ros-indigo-nlopt-1.0.57-0saucy-20150105-0658/lib

```

garaemon, 2015-01-07:
looks same

saucy
```
libtool: link: /usr/lib/ccache/cc -std=gnu99 -shared  -fPIC -DPIC  -Wl,--whole-archive direct/.libs/libdirect.a cdirect/.libs/libcdirect.a stogo/.libs/libstogo.a praxis/.libs/libpraxis.a luksan/.libs/libluksan.a crs/.libs/libcrs.a mlsl/.libs/libmlsl.a mma/.libs/libmma.a cobyla/.libs/libcobyla.a newuoa/.libs/libnewuoa.a neldermead/.libs/libneldermead.a auglag/.libs/libauglag.a bobyqa/.libs/libbobyqa.a isres/.libs/libisres.a slsqp/.libs/libslsqp.a api/.libs/libapi.a util/.libs/libutil.a -Wl,--no-whole-archive  -lm  -O2   -Wl,-soname -Wl,libnlopt_cxx.so.0 -o .libs/libnlopt_cxx.so.0.7.0
```

trusty
```
libtool: link: /usr/lib/ccache/i686-linux-gnu-gcc -std=gnu99 -shared  -fPIC -DPIC  -Wl,--whole-archive direct/.libs/libdirect.a cdirect/.libs/libcdirect.a stogo/.libs/libstogo.a praxis/.libs/libpraxis.a luksan/.libs/libluksan.a crs/.libs/libcrs.a mlsl/.libs/libmlsl.a mma/.libs/libmma.a cobyla/.libs/libcobyla.a newuoa/.libs/libnewuoa.a neldermead/.libs/libneldermead.a auglag/.libs/libauglag.a bobyqa/.libs/libbobyqa.a isres/.libs/libisres.a slsqp/.libs/libslsqp.a api/.libs/libapi.a util/.libs/libutil.a -Wl,--no-whole-archive  -lm  -O2   -Wl,-soname -Wl,libnlopt_cxx.so.0 -o .libs/libnlopt_cxx.so.0.7.0
```

garaemon, 2015-01-07:
I "successfully" reproduce the error with docker.
```
sudo docker pull ubuntu:13.10
sudo docker run -i -t ubuntu:13.10 /bin/bash
apt-get update
apt-get install build-essential wget libtool
wget http://ab-initio.mit.edu/nlopt/nlopt-2.3.tar.gz
tar xvzf nlopt-2.3.tar.gz
cd nlopt-2.3
./configure --enable-shared --with-cxx
make
```

garaemon, 2015-01-07:
`./configure --enable-shared --with-cxx LIBS='-lstdc++'` solves

garaemon, 2015-01-07:
see this issue https://github.com/sass/libsass/issues/421

garaemon, 2015-01-07:
Any other error on indigo buildfarm?

k-okada, 2015-01-07:
maybe no

closed 2015-01-07 by k-okada