jsk-ros-pkg/jsk_common #742: indigo urdfdom missing

garaemon, 2015-02-06:
[jenkins said](http://jenkins.ros.org/job/devel-indigo-jsk_common/ARCH_PARAM=amd64,UBUNTU_PARAM=trusty,label=devel/134/changes)

```
KeyError: 'urdfdom'
I: Copying back the cached apt archive contents
I: unmounting /var/cache/pbuilder/ccache filesystem

```

```
jskuser@fuji:~$ ROS_DISTRO=indigo rosdep db  | grep urdf
jskuser@fuji:~$ ROS_DISTRO=hydro rosdep db  | grep urdf
urdf_tutorial -> ros-hydro-urdf-tutorial
urdfdom -> ros-hydro-urdfdom
collada_urdf -> ros-hydro-collada-urdf
iai_urdf_msgs -> ros-hydro-iai-urdf-msgs
collada_urdf_jsk_patch -> ros-hydro-collada-urdf-jsk-patch
urdf -> ros-hydro-urdf
urdfdom_py -> ros-hydro-urdfdom-py
lizi_urdf -> ros-hydro-lizi-urdf
urdfdom_headers -> ros-hydro-urdfdom-headers
urdf_parser_plugin -> ros-hydro-urdf-parser-plugin
p2os_urdf -> ros-hydro-p2os-urdf
```

k-okada, 2015-02-06:
```
k-okada@kokada-t440s:~$ rosdep db | grep urdf
urdf_tutorial -> ros-indigo-urdf-tutorial
urdf_parser_plugin -> ros-indigo-urdf-parser-plugin
collada_urdf -> ros-indigo-collada-urdf
iai_urdf_msgs -> ros-indigo-iai-urdf-msgs
p2os_urdf -> ros-indigo-p2os-urdf
liburdfdom-headers-dev -> liburdfdom-headers-dev
urdf -> ros-indigo-urdf
urdfdom_py -> ros-indigo-urdfdom-py
liburdfdom-dev -> liburdfdom-dev
```
may be we should depend on collada_urdf, see https://github.com/jsk-ros-pkg/jsk_common/pull/557
but it is very difficult to pass both devel-inidog-jsk-common and devel-hydro-jsk-common from same branch, you can ignore hits.
this is disabled for release https://github.com/tork-a/jsk_common-release/blob/master/indigo.ignored

garaemon, 2015-02-06:
I see, it's reasonable

closed 2015-02-06 by garaemon