jsk-ros-pkg/jsk_common #789: failed to compile jskeus with jsk.rosbuild

garaemon, 2015-02-18:
```
jsk.rosbuild --from-source --rtm hydro
```

```
-------------------------------------------------------------------
Profile:                     default
Extending:             [env] /opt/ros/hydro
Workspace:                   /home/jskuser/ros/hydro_parent
Source Space:       [exists] /home/jskuser/ros/hydro_parent/src
Build Space:        [exists] /home/jskuser/ros/hydro_parent/build
Devel Space:       [missing] /home/jskuser/ros/hydro_parent/devel
Install Space:     [missing] /home/jskuser/ros/hydro_parent/install
DESTDIR:                     None
-------------------------------------------------------------------
Isolate Develspaces:         False
Install Packages:            False
Isolate Installs:            False
-------------------------------------------------------------------
Additional CMake Args:       None
Additional Make Args:        None
Additional catkin Make Args: None
-------------------------------------------------------------------
Workspace configuration appears valid.

```

```
  /bin/sh: 1:
  /home/jskuser/ros/hydro_parent/devel/share/euslisp/jskeus/eus//Linux64/bin/eusgl:
  not found

  make[5]: ***
  [/home/jskuser/ros/hydro_parent/devel/share/euslisp/jskeus/eus//Linux64/obj/irtgeo.o]
  Error 127

```

k-okada, 2015-02-18:
that has been fixed via latest setup, please check
http://jenkins.jsk.imi.i.u-tokyo.ac.jp:8080/job/jsk.rosbuild/
rtm failed due to https://github.com/fkanehiro/hrpsys-base/pull/488

garaemon, 2015-02-21:
not yet resolved

```
leus@leus-All-Series:~/prog$ ls -l /tmp/jsk.rosbuild 
-rw-rw-r-- 1 leus leus 14215  2月 21 20:42 /tmp/jsk.rosbuild
```

k-okada, 2015-02-22:
this has been fixed
http://jenkins.jsk.imi.i.u-tokyo.ac.jp:8080/job/jsk.rosbuild/, but still
have problem on nlopt compile, can you fix that -> @s-noda

closed 2015-08-27 by garaemon