jsk-ros-pkg/jsk_common #796: failed to build assimp_devel with jsk.rosbuild

otsubo, 2015-02-20:
````
(...snip...)
[assimp_devel] ==> '/home/otsubo/ros/hydro/build/assimp_devel/build_env.sh /usr/bin/make -j8 -l8' in '/home/otsubo/ros/hydro/build/assimp_devel'
Scanning dependencies of target libassimp_devel
[100%] Generating /home/otsubo/ros/hydro/devel/lib/libassimp_devel.so
/home/otsubo/ros/hydro/src/jsk-ros-pkg/jsk_common/3rdparty/assimp_devel/Makefile:21: /home/otsubo/ros/hydro/devel/share/mk/download_unpack_build.mk: No such file or directory
make[3]: warning: jobserver unavailable: using -j1.  Add `+' to parent make rule.
make[3]: *** No rule to make target `/home/otsubo/ros/hydro/devel/share/mk/download_unpack_build.mk'.  Stop.
make[2]: *** [/home/otsubo/ros/hydro/devel/lib/libassimp_devel.so] Error 2
make[1]: *** [CMakeFiles/libassimp_devel.dir/all] Error 2
make: *** [all] Error 2
[assimp_devel] <== '/home/otsubo/ros/hydro/build/assimp_devel/build_env.sh /usr/bin/make -j8 -l8' failed with return code '2
````
after i clean up $HOME/ros/hydro and $HOME/ros/hydro_parent using
    catkin clean --all
,clean up setting related to ros from .bashrc, restart the terminal, and build each workspace using 
    catkin build --force-cmake
,it is fixed.

i don't know how i should change jsk.rosbuild itself. just for your information.

mention: @ompugao, @furushchev

otsubo, 2015-02-20:
oops, no. not fixed.

k-okada, 2015-02-20:
if you 're missing `/home/otsubo/ros/hydro/devel/share/mk/
download_unpack_build.mk: No such file or directory`, sudo apt-get install
ros-hydro-mk

otsubo, 2015-02-20:
but i'm trying to install ros from source.
using 
````
bash /tmp/jsk.rosbuild indigo --from-full-source
````
is it a correct way to fix this problem?

otsubo, 2015-02-20:
s/indigo/hydro/

otsubo, 2015-02-20:
oh, it's already installed

k-okada, 2015-02-20:
do not use from-full-source, use from-source.

furushchev, 2015-03-10:
@otsubo Do you have still any problem?

otsubo, 2015-04-10:
No problem, thanks.

closed 2015-04-10 by otsubo