jsk-ros-pkg/jsk_common #917: Why don't we stop using hardtab in jsk-ros-pkg source codes?

wkentaro, 2015-05-05:
In source codes in jsk-ros-pkg, there are lots of hardtabs as well as softtabs.
The width of hardtab differs according to the settings so I think we should only softtab. Hardtab makes it hard to read the codes.

As example [pr2-interface.l](https://github.com/jsk-ros-pkg/jsk_pr2eus/blob/master/pr2eus/pr2-interface.l#L239-L269)::

**before converting hardtab -> softtab**
[image]

**after converting hardtab -> softtab**
[image]

I'm not emacs user, but [this](http://www.emacswiki.org/emacs/IndentationBasics) says in emacs, you can stop using hardtab by adding below in your .emacs or .emacs.d/init.el:
```
 (setq-default indent-tabs-mode nil)
```

And in vim, you can write on .vimrc like below:
```
set expandtab
```

garaemon, 2015-05-05:
just a historical reason.

You can untabify the codes when you find hard tab codes if you can maintain
them.

We can apply a script to untabify them but I don't recommend it because
1. need to write good script to estimate tab width
2. make it hard to track commit logs

I believe newer codes are written with soft tabs.

BTW, I had added soft-tab setting to jsk default .emacs but it's not used
these days.

wkentaro, 2015-05-06:
I'm willing to appreciate the history of these source codes,
but wondering the newer codes are really written with soft tabs.

I executed below in `jsk_robot`, `jsk_pr2eus`, `jsk_demos` for `.l` files to find lines which has hardtab and edited in each year (2013-2015).
```
git ls-files | grep ".*\.l$" | xargs -n 1 git blame | cat | grep $'\t' | grep "2015-[0-9][0-9]-[0-9][0-9]" | wc -l
```

**Result** (YEAR : NUMBER_OF_LINES)
* jsk_robot
  * 2013 : 52
  * 2014 : 161
  * 2015 : 88

* jsk_pr2eus
  * 2013 : 117
  * 2014 : 32
  * 2015 : 37

* jsk_demos
  * 2013 : 675
  * 2014 : 40
  * 2015 : 461

It looks there are still many hard tabs..

Perhaps emacs settings in some sharing PC in the lab is not finely configured, and that's why there are still many new hard tabs.

>
BTW, I had added soft-tab setting to jsk default .emacs but it's not used
these days.

I didn't know that, I'd like to share it with my friends in lab. Where can I find it?

garaemon, 2015-05-06:
You can find the settings in svn+ssh://
aries.jsk.t.u-tokyo.ac.jp/home/jsk/svn/trunk/scripts

wkentaro, 2015-05-08:
tnx!

closed 2015-05-08 by wkentaro