jsk-ros-pkg/jsk_common #919: Please split out stable packages into separate release units

tfoote, 2015-05-08:
With all these packages bundled together each release causes a lot of build time on the farm. (On the order of 24hours build time per rosdistro released) In each release very few packages appear to have changes applied to them, but because they are released together all of them are required to rebuild not just the changed packages but also the unchanged packages as well as anything that depends upon them.

I audited the length of time to build each package in this repository.  No individual package is a major problem, it's just that they all add up especially when repeated across the different architectures. 

This is one of the most frequently released packages, which we would like to encourage, however from reviewing our usage releasing the unchanged packages is causing the builds of this repository to consume a large fraction of our buildfarm's resources. 

Here's the details from my analysis. It's based on a manual sampling of the latest trusty amd64 builds for indigo. This analysis does not capture downstream dependency build time.

jsk_tilt_laser takes 12 minutes
speech_recognition_msgs takes 8 minutes
nlopt takes 11 minutes
multi_map_server takes 19 minutes
dynamic_tf_publisher takes 5 minutes
bayesian_belief_networks takes 6 minutes
posedetection_msgs takes 11 minutes
jsk_common takes 5 minutes
jsk_footstep_msgs takes 3 minutes
jsk_tools takes 7 minutes
jsk_topic_tools takes 17 minutes
jsk_network_tools takes 11 minutes
jsk_gui_msgs takes 5 minutes
rostwitter takes 13 minutes
assimp_devel takes 16 minutes
rospatlite takes 8 minutes
voice_text takes 7 minutes
ff takes 12 minutes
ffha takes 8 minutes
jsk_hark_msgs takes 4 minutes
image_view2 takes 17 minutes
jsk_data takes 2 minutes
rosping takes 10 minutes
julius takes 9 minutes
libsiftfast takes 11 minutes
virtual_force_publisher takes 11 minutes
opt_camera takes 15 minutes
mini_maxwell takes 7 minutes
downward takes 15 minutes
sklearn takes 11 minutes
Sum of the time per arch is 296 minutes
On 6 arches that's 29.6 hours
On 2 distros that's 59.2 hours

k-okada, 2015-05-09:
sorry to make you trouble, We'll fix them:

the list of recent updates:
```
May 9 1.0.70 jsk_data, jsk_footstep_msgs, jsk_netowrk_tools, jsk_tilt_laser, jsk_topic_tools
May 4 1.0.69 image_view2, jsk_tools, jsk_topic_tools,
may 4 1.0.68 jsk_data, jsk_totpic_tools
May 5 1.0.67 downward, image_view2, jsk_data, jsk_network_tools, jsk_tilt_laser, jsk_tools, jsk_\
topic_tools, posedetection_msgs, rostwitter
Apr 4 1.0.66 jsk_tilt_laser, jsk_tools,
```

One idea is move stable package to different repository, for example create new repo named jsk_3rdparty and jsk_comon_msgs and split them
```
jsk_common takes 5 minutes

jsk_tilt_laser takes 12 minutes
multi_map_server takes 19 minutes
dynamic_tf_publisher takes 5 minutes
jsk_tools takes 7 minutes
jsk_topic_tools takes 17 minutes
jsk_network_tools takes 11 minutes
image_view2 takes 17 minutes
jsk_data takes 2 minutes
virtual_force_publisher takes 11 minutes
                       (+ 12 19 5 7 17 11 17 2 11) 101

speech_recognition_msgs takes 8 minutes
posedetection_msgs takes 11 minutes
jsk_footstep_msgs takes 3 minutes
jsk_gui_msgs takes 5 minutes
jsk_hark_msgs takes 4 minutes
                       (+ 8 11 3 5 4) 31

nlopt takes 11 minutes
bayesian_belief_networks takes 6 minutes
rostwitter takes 13 minutes
assimp_devel takes 16 minutes
rospatlite takes 8 minutes
voice_text takes 7 minutes
ff takes 12 minutes
ffha takes 8 minutes
rosping takes 10 minutes
julius takes 9 minutes
libsiftfast takes 11 minutes
opt_camera takes 15 minutes
mini_maxwell takes 7 minutes
downward takes 15 minutes
sklearn takes 11 minutes
                      (+ 11 6 13 16 8 7 12 8 10 9 11 15 7 15 11) 159

(+ 5 101 31 159)
296
```

other is to move active package to down streams, 
```
jsk_footstep_msgs -> jsk_control
jsk_network_tools, jsk_topic_tools -> jsk_demo ?
jsk_tilt_laser -> jsk_perception
```

I think big problem is whether we should do this today, or one month later > @garaemon 
Anyway we'll release next version after solving this problem.

k-okada, 2015-06-09:
since, DRC has over, I think we can split jsk_common to several packages, as proposed above, if someone have different idea , nor changes not committed yet, please let me know.

k-okada, 2015-06-14:
Done

closed 2015-06-14 by k-okada

k-okada, 2015-06-14:
See #988