jsk-ros-pkg/jsk_common #989: Capture axis video using mjpeg_capture.sh in jsk_tools?

wkentaro, 2015-06-13:
To record baxter's motions I created a script to capture axis video's mjpeg stream.
https://github.com/start-jsk/2014-semi/blob/master/jsk_2014_picking_challenge/scripts/record_axis_video.py

I'd like to add this kind of script to `jsk_tools`, and found `mjpeg_capture.sh`.
Is that possible to capture axis video using the script?
or should I add a new one?

garaemon, 2015-06-14:
like this:
```
ffmpeg -f mjpeg -i http://user:passwd@133.11.216.150/axis-cgi/mjpg/video.cgi\?resolution\=800x600 hoge.mpeg
```

wkentaro, 2015-06-16:
So can I replace mjpeg_capture.sh with ffmpeg?
I don't understand the usage of mjepg_capture.sh which is now in the jsk_tools pkg.

garaemon, 2015-06-16:
Please use existing script

wkentaro, 2015-06-16:
I don't understand what these lines do.
https://github.com/jsk-ros-pkg/jsk_common/blob/master/jsk_tools/src/mjpeg_capture.sh#L32-L40

wkentaro, 2015-06-16:
```
$ ./mjpeg_capture.sh http://user:passwd@133.11.216.141/mjpg/video.mjpg output.mp4
touch: missing file operand
Try 'touch --help' for more information.
```
And timeout in while loop..

wkentaro, 2015-06-16:
I'd like to add a script like https://github.com/jsk-ros-pkg/jsk_common/pull/998 .

garaemon, 2015-06-16:
It just waits for other processes which listening to the port to finish for 30 seconds.
I don't know why its required though

wkentaro, 2015-06-17:
Closing with https://github.com/jsk-ros-pkg/jsk_common/pull/1000

closed 2015-06-17 by wkentaro