jsk-ros-pkg/jsk_common #993: missing libpython3.2.so when building hrpsys

furushchev, 2015-06-15:
continued from https://github.com/start-jsk/hrpsys/issues/141

```bash 
cd ros/hydro_parent/build/hrpsys
make
```
とすると
https://gist.github.com/de083065ba4c152e65e7
のようなエラーになりました。

```bash
sudo apt-get install python3.2-dev
make
```

とするとすべてうまくビルドできたので、依存関係不足だと思います。PRを別途送ります。

k-okada, 2015-06-15:
> continued from start-jsk/hrpsys#141
>
> cd ros/hydro_parent/build/hrpsys
> make
>
> とすると
> https://gist.github.com/de083065ba4c152e65e7
> のようなエラーになりました。
>
> sudo apt-get install python3.2-dev
> make
>
> とするとすべてうまくビルドできたので、依存関係不足だと思います。PRを別途送ります。

up to jade , the software should depends on python2,
http://www.ros.org/reps/rep-0003.html

furushchev, 2015-06-15:
wow, so what should we do?
add to install python3.2-dev by jsk.rosbuild? (not good solution i think)

k-okada, 2015-06-15:
super easy: remove python3 from your environment.
average: fix cmake or something, fix to compile hrpsys with python2 even if
python3 is installed.

furushchev, 2015-06-15:
Might be solved. 
My desktop shows `${PYTHON_LIBRARIES}` as `/usr/lib/python3.2/config/libpython3.2.so`in CMake.
But in my laptop `${PYTHON_LIBRARIES}` as usual libpython2.7.so
I checked the installation of python by `sudo aptitude search python3`, it shows `pi` that means "partly installed".
At last, the problem is only on my hardware.

closed 2015-06-15 by furushchev