jsk-ros-pkg/jsk_common #1036: jsk.rosbuild does not complete on jade

garaemon, 2015-07-17:
Failed to install gazebo2
```
executing command [sudo -H apt-get install -y -qq ros-jade-kdl-conversions]
ERROR: the following rosdeps failed to install
  apt: command [sudo -H apt-get install -y -qq gazebo2] failed
  apt: Failed to detect successful installation of [gazebo2]
++ error
++ echo 'export ROS_HOME=/home/leus/.ros'
export ROS_HOME=/home/leus/.ros
++ echo 'source /home/leus/ros/jade/devel/setup.bash'
source /home/leus/ros/jade/devel/setup.bash
++ cd /home/leus/ros/jade/build
/tmp/jsk.rosbuild: line 22: cd: /home/leus/ros/jade/build: No such file or directory
++ catkin_test_results --all
/tmp/jsk.rosbuild: line 22: catkin_test_results: command not found
++ echo done
done
++ exit 1
leus@leus-GB-BXi5G-760:~$ sudo apt-get install gazebo2
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 gazebo2 : Depends: libsdformat-dev (>= 1.4.11-1osrf1) but it is not going to be installed
           Depends: libsdformat-dev (< 2.0.0) but it is not going to be installed
E: Unable to correct problems, you have held broken packages.

```

garaemon, 2015-07-17:
It can be solved by following command
```
sudo apt-get install libsdformat1
sudo apt-get install gazebo2
```

k-okada, 2015-07-17:
from http://www.ros.org/reps/rep-0003.html, jade uses gazebo5

garaemon, 2015-07-17:
rospdep says gazebo is gazebo2
```
leus@leus-GB-BXi5G-760:~$ ROS_DISTRO=jade rosdep db| grep gazebo
gazebo_ros -> ros-jade-gazebo-ros
gazebo_ros_pkgs -> ros-jade-gazebo-ros-pkgs
libgazebo5-dev -> libgazebo5-dev
gazebo_msgs -> ros-jade-gazebo-msgs
gazebo -> gazebo2
gazebo5 -> gazebo5
gazebo_plugins -> ros-jade-gazebo-plugins
```

garaemon, 2015-07-17:
on indigo, gazebo is gazebo2
```
$ ROS_DISTRO=indigo rosdep resolve gazebo
#apt
gazebo2
```