jsk-ros-pkg/jsk_common #1049: jskrosbuildを実行すると,catkin buildでエラーが出てします.

otsubo, 2015-07-27:
jskrosbuildを実行したところ,catkin buildが実行されずに抜けていしまいます.
source spaceがないと言われてしまします.

```
+ cd /home/leus/ros/hydro_parent
+ LC_ALL=C
+ catkin build --no-status
catkin build: error: Unable to find source space `/home/leus/src`
++ error
++ echo 'export ROS_HOME=/home/leus/.ros'
export ROS_HOME=/home/leus/.ros
++ echo 'source /home/leus/ros/hydro/devel/setup.bash'
source /home/leus/ros/hydro/devel/setup.bash
++ cd /home/leus/ros/hydro/build
/tmp/jsk.rosbuild: line 23: cd: /home/leus/ros/hydro/build: No such file or directory
++ catkin_test_results --all
usage: catkin_test_results [-h] [--all] test_results_dir
catkin_test_results: error: too few arguments
++ echo done
done
++ exit 1
```
どこに問題があるのでしょうか.
すみませんが,よろしくお願いいたします.

otsubo, 2015-07-27:
以下,実行時のログです.
```
leus@Dr.Otsubo:~/prog/scripts$ bash /tmp/jsk.rosbuild --from-source hydro 
Install hydro to precise
+ '[' 0 -gt 0 ']'
+ setup-ros
+ '[' '' == -y ']'
+ read -p 'Are you sure? to run dpkg --configure? [y/N] ' -n 1 -r
Are you sure? to run dpkg --configure? [y/N] y+ echo

+ [[ y =~ ^[Yy]$ ]]
+ sudo dpkg --configure -a
+ sudo add-apt-repository ppa:webupd8team/java -y
gpg: keyring `/tmp/tmpXKUakC/secring.gpg' created
gpg: keyring `/tmp/tmpXKUakC/pubring.gpg' created
gpg: requesting key EEA14886 from hkp server keyserver.ubuntu.com
gpg: /tmp/tmpXKUakC/trustdb.gpg: trustdb created
gpg: key EEA14886: public key "Launchpad VLC" imported
gpg: Total number processed: 1
gpg:               imported: 1  (RSA: 1)
OK
+ sudo apt-get update
Hit http://dl.google.com stable Release.gpg
Hit http://jp.archive.ubuntu.com precise Release.gpg                                                                                     
Hit http://jp.archive.ubuntu.com precise-updates Release.gpg                                                                             
Hit http://jp.archive.ubuntu.com precise-backports Release.gpg                                                                           
Hit http://jp.archive.ubuntu.com precise Release                                                                                         
Hit http://jp.archive.ubuntu.com precise-updates Release                                                                                 
Hit http://dl.google.com stable Release                                                                                                  
Hit http://jp.archive.ubuntu.com precise-backports Release                                                                               
Hit http://jp.archive.ubuntu.com precise/main Sources                                                                                    
Hit http://jp.archive.ubuntu.com precise/restricted Sources                                                                              
Hit http://jp.archive.ubuntu.com precise/universe Sources                                                                                
Hit http://jp.archive.ubuntu.com precise/multiverse Sources                                                                              
Hit http://jp.archive.ubuntu.com precise/main amd64 Packages                                                                             
Hit http://jp.archive.ubuntu.com precise/restricted amd64 Packages                                                                       
Hit http://jp.archive.ubuntu.com precise/universe amd64 Packages                                                                         
Hit http://jp.archive.ubuntu.com precise/multiverse amd64 Packages                                                                       
Hit http://jp.archive.ubuntu.com precise/main i386 Packages                                                                              
Hit http://jp.archive.ubuntu.com precise/restricted i386 Packages                                                                        
Hit http://dl.google.com stable/main amd64 Packages                                                                                      
Hit http://jp.archive.ubuntu.com precise/universe i386 Packages                                                                          
Hit http://jp.archive.ubuntu.com precise/multiverse i386 Packages                                                                        
Hit http://jp.archive.ubuntu.com precise/main TranslationIndex                                                                           
Hit http://jp.archive.ubuntu.com precise/multiverse TranslationIndex                                                                     
Hit http://jp.archive.ubuntu.com precise/restricted TranslationIndex                                                                     
Hit http://dl.google.com stable/main i386 Packages                                                                                       
Hit http://jp.archive.ubuntu.com precise/universe TranslationIndex                                                                       
Hit http://jp.archive.ubuntu.com precise-updates/main Sources                                                                            
Hit http://jp.archive.ubuntu.com precise-updates/restricted Sources                                                                      
Hit http://jp.archive.ubuntu.com precise-updates/universe Sources                                                                        
Hit http://jp.archive.ubuntu.com precise-updates/multiverse Sources                                                                      
Ign http://dl.google.com stable/main TranslationIndex                                                                                    
Hit http://jp.archive.ubuntu.com precise-updates/main amd64 Packages                                                                     
Hit http://jp.archive.ubuntu.com precise-updates/restricted amd64 Packages                                                               
Hit http://jp.archive.ubuntu.com precise-updates/universe amd64 Packages                                                                 
Hit http://jp.archive.ubuntu.com precise-updates/multiverse amd64 Packages                                                               
[... cut ...]
catkin build: error: Unable to find source space `/home/leus/src`
++ error
++ echo 'export ROS_HOME=/home/leus/.ros'
export ROS_HOME=/home/leus/.ros
++ echo 'source /home/leus/ros/hydro/devel/setup.bash'
source /home/leus/ros/hydro/devel/setup.bash
++ cd /home/leus/ros/hydro/build
/tmp/jsk.rosbuild: line 23: cd: /home/leus/ros/hydro/build: No such file or directory
++ catkin_test_results --all
usage: catkin_test_results [-h] [--all] test_results_dir
catkin_test_results: error: too few arguments
++ echo done
done
++ exit 1
```

k-okada, 2015-07-27:
rm -fr ~/ros してから試してみてください．

--
◉ Kei Okada

On Mon, Jul 27, 2015 at 6:55 PM, Satoshi Otsubo <notifications@github.com>
wrote:

> 以下,実行時のログです.
>
> leus@Dr.Otsubo:~/prog/scripts$ bash /tmp/jsk.rosbuild --from-source hydro
> Install hydro to precise
> + '[' 0 -gt 0 ']'
> + setup-ros
> + '[' '' == -y ']'
> + read -p 'Are you sure? to run dpkg [...]

otsubo, 2015-07-27:
上の結果ですが,~/rosをremoveしてから,実行した結果になります.

k-okada, 2015-07-27:
では，
```
cd /home/leus/ros/hydro/
ls -al
env  | sort
cd
rm -fr ~/ros
```
してもう一度実行して下さい．

otsubo, 2015-07-27:
上のコマンドを実行して,rosbuildを実行した結果です.
```
leus@Dr.Otsubo:~/ros/hydro$ ls -al
合計 12
drwxrwxr-x  3 leus leus 4096  7月 27 22:00 .
drwxrwxr-x  4 leus leus 4096  7月 27 22:17 ..
drwxrwxr-x 32 leus leus 4096  7月 27 22:05 src
leus@Dr.Otsubo:~/ros/hydro$ env | sort
COLORTERM=gnome-terminal
COMPIZ_CONFIG_PROFILE=ubuntu
CVSDIR=/home/leus/prog
CVS_RSH=ssh
DBUS_SESSION_BUS_ADDRESS=unix:abstract=/tmp/dbus-pnrBdHXXtp,guid=2bc4d06c26b0e4ce898f676800000023
DEFAULTS_PATH=/usr/share/gconf/ubuntu.default.path
DESKTOP_SESSION=ubuntu
DISPLAY=:0.0
ELDKdir=/usr/local/ELDK4.1
GDMSESSION=ubuntu
GNOME_DESKTOP_SESSION_ID=this-is-deprecated
GNOME_KEYRING_CONTROL=/tmp/keyring-6uE1BK
GPG_AGENT_INFO=/tmp/keyring-6uE1BK/gpg:0:1
GTK_IM_MODULE=ibus
HISTCONTROL=ignoredups
HISTSIZE=999999
HOME=/home/leus
JRTMDIR=/home/leus/prog/jrtm
KOTAROPATH=/home/leus/prog/kotaro
LANG=ja_JP.UTF-8
LD_LIBRARY_PATH=.::/lib:/usr/local/svs/bin:/usr/local/lib:/usr/h8300-hitachi-hms/lib:/home/leus/prog/jrtm/arch/Linux/lib
LESSCLOSE=/usr/bin/lesspipe %s %s
LESSOPEN=| /usr/bin/lesspipe %s
LOCAL_SVS_PATH=/usr/local/svs
LOGNAME=leus
LS_COLORS=rs=0:di=01;34:ln=01;36:mh=00:pi=40;33:so=01;35:do=01;35:bd=40;33;01:cd=40;33;01:or=40;31;01:su=37;41:sg=30;43:ca=30;41:tw=30;42:ow=34;42:st=37;44:ex=01;32:*.tar=01;31:*.tgz=01;31:*.arj=01;31:*.taz=01;31:*.lzh=01;31:*.lzma=01;31:*.tlz=01;31:*.txz=01;31:*.zip=01;31:*.z=01;31:*.Z=01;31:*.dz=01;31:*.gz=01;31:*.lz=01;31:*.xz=01;31:*.bz2=01;31:*.bz=01;31:*.tbz=01;31:*.tbz2=01;31:*.tz=01;31:*.deb=01;31:*.rpm=01;31:*.jar=01;31:*.war=01;31:*.ear=01;31:*.sar=01;31:*.rar=01;31:*.ace=01;31:*.zoo=01;31:*.cpio=01;31:*.7z=01;31:*.rz=01;31:*.jpg=01;35:*.jpeg=01;35:*.gif=01;35:*.bmp=01;35:*.pbm=01;35:*.pgm=01;35:*.ppm=01;35:*.tga=01;35:*.xbm=01;35:*.xpm=01;35:*.tif=01;35:*.tiff=01;35:*.png=01;35:*.svg=01;35:*.svgz=01;35:*.mng=01;35:*.pcx=01;35:*.mov=01;35:*.mpg=01;35:*.mpeg=01;35:*.m2v=01;35:*.mkv=01;35:*.webm=01;35:*.ogm=01;35:*.mp4=01;35:*.m4v=01;35:*.mp4v=01;35:*.vob=01;35:*.qt=01;35:*.nuv=01;35:*.wmv=01;35:*.asf=01;35:*.rm=01;35:*.rmvb=01;35:*.flc=01;35:*.avi=01;35:*.fli=01;35:*.flv=01;35:*.gl=01;35:*.dl=01;35:*.xcf=01;35:*.xwd=01;35:*.yuv=01;35:*.cgm=01;35:*.emf=01;35:*.axv=01;35:*.anx=01;35:*.ogv=01;35:*.ogx=01;35:*.aac=00;36:*.au=00;36:*.flac=00;36:*.mid=00;36:*.midi=00;36:*.mka=00;36:*.mp3=00;36:*.mpc=00;36:*.ogg=00;36:*.ra=00;36:*.wav=00;36:*.axa=00;36:*.oga=00;36:*.spx=00;36:*.xspf=00;36:
MACRASYS=/home/leus/prog/macrasys
MANDATORY_PATH=/usr/share/gconf/ubuntu.mandatory.path
OB_DIR=/usr
OH_SDK_LICENSE_PATH=/usr/local/phantom/temp-license.lic
OLDPWD=/tmp
OPENHRPHOME=/home/leus/prog/OpenHRP
ORB=OMNIORB4
PATH=.:/home/leus/bin:/usr/local/bin:/usr/local/svs/bin:/usr/local/jsk/bin:/usr/java/j2sdk1.4.1/bin:/usr/lib/lightdm/lightdm:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/bin:/bin:/usr/h8300-hitachi-hms/bin:/usr/local/ELDK4.1/usr/bin:/home/leus/prog/scripts:/usr/local/src/gxp
PWD=/home/leus/ros/hydro
QT_IM_MODULE=xim
RBRAPPDIR=/home/leus/prog/rbrapp
ROBOT=sim
RtvDIR=/usr/local/RtvLib
SESSION_MANAGER=local/niyodo:@/tmp/.ICE-unix/2414,unix/niyodo:/tmp/.ICE-unix/2414
SHELL=/bin/bash
SHLVL=1
SSH_AGENT_PID=2452
SSH_AUTH_SOCK=/tmp/keyring-6uE1BK/ssh
SSH_USER=eus
SVN_SSH=ssh -l eus
TERM=xterm-color
UBUNTU_MENUPROXY=libappmenu.so
USER=leus
WINDOWID=58769189
XAUTHORITY=/home/leus/.Xauthority
XDG_CONFIG_DIRS=/etc/xdg/xdg-ubuntu:/etc/xdg
XDG_CURRENT_DESKTOP=Unity
XDG_DATA_DIRS=/usr/share/ubuntu:/usr/share/gnome:/usr/local/share/:/usr/share/
XDG_SEAT_PATH=/org/freedesktop/DisplayManager/Seat0
XDG_SESSION_COOKIE=2abfb9637fe1806eec5ac56400000016-1437987759.562085-601281927
XDG_SESSION_PATH=/org/freedesktop/DisplayManager/Session0
XMODIFIERS=@im=ibus
_=/usr/bin/env
leus@Dr.Otsubo:~/ros/hydro$ cd
leus@Dr.Otsubo:~$ rm -rf ~/ros
leus@Dr.Otsubo:~$ wget -q -O /tmp/jsk.rosbuild https://raw.github.com/jsk-ros-pkg/jsk_common/master/jsk.rosbuild
[... 2336 lines cut ...]
+++ '[' '!' http://localhost:11311 ']'
++ unset CATKIN_ENV_HOOK_WORKSPACE
++ _i=3
++ '[' 3 -lt 3 ']'
++ unset _i
++ unset _CATKIN_ENVIRONMENT_HOOKS_COUNT
+ cd /home/leus/ros/hydro_parent
+ LC_ALL=C
+ catkin build --no-status
catkin build: error: Unable to find source space `/home/leus/src`
++ error
++ echo 'export ROS_HOME=/home/leus/.ros'
export ROS_HOME=/home/leus/.ros
++ echo 'source /home/leus/ros/hydro/devel/setup.bash'
source /home/leus/ros/hydro/devel/setup.bash
++ cd /home/leus/ros/hydro/build
/tmp/jsk.rosbuild: line 23: cd: /home/leus/ros/hydro/build: No such file or directory
++ catkin_test_results --all
usage: catkin_test_results [-h] [--all] test_results_dir
catkin_test_results: error: too few arguments
++ echo done
done
++ exit 1
leus@Dr.Otsubo:~$ 
```

k-okada, 2015-07-27:
もしかして $HOME/.catkin_tools とかありませんか？

--
◉ Kei Okada

On Mon, Jul 27, 2015 at 11:52 PM, Satoshi Otsubo <notifications@github.com>
wrote:

> 上のコマンドを実行して,rosbuildを実行した結果です.
>
> leus@Dr.Otsubo:~/ros/hydro$ ls -al
> 合計 12
> drwxrwxr-x  3 leus leus 4096  7月 27 22:00 .
> drwxrwxr-x  4 leus leus 4096  7月 27 22:17 ..
> drwxrwxr-x 32 leus leus 4096  7月 27 [...]

otsubo, 2015-07-27:
ありました.
これをrmして実行すれば良いのでしょうか.

k-okada, 2015-07-27:
そうです．

k-okada, 2015-07-27:
そういう人もいる，ということで対応をPRにつくりました
https://github.com/jsk-ros-pkg/jsk_common/pull/1051

otsubo, 2015-07-27:
ちゃんと,catkin buildが実行されました.
お手数をお掛けしてしまいまして,申し訳ありませんでした.

closed 2015-07-27 by otsubo