jsk-ros-pkg/jsk_common #1105: gazebo上でのmultisense_laser_pipelineの使用について

orikuma, 2015-08-28:
1. https://github.com/jsk-ros-pkg/trans_system/pull/456 及び  を適用
2. scan_to_cloud_filter_chainのLaserScanIntensityFilterを削除
3. L/R_thk_palm -> L/R_thk_finger1/2 のtfをstatic_tf_publisherから出す
4. /multisense/joint_statesを/joint_statesにremap
5. roslaunch jaxon_multisense_local.launch RUN_DRIVER:=true
で/full_cloud2まで上がるようになりましたが, 2, 3の対策はどうするべきでしょうか.
stereoについてはまだ調査中です.

A. 2について
gazeboの/multisense/lidar_scanはintensityが全て0なためLaserScanIntensityFilterをつけると全ての入力がフィルタされます.
- multisense_laser_pipeline.launchにsimulatorかどうかでtilt_scan_to_cloudを別々に定義する(最も簡単だが共通化できてない)
- intensityを適当に入れるnodeを挟む(intensityの値をどう決めるのか?)
あたりが対策として考えられそうです.

B. 3について
実機では/joint_statesに/thk_hand_controllerが関節角度を出しているようですが,
- gazeboの時はgazebo_jaxon_no_controller.launchでstatic_tfを出してしまう(簡単だが手先のself_filterが正しくなくなる)
- L/R_thk_finger1/2を除いたjaxon_self_filter_param.yamlをgazebo用に別途作る(簡単だが共通化できてない)
- thk_handの関節角度を出すgazeboのpluginを作る(正しい解決策だが手間, 思いがけないところで嵌るかもしれない)
あたりが対策として考えられそうです.

YoheiKakiuchi, 2015-08-28:
> L/R_thk_palm -> L/R_thk_finger1/2 のtfをstatic_tf_publisherから出す

ハンドを使い物にしたければ、staroなどを参考にハンドも動くようにする。

ハンドを使わないなら応急処置としてはそれでもいいかも。
応急処置として、joint_stateを出すのはダメなのか？

> A. 2について

応急ではない方法は、gazeboからintensityが出るようにする。
もしくは、intensityの長さをゼロにする。 その時にフィルタしないようにする。

orikuma, 2015-08-31:
> ハンドを使い物にしたければ、staroなどを参考にハンドも動くようにする。
ハンドの使用は急ぎでないと思っているのでとりあえず応急で済ませようと思います.
>  応急処置として、joint_stateを出すのはダメなのか？
joint_statesを出すのが駄目な理由は特に無いです.
joint_statesを出すほうが実機との整合性が取れるので良さそうです.

> 応急ではない方法は、gazeboからintensityが出るようにする。
intensityを出すにしろ長さを0にするにしろplugin側の変更が必要そうなのでsrcを見て方針を考えてみます.

garaemon, 2015-09-29:
@orikuma  これもうできたかな?

orikuma, 2015-09-29:
今は応急処置的方法でお茶を濁しているので(ハンドはjoint_statesを固定値で出すnodeを上げている, intensityはIntensityFilterをコメントアウトしている), まだ正しい解決はできていません.
もうしばらくお待ちいただけると幸いです.