jsk-ros-pkg/jsk_common #1108: [enhancement] more subcommands for jsk_data (as well as ls, get)

wkentaro, 2015-09-01:
See also: #1107 

- [ ] `$ jsk_data put <filename>` to upload file
- [ ] `$ jsk_data ls --size` to get to know the file size

garaemon, 2015-09-01:
```
$ ls -l | awk 'BEGIN{mega=1048576} $5 >= mega {$5 = $5/mega "MB"} {print $9 " (" $5 ")"}'
accumulated_heightmap-2015-08-24.zip (2.62086MB)
Carousel.zip (11.5935MB)
cut-2015-08-23.zip (1.05763MB)
heightmap-2015-08-23.zip (11.6394MB)
HYDRA.zip (132290)
LSD_room.bag.zip (474.384MB)
```

garaemon, 2015-09-01:
`ls -sh` may be better

```
$ ls -sh
total 72K
4.0K action/        4.0K CMakeLists.txt  4.0K euslisp/  4.0K include/  4.0K msg/         4.0K README.md  4.0K scripts/  4.0K srv/
 12K CHANGELOG.rst  4.0K config/         4.0K images/   4.0K launch/   4.0K package.xml  4.0K sample/    4.0K src/      4.0K test/
```

garaemon, 2015-09-01:
```diff
```

```
$ make small-list
total 9.1G
2.6G 2014_10_15_fridge_demo.bag
935M amt_hrp2_lie_on_stomach_2011-07-19-10-17-25.bag
 26M dinosauor.pcd
467M hrp2016-rotate-elephant-onehand-20101021.bag
432M hrp2-hand-over-object-20100610.bag
310M hrp2-raw-camera-images-for-cr-capture.bag
 22M hrp2-sample-20100619.bag
1.8M hrp2w-room602-mapping-20100415-2.bag
 97M hrp4r-walk-hat-exp-without-passer-by-2012-02-15-20-42-10.bag
220M hrp4r-walk-hat-exp-without-passer-by-2012-02-15-21-08-01.bag
182M hrp4r-walk-hat-exp-with-passer-by-2012-02-15-21-22-45.bag
228M hrp4r-walk-hat-exp-with-passer-by-2012-02-15-21-34-09.bag
 28M kago1.pcd
 38M kago2.pcd
1.1G kenzo-armarker-20111216.bag
1.4G kojiro-view-20111218.bag
 15M mikita-experiment-set.pcd
708M motegi-arpose-20111216.bag
 27M room.pcd
4.0K screenshot-1324970785.cam
 24K screenshot-1324970785.png
 12M throw_blue_ball.pcd
 26M toaster1.pcd
 18M tsuda.pcd
388M velodyne_2012_0604.bag
```

wkentaro, 2015-09-05:
Thanks for ls with size info.
What should i do for put command?
Use rsync in jsk_data() or Makefile?

I'm not sure which should i edit.
If I edit jsk_data(), I feel it makes the Makefile be unnecessary. Because all ls, get and put command can be write on shell file not makefile.

garaemon, 2015-09-05:
In fact, I think we have no reason to use makefile. You can replace makefile with jsk_data command

wkentaro, 2015-09-05:
I was just concerned about compatibility.
Thanks, I will do that.

Also, I dont know whats difference b/w large and small for jsk_data. I
think it is confusing. and makes command complex.

wkentaro, 2015-09-05:
above is answered at #1114 .

wkentaro, 2015-09-08:
Now I think `jsk_data` should be executable to make it easy to create shell script and roslaunch.

closed 2015-09-28 by garaemon