jsk-ros-pkg/jsk_common #1112: gazeboのxtionをinputにしてimage_view2を立ち上げるとエラーが出る.

otsubo, 2015-09-04:
```bash
roslaunch hrpsys_gazebo_tutorials gazebo_hrp2jsknt_no_controllers.launch
rtmlaunch hrpsys_gazebo_tutorials hrp2jsknt_hrpsys_bringup.launch
```
して
```xml
<launch>
  <arg name="INPUT_IMAGE" default="/xtion/depth/image_raw"/>
  <arg name="INPUT_CAMERA_INFO" default="/xtion/rgb/camera_info"/>
  <arg name="INPUT_POINT_CLOUD" default="/xtion/depth/points" />
  <arg name="DEBUG_VIEW" default="false" />
  <node pkg="image_view2" type="image_view2" name="image_view2">
    <remap from="image" to="/xtion/depth/image_raw" />
  </node>
</launch>
```
を立ち上げると
[ERROR] [1441371960.898093994, 312.102200000]: Unable to convert 32FC1 image to bgr8
が出てしまいます.
解決策等はありますでしょうか.

garaemon, 2015-09-04:
pointsじゃなくてleft/image_rect_colorではない?

otsubo, 2015-09-04:
すみません,
depthではなくてrgbに変えたら動きました.
```xml
<launch>
  <arg name="INPUT_IMAGE" default="/xtion/rgb/image_raw"/>
  <arg name="INPUT_CAMERA_INFO" default="/xtion/rgb/camera_info"/>
  <arg name="INPUT_POINT_CLOUD" default="/xtion/depth/points" />
  <arg name="DEBUG_VIEW" default="false" />
  <node pkg="image_view2" type="image_view2" name="image_view2">
    <remap from="image" to="/xtion/rgb/image_raw" />
  </node>
</launch>
```
大変失礼いたしました.

closed 2015-09-04 by otsubo