jsk-ros-pkg/jsk_common #1114: [jsk_data] What's difference b/w large and small?

wkentaro, 2015-09-05:


garaemon, 2015-09-05:
I forget the original reason but... probably size of the file.

IMO, you can make them into one directory and remove large/small.

wkentaro, 2015-09-05:
thanks.
I'd like to hear @k-okada 's opinion also.

k-okada, 2015-09-07:
just a size.

wkentaro, 2015-09-07:
should we keep  both large and small?

k-okada, 2015-09-07:
i'm not sure what's happening in this package, but originally we design
`make small` for those who would like to check out useable datas and `make
lerge` for downloading all datas. so if new system can handle that, it ok.

k-okada, 2015-09-07:
it does not mean that we have to keep same command, just to keep same idea,
we want provide recommended dataset for beginners.

garaemon, 2015-09-07:
Something like...

* `jsk_data get --smaller-than 100M`
* `jsk_data get --small`
* `jsk_data get --newer-than 2015:08:20`
* `jsk_data ls --size-sort`
* `jsk_data ls --date-sort`

closed 2015-09-28 by garaemon