jsk-ros-pkg/jsk_common #1116: Why jsk-ros-pkg/* version is not set in jsk.rosinstall as master?

wkentaro, 2015-09-05:
https://github.com/jsk-ros-pkg/jsk_common/blob/master/jsk.rosinstall

garaemon, 2015-09-05:
Our policy does not allow maintain multiple branches

wkentaro, 2015-09-05:
What do you mean?

I mean it helps the version divergence of repositories in workspace (i mean
sanity).
like:
https://github.com/jsk-ros-pkg/jsk_common/blob/master/jsk_tools/bin/ws_doctor.py
wstool supports only divergence b/w local branches at present, but b/w
local and remote feature is coming.
https://github.com/vcstools/wstool/pull/38

2015年9月5日土曜日、Ryohei Ueda<notifications@github.com

garaemon, 2015-09-05:
Having version info in rosinstall is redundant because we have only one branch in jsk-ros-pkg 
only thinking about `wstool update`.

But the information can help something, you can add them.

wkentaro, 2015-09-05:
specifying master is helpful when user unexpectedly on the other branch
than master. it invokes V status flag as warning in wstool info.

commit 1108a9b6 (2015-09-05): Specify version: master in rosinstall for jsk-ros-pkg/*

Closes #1116
closed 2015-09-05 by garaemon