jsk-ros-pkg/jsk_common #1123: Fail to build jsk_topic_tools (undefined reference to `ros::WallTimer::setPeriod...)

wkentaro, 2015-09-14:
```
Starting ==> jsk_topic_tools                                                                                                                                                             
                                                                                                                                                                                         
[jsk_topic_tools] ==> '/home/wkentaro/ros/indigo/build/jsk_topic_tools/build_env.sh /usr/bin/make --jobserver-fds=3,5 -j' in '/home/wkentaro/ros/indigo/build/jsk_topic_tools'
Linking CXX executable /home/wkentaro/ros/indigo/devel/lib/jsk_topic_tools/block
Linking CXX executable /home/wkentaro/ros/indigo/devel/lib/jsk_topic_tools/deprecated_relay
Linking CXX executable /home/wkentaro/ros/indigo/devel/lib/jsk_topic_tools/hz_measure
[  0%] Built target _jsk_topic_tools_generate_messages_check_deps_PassthroughDuration
[  0%] Built target _jsk_topic_tools_generate_messages_check_deps_TopicInfo
[  0%] Built target _jsk_topic_tools_generate_messages_check_deps_ChangeTopic
[  0%] Built target _jsk_topic_tools_generate_messages_check_deps_Update
/opt/ros/indigo/lib/libbondcpp.so: undefined reference to `ros::WallTimer::setPeriod(ros::WallDuration const&)'
collect2: error: ld returned 1 exit status
make[2]: *** [/home/wkentaro/ros/indigo/devel/lib/jsk_topic_tools/block] Error 1
make[1]: *** [CMakeFiles/block.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
/opt/ros/indigo/lib/libbondcpp.so: undefined reference to `ros::WallTimer::setPeriod(ros::WallDuration const&)'
collect2: error: ld returned 1 exit status
make[2]: *** [/home/wkentaro/ros/indigo/devel/lib/jsk_topic_tools/hz_measure] Error 1
make[1]: *** [CMakeFiles/hz_measure.dir/all] Error 2
/opt/ros/indigo/lib/libbondcpp.so: undefined reference to `ros::WallTimer::setPeriod(ros::WallDuration const&)'
collect2: error: ld returned 1 exit status
make[2]: *** [/home/wkentaro/ros/indigo/devel/lib/jsk_topic_tools/deprecated_relay] Error 1
make[1]: *** [CMakeFiles/deprecated_relay.dir/all] Error 2
[  0%] Built target _jsk_topic_tools_generate_messages_check_deps_List
make: *** [all] Error 2
[jsk_topic_tools] <== '/home/wkentaro/ros/indigo/build/jsk_topic_tools/build_env.sh /usr/bin/make --jobserver-fds=3,5 -j' failed with return code '2'
                                                                                                                                                                                          
Failed   <== jsk_topic_tools      [ 0.6 seconds ]                                                                                                                                        
[build] There were '1' errors:                                                                                                                                                           

Failed to build package 'jsk_topic_tools' because the following command:

# Command to reproduce:
cd /home/wkentaro/ros/indigo/build/jsk_topic_tools && /home/wkentaro/ros/indigo/build/jsk_topic_tools/build_env.sh /usr/bin/make --jobserver-fds=3,5 -j; cd -

# Path to log:
cat /home/wkentaro/ros/indigo/build/build_logs/jsk_topic_tools.log

Exited with return code: 2 
                                                                                                                                                                                         
[build] Runtime: 1.8 seconds 
```

garaemon, 2015-09-14:
does it happen on jenkins?

wkentaro, 2015-09-14:
No, that's on my pc.
* Ubuntu 14.04
* ROS indigo
* jsk_topic_tools from source

garaemon, 2015-09-14:
roscpp should export rostime https://github.com/ros/ros_comm/blob/indigo-devel/clients/roscpp/CMakeLists.txt#L53

wkentaro, 2015-09-14:
Thanks.
Actually, https://github.com/jsk-ros-pkg/jsk_common/pull/1124 does not fix this error.

I found that the problem is caused by roscpp which is built from source. ( I had ros/ros_comm in my workspace )
It seems that jsk_topic_tools cannot build with roscpp when all packages are built from source.

garaemon, 2015-09-14:
hmm, compile with "VERBOSE=1" and check -l option arguments

garaemon, 2015-09-14:
Where did `ros::WallTimer::setPeriod` define? roscpp?

wkentaro, 2015-09-14:
seems at ros/time.h
/opt/ros/indigo/include/ros/time.h
http://docs.ros.org/diamondback/api/rostime/html/classros_1_1WallTime.html

wkentaro, 2015-09-14:
sorry, no
https://github.com/ros/ros_comm/blob/indigo-devel/clients/roscpp/include/ros/timer.h

garaemon, 2015-09-14:
hmm, catkin_LIBRARIES should contain libroscpp.so and the symbol should be resolved.

geetesh, 2017-04-25:
had similar issue elsewhere and stumbled upon this page. I solved it by 
`sudo apt-get install ros-indigo-bond*`