jsk-ros-pkg/jsk_common #1132: [enhancement] [jsk_topic_tools] warning when using rosrun and there is no subscriber for seconds

wkentaro, 2015-09-19:
why not add this feature to jsk_topic_tools/ConnectedBasedNodelet?
This feature is similar as warning when we run image_view without remapping `image` topic.

**notice**
* And this is only for `rosrun`.
* warning should be only once.
* warning will appear after around 2~5 seconds.

garaemon, 2015-09-19:
This kind of information is available on `/diagnostics`.
ConnectoinBasedNodelet publishes information about publishers and their subscribers to diagnostics.

wkentaro, 2015-09-19:
Oh, sorry. I mean this is not for professionals but for beginners. (I think
the warning of image_view is also for beginners.)

garaemon, 2015-09-21:
I agree that those kinds of warning helps us a lot, OTH I can easily imagine users say
"I have these warning message, this is a problem" and I will answer "It is an expected warning because no program require output yet ".

wkentaro, 2015-09-21:
This idea came from my experience, and I think this is useful for people
who are willing to resolve problems by theirselves. This can save their
time.

I think I understood your concern, that's true the warning does not work
for people who does not read the warning message well.

garaemon, 2015-09-29:
OK, let's see how it is useful. Please implement that feature.

wkentaro, 2015-09-29:
I'm planning to use [roscpp::Timer](http://wiki.ros.org/roscpp/Overview/Timers) with `oneshot` argument.
Do you agree with this way?

garaemon, 2015-09-29:
WallTimer sounds better

commit 5245992b (2015-09-29): [jsk_topic_tools] Warn when no connection in a few sec

Closes #1132

The warning message should be write with ROS_INFO,
for no many warning when running with roslaunch.
closed 2015-09-29 by garaemon