jsk-ros-pkg/jsk_common #1143: jsk_data put rename file destructively

garaemon, 2015-09-29:
I think it should not rename local file

wkentaro, 2015-09-29:
How can we upload file with rsync with renaming only the destination? 
Or should we use another tool to upload?

garaemon, 2015-09-29:
hmm, copy to `/tmp`?

wkentaro, 2015-09-29:
Or after uploading rename it remotely.

wkentaro, 2015-09-29:
So implementation without renaming seems possible, but I think current
bahaivor is simple and good enough.

People should add time stamp in file name, especially for log files
and movies. That is good practice.

Also I considered the renaming can be confusing so added user confirmation.

2015年9月29日火曜日、Kentaro Wada<www.kentaro.wada@gmail.com>さんは書きました:

> Or after uploading rename it remotely.
>
> 2015年9月29日火曜日、Ryohei Ueda<notifications@github.com
>
>> hmm, copy to /tmp?
>>
>>
>
>
> 和田 健太郎 / Kentaro Wada
> http://wkentaro.com

garaemon, 2015-10-01:
I understood

garaemon, 2015-10-01:
BTW, rsync with renaming file may be possible. http://stackoverflow.com/questions/1312411/using-rsync-to-rename-files-during-copying-with-files-from

closed 2015-10-01 by garaemon

wkentaro, 2015-10-01:
I see, so maybe we can change the behavior when the force renaming is realized really dangerous.
or add option --no-local-rename.