jsk-ros-pkg/jsk_common #1167: rosbagを再生しながらgazebo等のシミュレーションを実行したい

kindsenior, 2015-10-10:
rosbagを再生しながらgazebo等のシミュレーションを実行したいです

具体的には
1. kinectで骨格認識
2. 認識した骨格をeusで関節角度に変換
3. gazeboやopenhrp3のシミュレーション上のロボットに関節角度を送る
としたいですが，
新しいプログラムを試すたびにkinectの前でポーズをとるのが大変なので，
kinectの出力するtfをrosbagで記録しておいて，2回目以降はrosbagを再生しながらシミュレーション上のロボットを動かしたいです

しかし，シミュレータを起動した状態でrosbagを再生するとシミュレータがおかしくなります．

例えば
```
roslaunch hrpsys_gazebo_tutorials gazebo_hrp2jsknt_no_controllers.launch
rtmlaunch hrpsys_gazebo_tutorials hrp2jsknt_hrpsys_bringup.launch
roseus kinect-model-viewer.l "(test-kinect)"
(use_sim_timeはtrue)
```
で
①rosbag play hogehoge.bag (clockなし)
とすると
rosbagの再生が一回しかできず，2回目以降の再生でeusのインタープリタに
```
Warning: TF_OLD_DATA ignoring data from the past for frame openni_depth_frame at time 1.44429e+09 according to authority unknown_publisher
Possible reasons are listed at http://wiki.ros.org/tf/Errors%20explained
         at line 260 in /tmp/buildd/ros-indigo-tf2-0.5.12-0trusty-20150806-1828/src/buffer_core.cpp
Warning: TF_OLD_DATA ignoring data from the past for frame openni_depth_frame at time 1.44429e+09 according to authority unknown_publisher
```
となりsubscribeできません

②rosbag play hogehoge.bag --clock (clockあり)
とすると
(gazebo)
```
[ WARN] [1444464440.303329594, 1444294529.858223449]: timeout: loose synchronization / you should check hrpsys is working
```

(hrpsys)
```
[ERROR] [WallTime: 1444464420.221222] [23.130000] ROS time moved backwards: 1444294486.65s
iob::critical overrun (1444294486648.534668[ms]), w:1444294509.779535 -> 23.131000
Warning: TF_OLD_DATA ignoring data from the past for frame CARMINE at time 23.609 according to authority unknown_publisher
Possible reasons are listed at http://wiki.ros.org/tf/Errors%20explained
         at line 260 in /tmp/buildd/ros-indigo-tf2-0.5.12-0trusty-20150806-1828/src/buffer_core.cpp
```
となってシミュレータ等がおかしくなります

今の所解決案としてあるのは以下の2つなのですが，他にありますでしょうか
a)rosbagの再生をtf:=tf_oldとしてtf_oldをsubsucribeしてタイムスタンプだけを書き換えて再びpublishするnodeを実装する
b)マネキンを使う

garaemon, 2015-10-10:
> kinectの出力するtfをrosbagで記録しておいて

tf以外で出力できないのかな

garaemon, 2015-10-10:
これはあんまりよくないけど

1. rosbag record /tfしておく
2. `rosbag play foo.bag /tf:=/tf_bone`する
3. timestampをうまいこと解決して`/tf_bone` :arrow_right: `/tf`するスクリプトを作る

もしくは

3. eusのプログラム?で`/tf`ではなくて`/tf_bone`をsubscribeする。骨格以外のtfつかってないなら、`rosrun roseus foo.l /tf:=/tf_bone`

wkentaro, 2015-10-10:
a)でやったのが
https://gist.github.com/wkentaro/2b38d8eb914c729197c6
です。
rosbagが解決してくれるといいと思うのでissue切っておきました。
https://github.com/ros/ros_comm/issues/678

wkentaro, 2015-10-13:
>    3. timestampをうまいこと解決して/tf_bone [image: :arrow_right:] /tfするスクリプトを作る
>    1. eusのプログラム?で/tfではなくて/tf_boneをsubscribeする。骨格以外のtfつかってないなら、rosrun

* gazeboの/clockを別のトピックへリマップ
* rosbag play bagfile.bag --clock

でも出来ました。

wkentaro, 2015-11-16:
このIssueは閉じて大丈夫でしょうか。
@kindsenior

kindsenior, 2015-11-17:
はい，閉じます

closed 2015-11-17 by kindsenior