jsk-ros-pkg/jsk_common #1179: rosbagを右クリックでplotする

eisoku9618, 2015-10-14:
@oshiroy 先ほど言っていた右クリックでrosbagをプロットする方法です．

``mkdir -p ~/.gnome2/nautilus-scripts``して，以下のスクリプトを``plot.sh``などの名前で保存するとできました．

```bash
#!/bin/bash
gnome-terminal --window -e "rosrun jsk_tools bag_plotter.py $(rospack find jaxon_ros_bridge)/config/plotter.yaml  $NAUTILUS_SCRIPT_SELECTED_FILE_PATHS"
exit
```

garaemon, 2015-10-14:
インタラクティブにyamlを選択

```
#!/bin/bash
gnome-terminal --window -e "rosrun jsk_tools bag_plotter.py $( zenity --file-selection --filename=$(rospack find jaxon_ros_bridge)/config/plotter.yaml --file-filter=*.yaml) $NAUTILUS_SCRIPT_SELECTED_FILE_PATHS"
exit
```

garaemon, 2015-10-14:
[image]

garaemon, 2015-10-14:
zenityおもしろいですね

```
$ rostopic echo $(zenity --list --column topics $(rostopic list))
$ rosbag record $(zenity --list --column topics $(rostopic list) --multiple --separator=' ')
```

eisoku9618, 2015-10-21:
一番最初のスクリプトが動かなくなっている件ですが，手元では以下で動きました．

```bash
#!/bin/bash
gnome-terminal -t "my_title" -x $SHELL -ic 'rosrun jsk_tools bag_plotter.py `rospack find jaxon_ros_bridge`/config/plotter.yaml $NAUTILUS_SCRIPT_SELECTED_FILE_PATHS'
exit
```

eisoku9618, 2015-10-21:
ダブルクォートではなくシングルクォートであることがポイント？

```bash
#!/bin/bash
gnome-terminal -t "aho" -x $SHELL -ic 'rosrun jsk_tools bag_plotter.py $(zenity --file-selection --filename=$(rospack find jaxon_ros_bridge)/config/plotter.yaml --file-filter=*.yaml) $NAUTILUS_SCRIPT_SELECTED_FILE_PATHS'
exit
```

wkentaro, 2015-11-16:
同じことをpercolでやると
https://github.com/jsk-ros-pkg/jsk_common/pull/1217
になりますか。

このIssueは閉じて大丈夫でしょうか。

closed 2015-11-17 by eisoku9618