jsk-ros-pkg/jsk_common #1208: gazebo関連de

mmurooka, 2015-11-02:


mmurooka, 2015-11-02:
@furushchev と話していていまして，
hrpsysに依存しないgazebo関連のプログラムをおくパッケージ/リポジトリを作ったほうがよいと
思いますがいかがでしょうか．
例えば，以下のようなプログラムの置き場所に迷わなくなると思います．

・ kinectをインタラクティブマーカありで動かす（https://github.com/start-jsk/rtmros_tutorials/blob/master/hrpsys_gazebo_tutorials/launch/gazebo_kinect_interactive.launch）
・ voxelをurdfに変換してgazeboにspawnする（https://github.com/jsk-ros-pkg/jsk_recognition/pull/1295#issuecomment-152360643）
・ @otsubo のRSJに出てくる家電プラグイン（スイッチを押したら扉が開くみたいなプラグイン）

今まではhrpsys_gazebo_tutorialsにおいたりしていましたが，
これはhrpsys以外の人は標準で入れていないようで使いづらくなっております．

eusモデルをurdfモデルに変換するプログラムは，
もともとhrpsys_gazebo_tutorialsにありましたが，
PR2の人も使うということでeusurdfに移しました．
https://github.com/jsk-ros-pkg/jsk_model_tools/pull/122

furushchev, 2015-11-02:
程度問題かもしれませんが、新しくレポジトリを作るに +1です。

- 必要になる人が全員でない（実機のみの人はいらない）
- 他のレポジトリに入れるとコンパイルが遅くなる
- hrpsys_なんちゃらは `hrpsys_` prefixがついた途端に依存関係が難しくなる
- hrpsys_gazebo_tutorialsがリリースされていない

ぐらいの理由が考えられます。

k-okada, 2015-11-02:
> jsk-ros-pkg/jsk_model_tools#122

とりあえずgazebo_ros_pkg では？
作るのは簡単だけど，すぐに放置されるよね -> 
ros/rosdistro/pull/9742, ros/rosdistro/pull/9733

k-okada, 2015-11-02:
送信済みのPRのリストを書いてみましょうか．

mmurooka, 2015-11-02:
はい，そうします．
そこまで一般的でないプログラムのような気がしていますが，
とりあえずPRして返事を聞いてみようと思います．

k-okada, 2015-11-02:
一般的にするにはどうなるかも考えてみましょう．