jsk-ros-pkg/jsk_common #1228: travisが失敗しやすい

k-okada, 2015-11-11:
https://travis-ci.org/jsk-ros-pkg/jsk_common/jobs/90380856
```
+ catkin run_tests --no-deps --limit-status-rate 0.001 dynamic_tf_publisher image_view2 jsk_data jsk_network_tools jsk_tilt_laser jsk_tools jsk_topic_tools multi_map_server virtual_force_publisher -p4 --make-args -j8 --
==> Expanding alias 'run_tests' from 'catkin run_tests --no-deps --limit-status-rate 0.001 dynamic_tf_publisher image_view2 jsk_data jsk_network_tools jsk_tilt_laser jsk_tools jsk_topic_tools multi_map_server virtual_force_publisher -p4 --make-args -j8 --' to 'catkin build --verbose --catkin-make-args run_tests -- --no-deps --limit-status-rate 0.001 dynamic_tf_publisher image_view2 jsk_data jsk_network_tools jsk_tilt_laser jsk_tools jsk_topic_tools multi_map_server virtual_force_publisher -p4 --make-args -j8 --'
-----------------------------------------------------------------
Profile:                     default
Extending:          [cached] /opt/ros/indigo
Workspace:                   /workspace/ros/ws_jsk_common
Source Space:       [exists] /workspace/ros/ws_jsk_common/src
Build Space:        [exists] /workspace/ros/ws_jsk_common/build
Devel Space:        [exists] /workspace/ros/ws_jsk_common/devel
Install Space:     [missing] /workspace/ros/ws_jsk_common/install
DESTDIR:                     None
-----------------------------------------------------------------
Isolate Develspaces:         False
Install Packages:            False
Isolate Installs:            False
-----------------------------------------------------------------
Additional CMake Args:       None
Additional Make Args:        None
Additional catkin Make Args: run_tests
Internal Make Job Server:    True
-----------------------------------------------------------------
Whitelisted Packages:        None
Blacklisted Packages:        None
-----------------------------------------------------------------
Workspace configuration appears valid.
----------------------------------------------------------------- 
Found '9' packages in 0.0 seconds. 
Starting ==> dynamic_tf_publisher                                               
Starting ==> image_view2                                                        
Starting ==> jsk_data                                                           
Starting ==> jsk_network_tools                                                  
                                                                                
[dynamic_tf_publisher] ==> '/workspace/ros/ws_jsk_common/build/dynamic_tf_publisher/build_env.sh /usr/bin/make cmake_check_build_system' in '/workspace/ros/ws_jsk_common/build/dynamic_tf_publisher'
[dynamic_tf_publisher] <== '/workspace/ros/ws_jsk_common/build/dynamic_tf_publisher/build_env.sh /usr/bin/make cmake_check_build_system' finished with return code '0'
                                                                                 
                                                                                
[jsk_data] ==> '/workspace/ros/ws_jsk_common/build/jsk_data/build_env.sh /usr/bin/make cmake_check_build_system' in '/workspace/ros/ws_jsk_common/build/jsk_data'
[jsk_data] <== '/workspace/ros/ws_jsk_common/build/jsk_data/build_env.sh /usr/bin/make cmake_check_build_system' finished with return code '0'
                                                                                 
                                                                                
[jsk_network_tools] ==> '/workspace/ros/ws_jsk_common/build/jsk_network_tools/build_env.sh /usr/bin/make cmake_check_build_system' in '/workspace/ros/ws_jsk_common/build/jsk_network_tools'
[jsk_network_tools] <== '/workspace/ros/ws_jsk_common/build/jsk_network_tools/build_env.sh /usr/bin/make cmake_check_build_system' finished with return code '0'
                                                                                 
                                                                                
[image_view2] ==> '/workspace/ros/ws_jsk_common/build/image_view2/build_env.sh /usr/bin/make cmake_check_build_system' in '/workspace/ros/ws_jsk_common/build/image_view2'
[image_view2] <== '/workspace/ros/ws_jsk_common/build/image_view2/build_env.sh /usr/bin/make cmake_check_build_system' finished with return code '0'
                                                                                 
                                                                                
[dynamic_tf_publisher] ==> '/workspace/ros/ws_jsk_common/build/dynamic_tf_publisher/build_env.sh /usr/bin/make run_tests --jobserver-fds=3,5 -j' in '/workspace/ros/ws_jsk_common/build/dynamic_tf_publisher'
Scanning dependencies of target run_tests
Built target run_tests
[dynamic_tf_publisher] <== '/workspace/ros/ws_jsk_common/build/dynamic_tf_publisher/build_env.sh /usr/bin/make run_tests --jobserver-fds=3,5 -j' finished with return code '0'
                                                                                 
Finished <== dynamic_tf_publisher    [ 0.6 seconds ]                            
Starting ==> jsk_tilt_laser                                                     
                                                                                
[jsk_tilt_laser] ==> '/workspace/ros/ws_jsk_common/build/jsk_tilt_laser/build_env.sh /usr/bin/make cmake_check_build_system' in '/workspace/ros/ws_jsk_common/build/jsk_tilt_laser'
[jsk_tilt_laser] <== '/workspace/ros/ws_jsk_common/build/jsk_tilt_laser/build_env.sh /usr/bin/make cmake_check_build_system' finished with return code '0'
                                                                                 
                                                                                
[jsk_network_tools] ==> '/workspace/ros/ws_jsk_common/build/jsk_network_tools/build_env.sh /usr/bin/make run_tests --jobserver-fds=3,5 -j' in '/workspace/ros/ws_jsk_common/build/jsk_network_tools'
Scanning dependencies of target run_tests
Built target run_tests
[jsk_network_tools] <== '/workspace/ros/ws_jsk_common/build/jsk_network_tools/build_env.sh /usr/bin/make run_tests --jobserver-fds=3,5 -j' finished with return code '0'
                                                                                 
Finished <== jsk_network_tools       [ 0.8 seconds ]                            
Starting ==> jsk_tools                                                          
                                                                                
[... 714 lines cut ...]
rostest log file is in /workspace/jsk-ros-pkg/jsk_common/log/rostest-991c279ff3ae-16405.log
-- run_tests.py: verify result "/workspace/ros/ws_jsk_common/build/jsk_topic_tools/test_results/jsk_topic_tools/rostest-test_test_topic_buffer.xml"
Built target _run_tests_jsk_topic_tools_rostest_test_test_topic_buffer.test
Scanning dependencies of target _run_tests_jsk_topic_tools_rostest
Built target _run_tests_jsk_topic_tools_rostest
Scanning dependencies of target _run_tests_jsk_topic_tools
Built target _run_tests_jsk_topic_tools
Scanning dependencies of target run_tests
Built target run_tests
[jsk_topic_tools] <== '/workspace/ros/ws_jsk_common/build/jsk_topic_tools/build_env.sh /usr/bin/make run_tests --jobserver-fds=3,5 -j' finished with return code '0'
                                                                                 
Finished <== jsk_topic_tools         [ 1 minute and 30.5 seconds ]              
[build] Finished.                                                               
[build] Runtime: 1 minute and 31.8 seconds 
+ '[' catkin == catkin ']'
+ find build -iname 'MISSING*' -print -exec rm '{}' ';'
+ catkin_test_results build
jsk_topic_tools/test_results/jsk_topic_tools/rostest-test_test_topic_buffer_fixed_rate.xml: 3 tests, 0 errors, 1 failures
jsk_topic_tools/test_results/jsk_topic_tools/rostest-test_test_topic_buffer_fixed_rate_and_update_rate.xml: 3 tests, 0 errors, 1 failures
jsk_topic_tools/test_results/jsk_topic_tools/rostest-test_test_topic_buffer_update_rate.xml: 3 tests, 0 errors, 2 failures
Summary: 32 tests, 0 errors, 4 failures
+ error
+ travis_time_end 31
+ set +x
```

k-okada, 2015-11-11:
```
[testhztest_chatter][failed]
[testhztest_chatter_buffer][failed]
[testtest_connection][failed]
``
あたりか

wkentaro, 2015-11-16:
* https://github.com/jsk-ros-pkg/jsk_common/pull/1213
* https://github.com/jsk-ros-pkg/jsk_common/pull/1239

で安定してくると思います。

wkentaro, 2015-11-16:
@k-okada 
上の二つのPRがマージされたので一旦このIssueを閉じてもいいかと思います．
また再発したらReopenするという形の対応でどうでしょうか．

k-okada, 2015-11-16:
ああ，確認せずにマージしてしまった．
3 timesでいいんだけど，余りこれをやると時間以内にTravisが終わらなくなるので，根本原因をしらべたいところ．
ただ単にnice すれば解決するんじゃないか，とも思いました．

close #1228

k-okada, 2015-11-17:
travisの制限時間は50分なので，jsk_recognitionは危ないですね．
jsk_commonもjsk_3rdpartyとか分ける前は危なかったですが，今は大丈夫ですね．

リトライ増やすのはまぁまぁ正解ですが，でも，errorを大きくしたり，
リトライ増やすのが正解だとテンプレートで理解する人がでてくる危険
がありそうなので，一言．

closed 2015-11-17 by k-okada

wkentaro, 2015-11-17:
ありがとうございます，了解です．
リトライは使わなく済むようなテストを書くという方針だと思っておきます．

wkentaro, 2015-11-17:
30〜45分かかっています．
[image]