jsk-ros-pkg/jsk_common #1230: [jsk_topic_tools] Why roscpp_tutorials is listed as build_depend?

wkentaro, 2015-11-11:
@k-okada 
Why jsk_topic_tools depends on roscpp_tutorials?
We can not test general tools in jsk_topic_tools by changing programs in roscpp_tutorials because of that.

I found this change 9cd6f44a9dc62d8a79078ed0c686b95c2c0ac324

k-okada, 2015-11-12:
https://github.com/jsk-ros-pkg/jsk_common/pull/279/files

i think we used roscpp_tutorials at somewhere in test code

k-okada, 2015-11-12:
very hard to follow file history but may be originated at
https://github.com/jsk-ros-pkg/jsk_common/commit/8248415f156abb8e78fe0e613d2f006ebe71443d

k-okada, 2015-11-12:
and still used in
https://github.com/jsk-ros-pkg/jsk_common/blob/master/jsk_topic_tools/test/test_topic_buffer_close_wait.test#L8

k-okada, 2015-11-12:
that should be in test_depend but as far as I know ,rosdep did not install
package in test_depend

wkentaro, 2015-11-12:
That rosdep behavior seems changed at REP-0140.
https://github.com/ros-infrastructure/rep/pull/59/files#diff-3fa963019eccb75f6893ad65b7317a69R46

So currently, test_depend may work as we expect.

closed 2015-11-16 by k-okada