jsk-ros-pkg/jsk_common #1247: image_view2ga

mmurooka, 2015-11-26:


YuOhara, 2015-11-26:
image_view２のrectangleモードが、左上ー＞右下でない方向で選んだときに、
image_view2がエラーをはいて死ぬ気がします。

YuOhara, 2015-11-26:
忘れないようにissueにしましたが、暇になったら自分で直します。

garaemon, 2015-12-04:
前に直した気がするんだがなぁ

k-okada, 2015-12-05:
1f021573f4b76e03ba2ab8195deecdb779cdecac
の動画像がこれだとすると2.0.3か2.0.4ではうごいていたことになりますね
戻して確認してにみましょう @yuohara

YuOhara, 2015-12-09:
2.0.3だと動いてました・・・

YuOhara, 2015-12-09:
2.04だと動かないです・・・

YuOhara, 2015-12-09:
```
leus@leus-ThinkPad-W541:image_view2 [(detached from 2.0.4)]$ git log .
commit 560d701f62eddb2b914b4fe42d5d78e593f0aad9
Author: Kei Okada <kei.okada@gmail.com>
Date:   Wed Nov 25 22:27:50 2015 +0900

    2.0.4

commit 6bef6e7f6f4e0e23db7ea78e2cef53f1d7e1dd6f
Author: Kei Okada <kei.okada@gmail.com>
Date:   Wed Nov 25 22:27:30 2015 +0900

    update CHANGELOG

commit c5a2d8ba7e00deb6091a07e23f12799adc40a8a4
Merge: 81afab1 232078c
Author: Kei Okada <k-okada@jsk.t.u-tokyo.ac.jp>
Date:   Tue Nov 10 10:55:59 2015 +0900

    Merge pull request #1219 from wkentaro/image_view2-depth-image
    
    [image_view2] Visualize depth image

commit 232078c7fd9bfdc1e970961532a7763eef30e6df
Author: Kentaro Wada <www.kentaro.wada@gmail.com>
Date:   Mon Nov 9 22:24:38 2015 +0900

    [image_view2] Visualize depth image

commit a3ac6a1770c62a1ac6c63847ae47807b0bff0d3f
Author: Kentaro Wada <www.kentaro.wada@gmail.com>
Date:   Mon Nov 9 20:19:24 2015 +0900

    [image_view2] Describe about msg types with topics

commit db488f0cabc91e5a4cf36d3f68915707d1f0ed20
Author: Kentaro Wada <www.kentaro.wada@gmail.com>
Date:   Mon Nov 9 20:15:36 2015 +0900

    [image_view2] Describe about some publising topics

commit 3659edb2b42eca35d12f01d857135d4d8b63e134
Author: Kentaro Wada <www.kentaro.wada@gmail.com>
Date:   Mon Nov 9 20:05:43 2015 +0900

    [image_view2] Publish cropped image with roi
    
    Currently this is only work with rectangle mode.

commit b812656d855addca41f67c0958b755917beece33
Author: Kentaro Wada <www.kentaro.wada@gmail.com>
Date:   Mon Nov 9 19:48:49 2015 +0900

    [image_view2] Describe ~interactive_mode actual value

commit 1434087c9aa029a7143d5f397ae812cb12d04679
Author: Ryohei Ueda <garaemon@gmail.com>
Date:   Thu Oct 1 18:43:35 2015 +0900

    [image_view2] Add document about grabcut rect interaction

commit 3894a1865dde0e55e0e3e52ae32caffbfd5da4b8
Author: Ryohei Ueda <garaemon@gmail.com>
Date:   Thu Oct 1 16:02:58 2015 +0900

    [image_view2] Add document about grabcut interaction

commit e2df835de198e0b777c6b76c109ef6a2bfef7a8f
Author: Ryohei Ueda <garaemon@gmail.com>
Date:   Tue Sep 29 23:16:41 2015 +0900

    [image_view2] Add document about poly interaction

commit 01ad5677836013aa7a18b372fcfbb51b8c6eb9ab
Author: Ryohei Ueda <garaemon@gmail.com>
Date:   Tue Sep 29 21:47:48 2015 +0900

    [image_view2] Add document about line interaction

commit 1f021573f4b76e03ba2ab8195deecdb779cdecac
Author: Ryohei Ueda <garaemon@gmail.com>
Date:   Tue Sep 29 19:48:52 2015 +0900

    [image_view2] Add documentation about rectangle mode interaction

commit 4f7a0d6b2061ebe551559ddc74a73395f3711c36
Author: Ryohei Ueda <garaemon@gmail.com>
Date:   Tue Sep 29 18:45:40 2015 +0900

    [image_view2] Add document about subscribing topics and advertising services
Author: Ryohei Ueda <garaemon@gmail.com>
Date:   Tue Sep 29 18:45:40 2015 +0900

    [image_view2] Add document about subscribing topics and advertising services

commit d2c774c277e2b85a5893071fa91c5e45663cd1df
Author: Ryohei Ueda <garaemon@gmail.com>
Date:   Tue Sep 29 13:32:24 2015 +0900

    [image_view2] Fix typo in document: ImageMarker -> ImageMarker2

commit 98721798f03367c7224e98b88b142df76629e802
Author: Ryohei Ueda <garaemon@gmail.com>
Date:   Tue Sep 29 12:42:47 2015 +0900

    [image_view2] Add README.md

commit a6538aa0eb42c8220d34b798b4b82c23a53c110f
Author: Ryohei Ueda <garaemon@gmail.com>
Date:   Fri Jul 24 15:49:05 2015 +0900

    2.0.3

commit e5cf7e3ef1df816c8dd764274c8cd51f32063bca
```

k-okada, 2015-12-09:
https://github.com/jsk-ros-pkg/jsk_common/commits/master/jsk_ros_patch/image_view2
のどこかですね．

ところで，ダメなtagをさがすときは，
2.0.3 -> 2.0.6 -> 2.0.5
                      -> 2.0.7
とbinary search するとよいです．

k-okada, 2015-12-09:
```
k-okada@kokada-t440s:~/ros/indigo/src/jsk-ros-pkg/jsk_common/jsk_ros_patch/image_view2$ git diff 2.0.3...2.0.4 *
diff --git a/jsk_ros_patch/image_view2/CHANGELOG.rst b/jsk_ros_patch/image_view2/CHANGELOG.rst
index 9e3a9de..ceaaaa9 100644
--- a/jsk_ros_patch/image_view2/CHANGELOG.rst
+++ b/jsk_ros_patch/image_view2/CHANGELOG.rst
@@ -2,6 +2,23 @@
 Changelog for package image_view2
 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
 
+2.0.4 (2015-11-25)
+------------------
+* [image_view2] Visualize depth image
+* [image_view2] Describe about msg types with topics
+* [image_view2] Describe about some publising topics
+* [image_view2] Publish cropped image with roi Currently this is only work with rectangle mode.
+* [image_view2] Describe ~interactive_mode actual value
+* [image_view2] Add document about grabcut rect interaction
+* [image_view2] Add document about grabcut interaction
+* [image_view2] Add document about poly interaction
+* [image_view2] Add document about line interaction
+* [image_view2] Add documentation about rectangle mode interaction
+* [image_view2] Add document about subscribing topics and advertising services
+* [image_view2] Fix typo in document: ImageMarker -> ImageMarker2
+* [image_view2] Add README.md
+* Contributors: Kei Okada, Kentaro Wada, Ryohei Ueda
+
 2.0.3 (2015-07-24)
 ------------------
 
diff --git a/jsk_ros_patch/image_view2/README.md b/jsk_ros_patch/image_view2/README.md
new file mode 100644
index 0000000..c4041e6
--- /dev/null
+++ b/jsk_ros_patch/image_view2/README.md
@@ -0,0 +1,209 @@
+image_view2
+==========
+![](images/image_view2_rect_interaction.gif)
+
+`image_view2` is an extended [image_view](http://ros.org/wiki/image_view).
+
+Features
+* Draw markers on image view
+* Interact with user direction
+
+Interaction Modes
+-----------------
+image_view2 support several mode to interact with user.
+You can set interaction mode by `~interaction_mode` parameter or
+change mode by service interfaces.
+
+* Rectangle mode (`~interaction_mode:=rectangle`)
+
+  In rectangle mode, user can select region-of-interest by dragging mouse pointer.
+
+  ![](images/image_view2_rect_interaction.gif)
+
+  Selected regions are published to
+  `image/screenrectangle` (`geometry_msgs/PolygonStamped`) topic.
+
+  If user clicks on the view instead of dragging pointer, clicked position
+  is published to `image/screenpoint` (`geometry_msgs/PointStamped`) topic.
+
+* Line mode (`~interaction_mode:=line`)
+
+  In line mode, user can select a line by dragging mouse pointer.
+
+  ![](images/image_view2_line_interaction.gif)
+
[... 263 lines cut ...]
diff --git a/jsk_ros_patch/image_view2/images/image_view2_poly_interaction.gif b/jsk_ros_patch/image_view2/images/image_view2_poly_interaction.gif
new file mode 100644
index 0000000..8909c4b
Binary files /dev/null and b/jsk_ros_patch/image_view2/images/image_view2_poly_interaction.gif differ
diff --git a/jsk_ros_patch/image_view2/images/image_view2_rect_interaction.gif b/jsk_ros_patch/image_view2/images/image_view2_rect_interaction.gif
new file mode 100644
index 0000000..44b7138
Binary files /dev/null and b/jsk_ros_patch/image_view2/images/image_view2_rect_interaction.gif differ
diff --git a/jsk_ros_patch/image_view2/images/image_view2_screenrectangle_image.png b/jsk_ros_patch/image_view2/images/image_view2_screenrectangle_image.png
new file mode 100644
index 0000000..e1807c3
Binary files /dev/null and b/jsk_ros_patch/image_view2/images/image_view2_screenrectangle_image.png differ
diff --git a/jsk_ros_patch/image_view2/package.xml b/jsk_ros_patch/image_view2/package.xml
index 941fe74..270d2aa 100644
--- a/jsk_ros_patch/image_view2/package.xml
+++ b/jsk_ros_patch/image_view2/package.xml
@@ -1,6 +1,6 @@
 <package>
   <name>image_view2</name>
-  <version>2.0.3</version>
+  <version>2.0.4</version>
   <description>A simple viewer for ROS image topics with draw-on features</description>
   <maintainer email="k-okada@jsk.t.u-tokyo.ac.jp">Kei Okada</maintainer>
 
```

YuOhara, 2015-12-09:
https://github.com/jsk-ros-pkg/jsk_common/commit/3659edb2b42eca35d12f01d857135d4d8b63e134
調べたところこの前後で動かなくなってました。
原因がわかった気がするので、直してPR送ります。

wkentaro, 2015-12-09:
The regression is caused by my PR, sorry.

garaemon, 2015-12-10:
image_view2はマウスイベントをtopic経由でも送れる&&GUIなしモードがあるので、テストを書こうと思えばかけます。

closed 2015-12-14 by garaemon