jsk-ros-pkg/jsk_common #1271: Release 2.0.7

garaemon, 2015-12-06:
Tagged 2.0.7 but failed to release by jenkins job.

2.0.4-2.0.6 looks released by @k-okada manually.

wkentaro, 2015-12-06:
We need to resolve issue at https://github.com/jsk-ros-pkg/jsk_common/pull/1261#issuecomment-162285795. 

My question is why jsk_common need to be released for saucy and utopic.

k-okada, 2015-12-07:
> We need to resolve issue at #1261 (comment)

indigo supporting S and T, and jade supports T, U, V
http://www.ros.org/reps/rep-0003.html#indigo-igloo-may-2014

k-okada, 2015-12-07:
https://github.com/jsk-ros-pkg/jsk_common/commit/fd0a3155450c91ed8158e1c534551055a102f436
```
Releasing package 'dynamic_tf_publisher' for 'indigo' to:
'release/indigo/dynamic_tf_publisher'
Releasing package 'jsk_tilt_laser' for 'indigo' to:
'release/indigo/jsk_tilt_laser'
Releasing package 'jsk_data' for 'indigo' to: 'release/indigo/jsk_data'
 [git-bloom-patch import]: Applied 1 patches
Releasing package 'jsk_network_tools' for 'indigo' to:
'release/indigo/jsk_network_tools'
Releasing package 'multi_map_server' for 'indigo' to:
'release/indigo/multi_map_server'
Releasing package 'image_view2' for 'indigo' to:
'release/indigo/image_view2'
Releasing package 'jsk_common' for 'indigo' to: 'release/indigo/jsk_common'
Releasing package 'jsk_topic_tools' for 'indigo' to:
'release/indigo/jsk_topic_tools'
 [git-bloom-patch import]: Applied 1 patches
Releasing package 'jsk_tools' for 'indigo' to: 'release/indigo/jsk_tools'
 [git-bloom-patch import]: 'execute_command' failed to call 'git am
/tmp/tmpMLHcNu/*.patch' which had a return code (1):
 [git-bloom-patch import]: ```
Applying: pip packages need to removed to release as deb
error: patch failed: package.xml:17
error: package.xml: patch does not apply
Patch failed at 0001 pip packages need to removed to release as deb
When you have resolved this problem run "git am --resolved".
If you would prefer to skip this patch, instead run "git am --skip".
To restore the original branch and stop patching run "git am --abort".

 [git-bloom-patch import]: ```
 [git-bloom-patch import]: Failed to apply one or more patches for the
'Command 'git am /tmp/tmpMLHcNu/*.patch' returned non-zero exit status 1'
branch.

>>> Resolve any conflicts and when you have resolved this problem run 'git
am --resolved' and then exit the shell using 'exit 0'. <<<
    To abort use 'exit 1'
(bloom)cat
/tmp/tmpMLHcNu/0001-pip-packages-need-to-removed-to-release-as-deb.patch
From b8f2a6e3181995f09f64a09616cb80c360dfa304 Mon Sep 17 00:00:00 2001
From: Kei Okada <kei.okada@gmail.com>
Date: Mon, 4 May 2015 08:24:09 +0900
Subject: [PATCH] pip packages need to removed to release as deb

---
 package.xml |   10 +++++-----
 1 file changed, 5 insertions(+), 5 deletions(-)

diff --git a/package.xml b/package.xml
index 69b49d1..4ef952d 100644
--- a/package.xml
+++ b/package.xml
@@ -17,10 +17,10 @@

   <build_depend>git</build_depend>
   <build_depend>rosgraph_msgs</build_depend>
-  <run_depend>python-percol</run_depend>
-  <run_depend>python-colorama</run_depend>
-  <run_depend>python-pygithub3</run_depend>
-  <run_depend>python-slacker-cli</run_depend>
+  <!-- <run_depend>python-percol</run_depend> -->
+  <!-- <run_depend>python-colorama</run_depend> -->
+  <!-- <run_depend>python-pygithub3</run_depend> -->
+  <!-- <run_depend>python-slacker-cli</run_depend> -->
   <run_depend>rosgraph_msgs</run_depend>
   <run_depend>rospy</run_depend>
   <run_depend>rqt_reconfigure</run_depend>
@@ -29,7 +29,7 @@
   <run_depend>jsk_network_tools</run_depend>
   <run_depend>jsk_topic_tools</run_depend>
   <run_depend>rosbag</run_depend>
-  <run_depend>rosemacs-el</run_depend>
+  <run_depend>rosemacs</run_depend>
   <!-- <build_depend>mjpeg_server</build_depend> -->
   <!-- <run_depend>mjpeg_server</run_depend> -->
   <!-- <test_depend>mjpeg_server</test_depend> -->
--
1.7.9.5
```

garaemon, 2015-12-07:
on ros jenkins

```
CMake Error at CMakeLists.txt:37 (find_package):
  By not providing "Findrostest.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "rostest", but
  CMake did not find one.

  Could not find a package configuration file provided by "rostest" with any
  of the following names:

    rostestConfig.cmake
    rostest-config.cmake

  Add the installation prefix of "rostest" to CMAKE_PREFIX_PATH or set
  "rostest_DIR" to a directory containing one of the above files.  If
  "rostest" provides a separate development package or SDK, be sure it has
  been installed.
```

garaemon, 2015-12-07:
jsk_tools misses rostest in package.xml

k-okada, 2015-12-07:
can you paste link to this?

garaemon, 2015-12-07:
http://jenkins.ros.org/job/ros-indigo-jsk-tools_binarydeb_saucy_i386/45/

wkentaro, 2015-12-07:
> indigo supporting S and T, and jade supports T, U, V

I see, I will add these.

garaemon, 2015-12-07:
http://jenkins.jsk.imi.i.u-tokyo.ac.jp:8080/job/bloom_release/5871/console

Failed to release 2.0.8

wkentaro, 2015-12-07:
https://github.com/ros/rosdistro/pull/9985
is merged.

garaemon, 2015-12-08:
released, thanks

closed 2015-12-08 by garaemon