jsk-ros-pkg/jsk_common #1284: multisenseの画像topicの周期が遅い

orikuma, 2015-12-10:
まだ詳しく調べられていないので現象だけですが,
JAXON_REDにおいて/multisense_local/left/image_rectのtopicがpublishされる周期が遅いです.

JAXON_REDはmultisenseの解像度1024x544, 256 disparity, fps 30Hzで運用していますが,
現在30Hzが期待される/multisense_local/left/image_rectが8-5Hz程度しか出ていません.
ちなみに解像度を1024x272, 64disparityまで落とすと15Hz程度まで回復します.

```
$ rostopic hz /multisense_local/left/image_rect -w 10
subscribed to [/multisense_local/left/image_rect]
average rate: 3.965
        min: 0.252s max: 0.252s std dev: 0.00000s window: 2
average rate: 5.305
        min: 0.124s max: 0.500s std dev: 0.12468s window: 9
average rate: 5.962
        min: 0.124s max: 0.377s std dev: 0.08401s window: 10
average rate: 5.949
        min: 0.125s max: 0.253s std dev: 0.05899s window: 10
average rate: 5.961
        min: 0.124s max: 0.253s std dev: 0.05943s window: 10
average rate: 5.500
        min: 0.123s max: 0.379s std dev: 0.08705s window: 10
average rate: 7.103
        min: 0.123s max: 0.255s std dev: 0.04045s window: 10
```

garaemon, 2015-12-10:
ここではなくて、本家で聞いてください

k-okada, 2015-12-10:
聞いた先のリンクをはっておいてください @orikuma

orikuma, 2015-12-17:
https://bitbucket.org/crl/multisense_ros/issues/55/execution-rate-of-multisense-stereo-images

orikuma, 2015-12-17:
ついでにfirmwareの更新について:
https://bitbucket.org/crl/multisense_ros/issues/56/firmware-upgrade-for-multisense-sl

garaemon, 2015-12-19:
この問題、かなり複雑で面白い原因が複数あるので再オープンします

現在、`/tf`の周期を何とかして落とさないと画像の周期が遅くなる、というところまで来ています

@orikuma 問題を整理してください

1. linuxのUDP設定の問題
2. tfのような高周期トピックの問題

orikuma, 2015-12-19:
* 症状
JAXON_REDで/multisense_local/left/image_rectがpublishされる周期が理論周期よりも遅いという現象が発生しています(HRP2016でも同様の現象が発生しているが以降はJAXON_REDを対象として記述).
現在JAXON_REDでは解像度1024x544x256でmultisenseを使用しており,
スペック情報( http://carnegierobotics.com/support-updates/2015/7/16/frame-rates-at-various-resolutions-and-disparity-search-windows )を見るとこの場合の理論周期は15Hzとなっています.
しかし実際には上述の通り5Hz程度しか出ていません.

* 検証
a. multisense_driverから出てくる画像の周期がそもそも遅い問題
https://bitbucket.org/crl/multisense_ros/issues/55/execution-rate-of-multisense-stereo-images
で回答されたとおりPC側のネットワーク及びカーネルパラメータの変更を行うことで
multisense_driverを単独で上げた時に/multisense/multisense_driverから出てくる画像の周期が改善します(詳細は上記リンク参照).
しかしこの変更だけではmultisense_laser_pipeline等の他のnodeを上げた際に画像の周期が再度遅くなります(5-6Hz程度)
しかしこれだけだとmultisense_laser_pipeline等の

iory, 2018-07-23:
@k-okada
こちらassignされましたが、具体的に解決すればいい問題はどの部分になるでしょうか？

> aの変更だけではmultisense_laser_pipeline等の他のnodeを上げた際に画像の周期が再度遅くなります(5-6Hz程度).
この問題については/joint_states, /tfなどの複数のnodeが吐いていることにより高周期化しているtopicをsubscribeするnodeが増えるとある段階で画像の周期が遅くなっていく, という現象が確認されています.

そもそもの高周期のtopicをsubscribeすると画像の周期が遅くなっているという現象はroscoreなどのレベルでの問題になると思いますが、そちらを解決するのか、

>  高周期topicの周期を落として対処しようとしています.

こちらのように周期をおとして対処するということに焦点を当てて、
multisenseのhzが15Hzになるようにすればいいでしょうか？

k-okada, 2018-07-24:
現状どうなっているか再確認しましょう．tf2とかで良くなっているかもしれないです．

otsubo, 2018-07-27:
multisense周りのソースを最新にする前とした後で熊谷さんが報告していた解像度1024x544, 256 disparityでrostopic hz してみたところ
```
;;before update
[http://hrp2016v:10016][192.168.96.103] leus@fuji:~$ rostopic hz /multisense_local/left/image_rect -w 10
subscribed to [/multisense_local/left/image_rect]
average rate: 11.745
	min: 0.063s max: 0.128s std dev: 0.02954s window: 10
average rate: 12.822
	min: 0.063s max: 0.130s std dev: 0.02726s window: 10
average rate: 10.845
	min: 0.063s max: 0.129s std dev: 0.03169s window: 10
average rate: 10.877
	min: 0.062s max: 0.128s std dev: 0.03183s window: 10
average rate: 12.859
	min: 0.062s max: 0.128s std dev: 0.02678s window: 10

;;after update
[http://hrp2016v:10016][192.168.96.103] leus@fuji:~$ rostopic hz /multisense_local/left/image_rect -w 10
subscribed to [/multisense_local/left/image_rect]
average rate: 15.044
	min: 0.065s max: 0.068s std dev: 0.00092s window: 10
average rate: 15.000
	min: 0.065s max: 0.068s std dev: 0.00084s window: 10
average rate: 15.012
	min: 0.066s max: 0.067s std dev: 0.00047s window: 10
average rate: 14.979
	min: 0.066s max: 0.068s std dev: 0.00085s window: 10
average rate: 12.257
	min: 0.066s max: 0.134s std dev: 0.02776s window: 10
````
という感じの値になっていました.
2016年2月の時点で10~12Hzぐらい出ていたようです.
最新では理論値に近い値が出ています.
ただ,最新にするとエンバグしているのかhokuyoのtfがつながらなくなってしまって,原因を確認しています.

YoheiKakiuchi, 2018-07-27:
tfの問題は、
https://github.com/start-jsk/rtmros_tutorials/pull/504
じゃないでしょうか。

iory, 2018-07-27:
なるほど。
現状は3.xを使っているから起きていて、
ようやく4.0.0を使う必要性が出てきたということでしょうか？

ishiguroJSK, 2019-03-11:
数週間前(2019/03)からずっとmultisense_rosがcheckoutできないんですが，これは本家に報告しろという話なんでしょうか？
```
wstool update multisense 
[multisense] Updating /home/ishiguro/catkin_ws/master/src/multisense
中止: エラー: _ssl.c:510: error:1408F10B:SSL routines:SSL3_GET_RECORD:wrong version number
Exception caught during install: Error processing 'multisense' : [multisense] Update Failed of /home/ishiguro/catkin_ws/master/src/multisense

ERROR in config: Error processing 'multisense' : [multisense] Update Failed of /home/ishiguro/catkin_ws/master/src/multisense
```

```
hg clone https://bitbucket.org/crl/multisense_ros
中止: エラー: _ssl.c:510: error:1408F10B:SSL routines:SSL3_GET_RECORD:wrong version number
```

ishiguroJSK, 2019-03-11:
bitbucketがTLS辞めただの何だのが原因らしく，
~/.hgrcを作って
```
[ui]
tls = False
```
と書いておくという謎のソリューションに至りました．もっと賢い方法あったら教えてください

https://stackoverflow.com/questions/21477683/mercurial-https-clone-abort-error-wrong-version-number

k-okada, 2019-03-11:
hgのバージョンでしょうか。ところで、これでダメならtravisで引っかかるはずなんだけどな。
というのと、ソースで落として来なければいけな理由があるんだっけ？というのと、
別Issueにするとよいですね。

```
user@ubuntu-16:~$ hg clone https://bitbucket.org/crl/multisense_ros
destination directory: multisense_ros
requesting all changes
adding changesets
adding manifests
adding file changes
added 117 changesets with 1542 changes to 393 files
updating to branch default
s250 files updated, 0 files merged, 0 files removed, 0 files unresolved
```

ishiguroJSK, 2019-03-11:
> hgのバージョンでしょうか。ところで、これでダメならtravisで引っかかるはずなんだけどな。

2台の14.04で，片方はapt-get upgradeしてからも再現していたはずです．ただ，確かにtravisも14.04あるはずなので，落ち着いてまた確かめます．

> というのと、ソースで落として来なければいけな理由があるんだっけ？というのと、

最新のmultisense_rosは構成が大きく変わっていて，いろいろ対応しきれていないから，ということでバージョン固定していたはずです

> 別Issueにするとよいですね。

はい，何か意味のある展開になりましたら立てます．