jsk-ros-pkg/jsk_common #1287: 'Enum' object has no attribute 'SUBSCRIBED'

garaemon, 2015-12-10:
cannot import jsk_topic_tools.ConnectionBasedTransport

```
  File "./unwrap_histogram_with_range_array.py", line 7, in <module>
    from jsk_topic_tools import jsk_logdebug, jsk_loginfo, ConnectionBasedTransport
  File "/home/lueda/ros/hydro/devel/lib/python2.7/dist-packages/jsk_topic_tools/__init__.py", line 34, in <module>
    exec(__fh.read())
  File "<string>", line 4, in <module>
  File "/home/lueda/ros/hydro/src/jsk-ros-pkg/jsk_common/jsk_topic_tools/src/jsk_topic_tools/transport.py", line 18, in <module>
    SUBSCRIBED = ConnectionStatus.SUBSCRIBED
AttributeError: 'Enum' object has no attribute 'SUBSCRIBED'
```

I think enum should be 
```
ConnectionStatus = Enum('ConnectionStatus', 'SUBSCRIBED', 'NOT_SUBSCRIBED')
```

garaemon, 2015-12-10:
enum version is 0.4.6
```
$ pip freeze | grep enum
Warning: cannot find svn location for distribute==0.6.24dev-r0
enum==0.4.6
```

garaemon, 2015-12-10:
https://pypi.python.org/pypi/enum

wkentaro, 2015-12-10:
I think `enum34` is more common plugin, so `jsk_topic_tools` uses it, by `pip install enum34`.
so that problem should be resolved by
```
sudo pip install uninstall enum
sudo pip install enum34
```
http://stackoverflow.com/questions/36932/how-can-i-represent-an-enum-in-python
https://github.com/jsk-ros-pkg/jsk_common/blob/master/jsk_topic_tools/package.xml#L44

But someone already installed `enum` library so this can be problem, so I will remove enum and use just int.

PR #1288 (2015-12-10): [jsk_topic_tools] Stop using enum34 and use just int
Closes #1287
closed 2015-12-10 by garaemon