jsk-ros-pkg/jsk_common #1301: Failed to release new version from jenkins

garaemon, 2015-12-14:
http://jenkins.jsk.imi.i.u-tokyo.ac.jp:8080/job/bloom_release/5877/console

pip and rosemcas-el problem.

Need to run bloom-release manually?
```
22:29:00 Releasing for rosdistro: indigo[0m
22:29:00 
22:29:00 Pre-verifying Debian dependency keys...[0m
22:29:00 Running 'rosdep update'...[0m
22:29:00 Failed to resolve [36m[1mpython-click[0m on [34mubuntu[0m:[36m[1msaucy[0m with: Error running generator: The Debian generator does not support dependencies which are installed with the 'pip' installer.[0m
22:29:00 [36m[1mpython-click[0m is depended on by these packages: [0m['jsk_data']
22:29:00 [30m[1m<== [31m[1mFailed[0m[0m
22:29:00 Failed to resolve [36m[1mpython-click[0m on [34mubuntu[0m:[36m[1mtrusty[0m with: Error running generator: The Debian generator does not support dependencies which are installed with the 'pip' installer.[0m
22:29:00 [36m[1mpython-click[0m is depended on by these packages: [0m['jsk_data']
22:29:00 [30m[1m<== [31m[1mFailed[0m[0m
22:29:00 Failed to resolve [36m[1mpython-freezegun-pip[0m on [34mubuntu[0m:[36m[1msaucy[0m with: Error running generator: The Debian generator does not support dependencies which are installed with the 'pip' installer.[0m
22:29:00 [36m[1mpython-freezegun-pip[0m is depended on by these packages: [0m['jsk_data']
22:29:00 [30m[1m<== [31m[1mFailed[0m[0m
22:29:00 Failed to resolve [36m[1mpython-freezegun-pip[0m on [34mubuntu[0m:[36m[1mtrusty[0m with: Error running generator: The Debian generator does not support dependencies which are installed with the 'pip' installer.[0m
22:29:00 [36m[1mpython-freezegun-pip[0m is depended on by these packages: [0m['jsk_data']
22:29:00 [30m[1m<== [31m[1mFailed[0m[0m
22:29:00 Failed to resolve [36m[1mpython-percol[0m on [34mubuntu[0m:[36m[1msaucy[0m with: Error running generator: The Debian generator does not support dependencies which are installed with the 'pip' installer.[0m
22:29:00 [36m[1mpython-percol[0m is depended on by these packages: [0m['jsk_tools']
22:29:00 [30m[1m<== [31m[1mFailed[0m[0m
22:29:00 Failed to resolve [36m[1mpython-percol[0m on [34mubuntu[0m:[36m[1mtrusty[0m with: Error running generator: The Debian generator does not support dependencies which are installed with the 'pip' installer.[0m
22:29:00 [36m[1mpython-percol[0m is depended on by these packages: [0m['jsk_tools']
22:29:00 [30m[1m<== [31m[1mFailed[0m[0m
22:29:00 Failed to resolve [36m[1mpython-pygithub3[0m on [34mubuntu[0m:[36m[1msaucy[0m with: Error running generator: The Debian generator does not support dependencies which are installed with the 'pip' installer.[0m
22:29:00 [36m[1mpython-pygithub3[0m is depended on by these packages: [0m['jsk_tools']
22:29:00 [30m[1m<== [31m[1mFailed[0m[0m
22:29:00 Failed to resolve [36m[1mpython-pygithub3[0m on [34mubuntu[0m:[36m[1mtrusty[0m with: Error running generator: The Debian generator does not support dependencies which are installed with the 'pip' installer.[0m
22:29:00 [36m[1mpython-pygithub3[0m is depended on by these packages: [0m['jsk_tools']
22:29:00 [30m[1m<== [31m[1mFailed[0m[0m
22:29:00 Failed to resolve [36m[1mpython-slacker-cli[0m on [34mubuntu[0m:[36m[1msaucy[0m with: Error running generator: The Debian generator does not support dependencies which are installed with the 'pip' installer.[0m
22:29:00 [36m[1mpython-slacker-cli[0m is depended on by these packages: [0m['jsk_tools']
22:29:00 [30m[1m<== [31m[1mFailed[0m[0m
22:29:00 Failed to resolve [36m[1mpython-slacker-cli[0m on [34mubuntu[0m:[36m[1mtrusty[0m with: Error running generator: The Debian generator does not support dependencies which are installed with the 'pip' installer.[0m
22:29:00 [36m[1mpython-slacker-cli[0m is depended on by these packages: [0m['jsk_tools']
22:29:00 [30m[1m<== [31m[1mFailed[0m[0m
22:29:00 No definition of [rosemacs-el] for OS version [saucy]
22:29:00 	rosdep key : rosemacs-el
22:29:00 	OS name    : ubuntu
22:29:00 	OS version : saucy
22:29:00 	Data: ubuntu:
22:29:00   lucid:
22:29:00   - rosemacs-el
22:29:00   oneiric:
22:29:00   - rosemacs-el
22:29:00   precise:
22:29:00   - rosema[31m[1mCould not resolve rosdep key 'rosemacs-el' for distro 'trusty':[0m
22:29:01 [31m[1mSome of the dependencies for packages in this repository could not be resolved by rosdep.[0m
22:29:01 [31m[1mYou can try to address the issues which appear above and try again if you wish.[0m
22:29:01 Traceback (most recent call last):[39;49;00m
22:29:01   File [39;49;00m[36m"/usr/bin/git-bloom-generate"[39;49;00m, line [39;49;00m[34m9[39;49;00m, in [39;49;00m<module>[39;49;00m
22:29:01     [39;49;00mload_entry_point[39;49;00m([39;49;00m[33m'[39;49;00m[33mbloom==0.5.20[39;49;00m[33m'[39;49;00m,[39;49;00m [39;49;00m[33m'[39;49;00m[33mconsole_scripts[39;49;00m[33m'[39;49;00m,[39;49;00m [39;49;00m[33m'[39;49;00m[33mgit-bloom-generate[39;49;00m[33m'[39;49;00m)[39;49;00m([39;49;00m)[39;49;00m
22:29:01   File [39;49;00m[36m"/usr/lib/python2.7/dist-packages/bloom/commands/git/generate.py"[39;49;00m, line [39;49;00m[34m252[39;49;00m, in [39;49;00mmain[39;49;00m
22:29:01     [39;49;00mrun_generator[39;49;00m([39;49;00mgenerator[39;49;00m,[39;49;00m [39;49;00margs[39;49;00m)[39;49;00m
22:29:01   File [39;49;00m[36m"/usr/lib/python2.7/dist-packages/bloom/commands/git/generate.py"[39;49;00m, line [39;49;00m[34m127[39;49;00m, in [39;49;00mrun_generator[39;49;00m
22:29:01     [39;49;00mgen[39;49;00m.[39;49;00mpre_modify[39;49;00m)[39;49;00m
22:29:01   File [39;49;00m[36m"/usr/lib/python2.7/dist-packages/bloom/commands/git/generate.py"[39;49;00m, line [39;49;00m[34m101[39;49;00m, in [39;49;00mtry_execute[39;49;00m
22:29:01     [39;49;00mretcode[39;49;00m [39;49;00m=[39;49;00m [39;49;00mfunc[39;49;00m([39;49;00m*[39;49;00margs[39;49;00m,[39;49;00m [39;49;00m*[39;49;00m*[39;49;00mkwargs[39;49;00m)[39;49;00m
22:29:01   File [39;49;00m[36m"/usr/lib/python2.7/dist-packages/bloom/generators/debian/generator.py"[39;49;00m, line [39;49;00m[34m613[39;49;00m, in [39;49;00mpre_modify[39;49;00m
22:29:01     [39;49;00m[34mif[39;49;00m [39;49;00m[35mnot[39;49;00m [39;49;00mmaybe_continue[39;49;00m([39;49;00mmsg[39;49;00m=[39;49;00m[33m"[39;49;00m[33mWould you like to try again?[39;49;00m[33m"[39;49;00m)[39;49;00m:[39;49;00m
22:29:01   File [39;49;00m[36m"/usr/lib/python2.7/dist-packages/bloom/util.py"[39;49;00m, line [39;49;00m[34m387[39;49;00m, in [39;49;00mmaybe_continue[39;49;00m
22:29:01     [39;49;00mresponse[39;49;00m [39;49;00m=[39;49;00m [39;49;00msafe_input[39;49;00m([39;49;00mmsg[39;49;00m)[39;49;00m
22:29:01   File [39;49;00m[36m"/usr/lib/python2.7/dist-packages/bloom/util.py"[39;49;00m, line [39;49;00m[34m97[39;49;00m, in [39;49;00msafe_input[39;49;00m
22:29:01     [39;49;00mflush_stdin[39;49;00m([39;49;00m)[39;49;00m
22:29:01   File [39;49;00m[36m"/usr/lib/python2.7/dist-packages/bloom/util.py"[39;49;00m, line [39;49;00m[34m90[39;49;00m, in [39;49;00mflush_stdin[39;49;00m
22:29:01     [39;49;00mtcflush[39;49;00m([39;49;00msys[39;49;00m.[39;49;00mstdin[39;49;00m,[39;49;00m [39;49;00mTCIFLUSH[39;49;00m)[39;49;00m
[... 7 lines cut ...]
22:29:01 No definition of [rosemacs-el] for OS version [trusty]
22:29:01 	rosdep key : rosemacs-el
22:29:01 	OS name    : ubuntu
22:29:01 	OS version : trusty
22:29:01 	Data: ubuntu:
22:29:01   lucid:
22:29:01   - rosemacs-el
22:29:01   oneiric:
22:29:01   - rosemacs-el
22:29:01   precise:
22:29:01   - rosemacs-el
22:29:01 
22:29:01 Failed to resolve [36m[1mrosemacs-el[0m on [34mubuntu[0m:[36m[1mtrusty[0m with: Error running generator: Failed to resolve rosdep key 'rosemacs-el', aborting.[0m
22:29:01 [36m[1mrosemacs-el[0m is depended on by these packages: [0m['jsk_tools']
22:29:01 [30m[1m<== [31m[1mFailed[0m[0m
22:29:01 Processing release track settings for 'indigo'[0m
22:29:01 Checking upstream devel branch for package.xml(s)[0m
22:29:01 Submodule '.travis' (git://github.com/jsk-ros-pkg/jsk_travis) registered for path '.travis'
22:29:01 Cloning into '.travis'...
22:29:01 Submodule path '.travis': checked out '6b0f85792f6cc27507e12c1cd633013549c5dba0'
22:29:01 Looking for packages in 'master' branch... [0mfound 10 packages.
22:29:01 Detected version '2.0.9' from package(s): ['dynamic_tf_publisher', 'jsk_tilt_laser', 'jsk_data', 'jsk_network_tools', 'multi_map_server', 'image_view2', 'jsk_common', 'jsk_topic_tools', 'jsk_tools', 'virtual_force_publisher'][0m
22:29:01 
22:29:01 Executing release track 'indigo'[0m
```

k-okada, 2015-12-14:
need to apply patch manually, because of
https://github.com/jsk-ros-pkg/jsk_common/commit/347a11bf196bb2a5a756c8fad5eb24d276905189

k-okada, 2015-12-15:
and
https://github.com/jsk-ros-pkg/jsk_common/commit/61a69e9046722483f7c8236b764c7f584f23e3a0
memo axis_camera, pr2_computer_monitor is not released on jade.

-> done https://github.com/ros/rosdistro/pull/10060,
https://github.com/ros/rosdistro/pull/10061

closed 2015-12-16 by garaemon