jsk-ros-pkg/jsk_common #1304: ImportError: No module named enum

k-okada, 2015-12-15:
https://travis-ci.org/jsk-ros-pkg/jsk_recognition/jobs/96754581

```
[jsk_perception]   File "/workspace/ros/ws_jsk_recognition/src/jsk_recognition/jsk_perception/node_scripts/image_publisher.py", line 14, in <module> 
[jsk_perception]     from jsk_topic_tools import jsk_logerr                     
[jsk_perception]   File "/opt/ros/indigo/lib/python2.7/dist-packages/jsk_topic_tools/__init__.py", line 4, in <module> 
[jsk_perception]     from .transport import *                                   
[jsk_perception]   File "/opt/ros/indigo/lib/python2.7/dist-packages/jsk_topic_tools/transport.py", line 7, in <module> 
[jsk_perception]     from enum import Enum                                      
[jsk_perception] ImportError: No module named enum                              
[jsk_perception] Traceback (most recent call last):                             
[jsk_perception]   File "/workspace/ros/ws_jsk_recognition/src/jsk_recognition/jsk_perception/node_scripts/image_publisher.py", line 14, in <module> 
[jsk_perception]     from jsk_topic_tools import jsk_logerr                     
[jsk_perception]   File "/opt/ros/indigo/lib/python2.7/dist-packages/jsk_topic_tools/__init__.py", line 4, in <module> 
[jsk_perception]     from .transport import *                                   
[jsk_perception]   File "/opt/ros/indigo/lib/python2.7/dist-packages/jsk_topic_tools/transport.py", line 7, in <module> 
[jsk_perception]     from enum import Enum                                      
[jsk_perception] ImportError: No module named enum                              
[jsk_pcl_ros]     [                     <=>               ] 65,887,958  12.6MB/s 
[jsk_perception] libdc1394 error: Failed to initialize libdc1394                
[jsk_perception] Traceback (most recent call last):                             
[jsk_perception]   File "/workspace/ros/ws_jsk_recognition/src/jsk_recognition/jsk_perception/node_scripts/image_publisher.py", line 14, in <module> 
[jsk_perception]     from jsk_topic_tools import jsk_logerr                     
[jsk_perception]   File "/opt/ros/indigo/lib/python2.7/dist-packages/jsk_topic_tools/__init__.py", line 4, in <module> 
[jsk_perception]     from .transport import *                                   
[jsk_perception]   File "/opt/ros/indigo/lib/python2.7/dist-packages/jsk_topic_tools/transport.py", line 7, in <module> 
[jsk_perception]     from enum import Enum                                      
[jsk_perception] ImportError: No module named enum                              
[jsk_perception] Traceback (most recent call last):                             
[jsk_perception]   File "/workspace/ros/ws_jsk_recognition/src/jsk_recognition/jsk_perception/node_scripts/image_publisher.py", line 14, in <module> 
[jsk_perception]     from jsk_topic_tools import jsk_logerr                     
[jsk_perception]   File "/opt/ros/indigo/lib/python2.7/dist-packages/jsk_topic_tools/__init__.py", line 4, in <module> 
[jsk_perception]     from .transport import *                                   
[jsk_perception]   File "/opt/ros/indigo/lib/python2.7/dist-packages/jsk_topic_tools/transport.py", line 7, in <module> 
[jsk_perception]     from enum import Enum                                      
[jsk_perception] ImportError: No module named enum                              
[jsk_pcl_ros]     [                      <=>              ] 68,280,012  12.6MB/s 
[jsk_perception] libdc1394 error: Failed to initialize libdc1394                
[jsk_pcl_ros]     [                       <=>             ] 70,999,746  12.4MB/s 
[jsk_perception] Traceback (most recent call last):                             
[jsk_perception]   File "/workspace/ros/ws_jsk_recognition/src/jsk_recognition/jsk_perception/node_scripts/label_image_decomposer.py", line 7, in <module> 
[jsk_perception]     from jsk_recognition_utils import bounding_rect_of_mask    
[jsk_perception] Traceback (most recent call last):                             
[jsk_perception]   File "/workspace/ros/ws_jsk_recognition/devel/lib/python2.7/dist-packages/jsk_recognition_utils/__init__.py", line 35, in <module> 
[jsk_perception]   File "/workspace/ros/ws_jsk_recognition/src/jsk_recognition/jsk_perception/node_scripts/label_image_decomposer.py", line 7, in <module> 
[jsk_perception]     exec(__fh.read())                                          
[jsk_perception]       File "<string>", line 6, in <module>                     
[jsk_perception] from jsk_recognition_utils import bounding_rect_of_mask        
[jsk_perception]   File "/workspace/ros/ws_jsk_recognition/devel/lib/python2.7/dist-packages/jsk_recognition_utils/__init__.py", line 35, in <module> 
[jsk_perception]     exec(__fh.read())                                          
[jsk_perception]   File "<string>", line 6, in <module>                         
[jsk_perception]   File "/workspace/ros/ws_jsk_recognition/src/jsk_recognition/jsk_recognition_utils/python/jsk_recognition_utils/visualize.py", line 7, in <module> 
[jsk_perception]   File "/workspace/ros/ws_jsk_recognition/src/jsk_recognition/jsk_recognition_utils/python/jsk_recognition_utils/visualize.py", line 7, in <module> 
[jsk_perception]     import matplotlib.pyplot as plt                            
[jsk_perception]   File "/usr/local/lib/python2.7/dist-packages/matplotlib/pyplot.py", line 29, in <module> 
[jsk_perception]     import matplotlib.pyplot as plt                            
[jsk_perception]   File "/usr/local/lib/python2.7/dist-packages/matplotlib/pyplot.py", line 29, in <module> 
[jsk_perception]     import matplotlib.colorbar                                 
[jsk_perception]   File "/usr/local/lib/python2.7/dist-packages/matplotlib/colorbar.py", line 34, in <module> 
[jsk_perception]     import matplotlib.colorbar                                 
[jsk_perception]   File "/usr/local/lib/python2.7/dist-packages/matplotlib/colorbar.py", line 34, in <module> 
[jsk_perception]     import matplotlib.collections as collections               
[jsk_perception]   File "/usr/local/lib/python2.7/dist-packages/matplotlib/collections.py", line 27, in <module> 
[jsk_perception]     import matplotlib.collections as collections               
[jsk_perception]   File "/usr/local/lib/python2.7/dist-packages/matplotlib/collections.py", line 27, in <module> 
[jsk_perception]     import matplotlib.backend_bases as backend_bases           
[jsk_perception]   File "/usr/local/lib/python2.7/dist-packages/matplotlib/backend_bases.py", line 62, in <module> 
[jsk_perception]     import matplotlib.backend_bases as backend_bases           
[jsk_perception]   File "/usr/local/lib/python2.7/dist-packages/matplotlib/backend_bases.py", line 62, in <module> 
[jsk_perception]     import matplotlib.textpath as textpath                     
[... 10 lines cut ...]
[jsk_perception]   File "/usr/local/lib/python2.7/dist-packages/matplotlib/font_manager.py", line 1405, in _rebuild 
[jsk_perception]     fontManager = FontManager()                                
[jsk_perception]       File "/usr/local/lib/python2.7/dist-packages/matplotlib/font_manager.py", line 1043, in __init__ 
[jsk_perception] fontManager = FontManager()                                    
[jsk_perception]   File "/usr/local/lib/python2.7/dist-packages/matplotlib/font_manager.py", line 1043, in __init__ 
[jsk_perception]     self.ttffiles = findSystemFonts(paths) + findSystemFonts() 
[jsk_perception]       File "/usr/local/lib/python2.7/dist-packages/matplotlib/font_manager.py", line 323, in findSystemFonts 
[jsk_perception] self.ttffiles = findSystemFonts(paths) + findSystemFonts()     
[jsk_perception]   File "/usr/local/lib/python2.7/dist-packages/matplotlib/font_manager.py", line 334, in findSystemFonts 
[jsk_perception]     for f in get_fontconfig_fonts(fontext):                    
[jsk_perception]       File "/usr/local/lib/python2.7/dist-packages/matplotlib/font_manager.py", line 276, in get_fontconfig_fonts 
[jsk_perception] return [fname for fname in six.iterkeys(fontfiles) if os.path.exists(fname)] 
[jsk_perception]   File "/usr/lib/python2.7/genericpath.py", line 18, in exists 
[jsk_perception]         output = pipe.communicate()[0]                         
[jsk_perception] os.stat(path)                                                  
[jsk_perception]   File "/usr/lib/python2.7/subprocess.py", line 799, in communicate 
[jsk_perception] KeyboardInterrupt                                              
[... cut ...]              
[jsk_perception]   File "/usr/lib/python2.7/subprocess.py", line 1401, in _communicate 
[jsk_perception]     stdout, stderr = self._communicate_with_poll(input)        
[jsk_perception]   File "/usr/lib/python2.7/subprocess.py", line 1455, in _communicate_with_poll 
[jsk_perception]     ready = poller.poll()                                      
[jsk_perception] KeyboardInterrupt                                              
[jsk_pcl_ros]     [                        <=>       
```

k-okada, 2015-12-15:
maybe 2.0.9 fixed this problem

wkentaro, 2015-12-15:
That is because apt does not resolve dependency on pip.
by removing it from package.xml fixes this problem.

garaemon, 2015-12-16:
fixed

closed 2015-12-16 by garaemon