jsk-ros-pkg/jsk_common #1401: using voice_text with sound_play occurs high load?

furushchev, 2016-05-28:
I suspect that `sound_play` process with japanese voice (`voice_text`) make cpu load super high on PR2 real robot when deployed long time.

Once PR2 cpu load is over 50, which is too high to perform any task, and soon after `rosnode kill sound_node_jp`, cpu load is back to just 6.

I'm not yet sure what part are actually problem, but I just wanted to notify this.

k-okada, 2016-05-30:
do you mean the  cpu load is not so high just after starting pr2.launch and
becomes 50 after few hours?

furushchev, 2016-05-30:
after a lot of `say` ing, I think. I suspect that `sound_play` process has a lot of file descriptors left to be open.

k-okada, 2016-05-30:
https://github.com/ros-drivers/audio_common/blob/hydro-devel/sound_play/scripts/soundplay_node.py
uses threading so that's may be problem?

furushchev, 2016-06-20:
I think the main reason of high load is comming from keeping many file descriptors.
Opened issue at upstream.
https://github.com/ros-drivers/audio_common/issues/75

furushchev, 2016-07-01:
Sent Pull request to upstream.
Now I'm working on porting to hydro with minimum change.
(But some bugs cannot be fixed in minimum change)

furushchev, 2016-07-01:
done https://github.com/ros-drivers/audio_common/pull/78

furushchev, 2016-07-01:
created deb binary
[ros-hydro-sound-play_0.2.99-0precise_amd64.zip](https://github.com/jsk-ros-pkg/jsk_common/files/343037/ros-hydro-sound-play_0.2.99-0precise_amd64.zip)

closed 2016-07-01 by furushchev