jsk-ros-pkg/jsk_common #1411: Request of next release: version 2.0.15

wkentaro, 2016-06-13:
Could you please release next patch version? 2.0.15? @k-okada

k-okada, 2016-06-13:
```
Releasing package 'image_view2' for 'indigo' to: 'release/indigo/image_view2'
Releasing package 'jsk_common' for 'indigo' to: 'release/indigo/jsk_common'
Releasing package 'jsk_topic_tools' for 'indigo' to: 'release/indigo/jsk_topic_tools'
Releasing package 'jsk_tools' for 'indigo' to: 'release/indigo/jsk_tools'
Applying: comment out pip package
error: patch failed: package.xml:17
error: package.xml: patch does not apply
Patch failed at 0001 comment out pip package
When you have resolved this problem run "git am --resolved".
If you would prefer to skip this patch, instead run "git am --skip".
To restore the original branch and stop patching run "git am --abort".

 [git-bloom-patch import]: Failed to apply one or more patches for the 'Command 'git am /tmp/tmptSPw7I/*.patch' returned non-zero exit status 1' branch.

>>> Resolve any conflicts and when you have resolved this problem run 'git am --resolved' and then exit the shell using 'exit 0'. <<<
    To abort use 'exit 1'
 [git-bloom-patch import]: User reports that conflicts have been resolved, continuing.
 [git-bloom-patch import]: Applied 1 patches
Releasing package 'virtual_force_publisher' for 'indigo' to: 'release/indigo/virtual_force_publisher'
Key 'python-percol' resolved to '['percol']' with installer 'pip', which does not match the default installer 'apt'.
Key 'python-percol' resolved to '['percol']' with installer 'pip', which does not match the default installer 'apt'.
Key 'python-pygithub3' resolved to '['pygithub3']' with installer 'pip', which does not match the default installer 'apt'.
Key 'python-pygithub3' resolved to '['pygithub3']' with installer 'pip', which does not match the default installer 'apt'.
Key 'python-slacker-cli' resolved to '['slacker-cli']' with installer 'pip', which does not match the default installer 'apt'.
Key 'python-slacker-cli' resolved to '['slacker-cli']' with installer 'pip', which does not match the default installer 'apt'.
Key 'python-termcolor' resolved to '['termcolor']' with installer 'pip', which does not match the default installer 'apt'.
Could not resolve rosdep key 'rosemacs-el' for distro 'saucy':
Could not resolve rosdep key 'rosemacs-el' for distro 'trusty':
Some of the dependencies for packages in this repository could not be resolved by rosdep.
You can try to address the issues which appear above and try again if you wish.
Traceback (most recent call last):
  File "/usr/bin/git-bloom-generate", line 9, in <module>
    load_entry_point('bloom==0.5.21', 'console_scripts', 'git-bloom-generate')()
  File "/usr/lib/python2.7/dist-packages/bloom/commands/git/generate.py", line 252, in main
    run_generator(generator, args)
  File "/usr/lib/python2.7/dist-packages/bloom/commands/git/generate.py", line 127, in run_generator
    gen.pre_modify)
  File "/usr/lib/python2.7/dist-packages/bloom/commands/git/generate.py", line 101, in try_execute
    retcode = func(*args, **kwargs)
  File "/usr/lib/python2.7/dist-packages/bloom/generators/debian/generator.py", line 644, in pre_modify
    if not maybe_continue(msg="Would you like to try again?"):
  File "/usr/lib/python2.7/dist-packages/bloom/util.py", line 397, in maybe_continue
    response = safe_input(msg)
  File "/usr/lib/python2.7/dist-packages/bloom/util.py", line 97, in safe_input
    flush_stdin()
  File "/usr/lib/python2.7/dist-packages/bloom/util.py", line 90, in flush_stdin
    tcflush(sys.stdin, TCIFLUSH)
error: (25, 'Inappropriate ioctl for device')

Generating source debs for the packages: ['image_view2', 'dynamic_tf_publisher', 'nlopt', 'ff', 'multi_map_server', 'jsk_network_tools', 'jsk_topic_tools', 'rosping', 'speech_recognition_msgs', 'jsk_tools', 'rostwitter', 'virtual_force_publisher', 'jsk_common', 'mini_maxwell', 'jsk_data', 'jsk_tilt_laser', 'assimp_devel']
Debian Incremental Version: 0
Debian Distributions: ['saucy', 'trusty']
Releasing for rosdistro: indigo

Pre-verifying Debian dependency keys...
Running 'rosdep update'...
Failed to resolve python-percol on ubuntu:saucy with: Error running generator: The Debian generator does not support dependencies which are installed with the 'pip' installer.
python-percol is depended on by these packages: ['jsk_tools']
<== Failed
Failed to resolve python-percol on ubuntu:trusty with: Error running generator: The Debian generator does not support dependencies which are installed with the 'pip' installer.
python-percol is depended on by these packages: ['jsk_tools']
<== Failed
Failed to resolve python-pygithub3 on ubuntu:saucy with: Error running generator: The Debian generator does not support dependencies which are installed with the 'pip' installer.
python-pygithub3 is depended on by these packages: ['jsk_tools']
<== Failed
Failed to resolve python-pygithub3 on ubuntu:trusty with: Error running generator: The Debian generator does not support dependencies which are installed with the 'pip' installer.
python-pygithub3 is depended on by these packages: ['jsk_tools']
<== Failed
Failed to resolve python-slacker-cli on ubuntu:saucy with: Error running generator: The Debian generator does not support dependencies which are installed with the 'pip' installer.
[... 56 lines cut ...]
New ROS Distro index url: 'https://raw.githubusercontent.com/ros/rosdistro/959cc3f22c0b93ab6d97a850a8e88690f4939ae6/index.yaml'
==> Fetching 'jsk_common' repository from 'https://github.com/tork-a/jsk_common-release.git'
==> Setting release repository remote url to 'git@github.com:tork-a/jsk_common-release'
[... cut ...]
+ /var/lib/jenkins/send_to_bloom.py jsk-ros-pkg jsk_common_msgs
repository jsk-ros-pkg/jsk_common_msgs -> 2.0.1
release indigo/jsk_common_msgs -> 2.0.1
release jade/jsk_common_msgs -> 2.0.1
release kinetic/jsk_common_msgs -> 2.0.1
+ /var/lib/jenkins/send_to_bloom.py jsk-ros-pkg jsk_3rdparty
repository jsk-ros-pkg/jsk_3rdparty -> 2.0.14
release indigo/jsk_3rdparty -> 2.0.14
release jade/jsk_3rdparty -> 2.0.14
release kinetic/jsk_3rdparty -> None
+ /var/lib/jenkins/send_to_bloom.py jsk-ros-pkg jsk_control
```
...

k-okada, 2016-06-13:
https://github.com/ros/rosdistro/pull/11811 done

closed 2016-06-13 by k-okada