jsk-ros-pkg/jsk_common #1413: jsk_tools is not released well

wkentaro, 2016-06-15:
I have error in https://github.com/jsk-ros-pkg/jsk_recognition/pull/1747

https://github.com/tork-a/jsk_common-release/blob/debian/indigo/jsk_tools/package.xml#L18-L40

Below should be included in apt depends:

- python-colorama
- python-termcolor
- python-texttable

```
% pkgs=(python-percol python-colorama python-pygithub3 python-slacker-cli python-termcolor python-texttable)
for pkg in $pkgs; do
rosdep resolve $pkg;
done
#pip
percol
#apt
python-colorama
#pip
pygithub3
#pip
slacker-cli
#apt
python-termcolor
#apt
python-texttable
```

wkentaro, 2016-06-15:
Note that this is **critical** issue.

k-okada, 2016-06-17:
are you saying deb package of jsk_tools should have dependency to these python debs? may be I over-removed python depends, I'll fix this when we release next time. 
```
k-okada@kokada-t440s:~$ aptitude show ros-indigo-jsk-tools
Package: ros-indigo-jsk-tools            
State: installed
Automatically installed: no
Version: 2.0.14-0trusty-20160514-212155-0700
Priority: extra
Section: misc
Maintainer: Ryohei Ueda <ueda@jsk.t.u-tokyo.ac.jp>
Architecture: amd64
Uncompressed Size: 384 k
Depends: ros-indigo-axis-camera, ros-indigo-jsk-network-tools,
         ros-indigo-jsk-topic-tools, ros-indigo-pr2-computer-monitor,
         ros-indigo-rosbag, ros-indigo-rosemacs, ros-indigo-rosgraph-msgs,
         ros-indigo-rospy, ros-indigo-rqt-reconfigure
Conflicts: ros-indigo-jsk-tools
Description: Includes emacs scripts, ros tool alias generator, and launch doc generator.
 
Homepage: http://ros.org/wiki/jsk_tools

```

wkentaro, 2016-06-17:
> are you saying deb package of jsk_tools should have dependency to these python debs?

Yes, and it causes import error in travis testing.

k-okada, 2016-06-19:
python-termcolor is not available on saucy
```
k-okada@kokada-t440s:~$ rosdep resolve python-termcolor --os ubuntu:trusty
#apt
python-termcolor
k-okada@kokada-t440s:~$ rosdep resolve python-termcolor --os ubuntu:saucy
#pip
termcolor
```

wkentaro, 2016-06-19:
I didn't notice that, sorry.
I will remove dependency on termcolor and use colorama only.

wkentaro, 2016-06-19:
Sent #1417

PR #1417 (2016-06-22): Remove dependency on python-termcolor
Fix #1413
closed 2016-06-22 by YoheiKakiuchi