jsk-ros-pkg/jsk_common #1425: rosmaster間での通信方法

h-kamada, 2016-07-12:
複数台のロボットを用いて協調して動作を行う上で、rosmaster間の通信を可能にする必要があります。
どんなやり方が良いかを調べている所ですが、良いやり方を知っている人はいますか？

一つの可能性としてmultimaster_fkie(http://wiki.ros.org/multimaster_fkie)を使う方法があると思っています。
https://docs.google.com/presentation/d/1XM7x1SQDECO6A7phvw0mEaaMID0NNKweFHU4k2inUAk/edit#slide=id.p
で紹介されており、
http://www.iri.upc.edu/files/scidoc/1607-Multi-master-ROS-systems.pdf
でこれを使ったものが書かれています。

k-okada, 2016-07-12:
https://github.com/jsk-ros-pkg/jsk_robot/pull/502

closed 2016-07-15 by h-kamada

k-okada, 2024-01-09:
@tkmtnt7000 websoketを使った例について、以下のやり取りが pr2@ にありました、これはfirewall を超えられるので、topicにprefix つければmultimasater的につかえる、、、かもしれない。

https://gist.github.com/k-okada/8342fdcc684bb5db3e4794883516fb32

==
なるほど、ありがとうございます。
rosbridgeのwebsocker<->ROSのクライアント側ということですね、すごいですね
URLをプログラムを実行するときのargumentで渡せばjsk_robotとかjsk_3rdpartyでも追加できる気もしましたけどどうでしょうか？
急ぎで使いたいというわけではないですが、Github上のどこかにあったほうが検索もしやすいし、見やすい気もするので

--
北川 晋吾
Shingo Kitagawa
the University of Tokyo, JSK Lab
[s-kitagawa@jsk.imi.i.u-tokyo.ac.jp](mailto:s-kitagawa@jsk.imi.i.u-tokyo.ac.jp)

2020年9月5日(土) 13:57 Ishiguro Yasuhiro <[ishiguro@jsk.imi.i.u-tokyo.ac.jp](mailto:ishiguro@jsk.imi.i.u-tokyo.ac.jp)>:

これ自体は自宅PCのroscore環境からPR2などとtopic通信を可能にするクライアント側のプログラムです
すでにロボット体内で立ち上がっているrosbridge_serverに繋ぎに行きます(pr1040とfetch15で確認済み)
家から簡単なROS操作したい人向けに作られています

本格運用時の負荷分散用サーバー構成や管理レポジトリなどはこれから話し合っていくんだと思います

2020年9月5日(土) 4:55 Shingo Kitagawa <[s-kitagawa@jsk.imi.i.u-tokyo.ac.jp](mailto:s-kitagawa@jsk.imi.i.u-tokyo.ac.jp)>:
これを７３B2のどこかのマシンでたてますか？
あとPrivateにするのであれば、gitlabとかで管理したほうが良さそうな気がしますがどうでしょうか？

--
北川 晋吾
Shingo Kitagawa
the University of Tokyo, JSK Lab
[s-kitagawa@jsk.imi.i.u-tokyo.ac.jp](mailto:s-kitagawa@jsk.imi.i.u-tokyo.ac.jp)

2020年9月5日(土) 0:22 Ishiguro Yasuhiro <[ishiguro@jsk.imi.i.u-tokyo.ac.jp](mailto:ishiguro@jsk.imi.i.u-tokyo.ac.jp)>:
rosbridge_serverが上がっているロボット(PR2, fetch)共通

※注意点
- melodic以降前提
- WebSocketのポートは現状インターネットから誰でも繋げてしまうのでこのスクリプトをpublicにしない
- 大量のデータを扱うとロボット体内のrosbridge_serverのCPU負荷がそこそこ上がるので実験中に迷惑かけないように
- 帯域上限で止まる．特に大量のデータ送信は苦手．画像やPointCloudはcompressedやthrottleで
- クライアント側の全トピックをサーバー側に送ろうとするので，クリーンなROS_MASTER環境で起動すること

pip install roslibpy
roscore
websocket_to_client_ros.py
rviz とか rostopic pub とか

--
------------------------------------------------------

Naoki-Hiraoka, 2024-01-09:
自分は自前で共有メモリを使った単純なブリッジを書いて、rosmaster間でtopicをやりとりしていました。
https://github.com/Naoki-Hiraoka/connect_two_ros_master/tree/master/sample

knorth55, 2024-01-09:
we can use xmlrpc to connect two roscore masteras written in`app_manager`.
https://github.com/PR2/app_manager/blob/kinetic-devel/src/app_manager/master_sync.py