jsk-ros-pkg/jsk_common #1439: How to get download_data url

mmurooka, 2016-08-19:
@wkentaro , Cloud you tell me how to get the url of https://github.com/jsk-ros-pkg/jsk_common/blob/master/jsk_data/src/jsk_data/download_data.py#L71 ?

wkentaro, 2016-08-19:
Firstly you need to upload a file to public area like Google Drive and Dropbox.

There is a tool to help users to upload files to Google Drive: http://jsk-docs.readthedocs.io/en/latest/jsk_common/doc/jsk_data/scripts/jsk_data_cli.html
but you can upload data to Google Drive manually.

A sample is here: https://github.com/jsk-ros-pkg/jsk_recognition/blob/master/jsk_pcl_ros/scripts/install_sample_data.py#L12
Note that the link must be public to anyone.

wkentaro, 2016-08-23:
Fixed?

mmurooka, 2016-08-23:
Thank you. The problem was that I didn't make file public.

closed 2016-08-23 by mmurooka