jsk-ros-pkg/jsk_common #1460: failed to install slacker-cli

k-okada, 2016-10-20:
```
executing command [sudo -H pip install -U slacker-cli]
Traceback (most recent call last):
  File "/usr/bin/pip", line 11, in <module>
    sys.exit(main())
  File "/usr/lib/python2.7/dist-packages/pip/__init__.py", line 215, in main
    locale.setlocale(locale.LC_ALL, '')
  File "/usr/lib/python2.7/locale.py", line 581, in setlocale
    return _setlocale(category, locale)
locale.Error: unsupported locale setting
```

k-okada, 2016-10-20:
```
perl: warning: Setting locale failed.
perl: warning: Please check that your locale settings:
        LANGUAGE = (unset),
        LC_ALL = (unset),
        LC_TIME = "ja_JP.UTF-8",
        LC_MONETARY = "ja_JP.UTF-8",
        LC_ADDRESS = "ja_JP.UTF-8",
        LC_TELEPHONE = "ja_JP.UTF-8",
        LC_NAME = "ja_JP.UTF-8",
        LC_MEASUREMENT = "ja_JP.UTF-8",
        LC_IDENTIFICATION = "ja_JP.UTF-8",
        LC_NUMERIC = "ja_JP.UTF-8",
        LC_PAPER = "ja_JP.UTF-8",
        LANG = "en_US.UTF-8"
    are supported and installed on your system.
perl: warning: Falling back to a fallback locale ("en_US.UTF-8")
```

k-okada, 2016-10-20:
```executing command [sudo -H pip install -U pygithub3]
Traceback (most recent call last):
  File "/usr/bin/pip", line 11, in <module>
    sys.exit(main())
  File "/usr/lib/python2.7/dist-packages/pip/__init__.py", line 215, in main
    locale.setlocale(locale.LC_ALL, '')
  File "/usr/lib/python2.7/locale.py", line 581, in setlocale
    return _setlocale(category, locale)
locale.Error: unsupported locale setting
```
```
executing command [sudo -H pip install -U percol]
Traceback (most recent call last):
  File "/usr/bin/pip", line 11, in <module>
    sys.exit(main())
  File "/usr/lib/python2.7/dist-packages/pip/__init__.py", line 215, in main
    locale.setlocale(locale.LC_ALL, '')
  File "/usr/lib/python2.7/locale.py", line 581, in setlocale
    return _setlocale(category, locale)
locale.Error: unsupported locale setting
executing command [sudo -H pip install -U freezegun]
Traceback (most recent call last):
  File "/usr/bin/pip", line 11, in <module>
    sys.exit(main())
  File "/usr/lib/python2.7/dist-packages/pip/__init__.py", line 215, in main
    locale.setlocale(locale.LC_ALL, '')
  File "/usr/lib/python2.7/locale.py", line 581, in setlocale
    return _setlocale(category, locale)
locale.Error: unsupported locale setting
```

wkentaro, 2016-10-20:
```
% pip show setuptools | grep Version
Metadata-Version: 2.0
Version: 28.6.0

% pip show pip | grep Version
Metadata-Version: 1.1
Version: 8.1.2

% sudo -H pip install -U slacker-cli
[sudo] password for wkentaro:
Requirement already up-to-date: slacker-cli in /usr/local/lib/python2.7/dist-packages
Collecting slacker>=0.7.3 (from slacker-cli)
  Downloading slacker-0.9.28.tar.gz
Collecting requests (from slacker-cli)
  Downloading requests-2.11.1-py2.py3-none-any.whl (514kB)
    100% |████████████████████████████████| 522kB 1.3MB/s
Building wheels for collected packages: slacker
  Running setup.py bdist_wheel for slacker ... done
  Stored in directory: /root/.cache/pip/wheels/d1/9b/37/e62406417232828c72cbade126536070e526e167ae1ad9236a
Successfully built slacker
Installing collected packages: requests, slacker
  Found existing installation: requests 2.8.1
    Uninstalling requests-2.8.1:
      Successfully uninstalled requests-2.8.1
  Found existing installation: slacker 0.9.18
    Uninstalling slacker-0.9.18:
      Successfully uninstalled slacker-0.9.18
Successfully installed requests-2.11.1 slacker-0.9.28
```

k-okada, 2016-10-20:
```
user@ubuntu-16:~/catkin_ws$ pip show setuptools | grep Version
Traceback (most recent call last):
  File "/usr/bin/pip", line 11, in <module>
    sys.exit(main())
  File "/usr/lib/python2.7/dist-packages/pip/__init__.py", line 215, in main
    locale.setlocale(locale.LC_ALL, '')
  File "/usr/lib/python2.7/locale.py", line 581, in setlocale
    return _setlocale(category, locale)
locale.Error: unsupported locale setting
user@ubuntu-16:~/catkin_ws$ pip show pip | grep Version
Traceback (most recent call last):
  File "/usr/bin/pip", line 11, in <module>
    sys.exit(main())
  File "/usr/lib/python2.7/dist-packages/pip/__init__.py", line 215, in main
    locale.setlocale(locale.LC_ALL, '')
  File "/usr/lib/python2.7/locale.py", line 581, in setlocale
    return _setlocale(category, locale)
locale.Error: unsupported locale setting
```

wkentaro, 2016-10-21:
hahaha... what about below?

```bash
$ pip --help
$ pip --version
```

closed 2016-10-24 by k-okada

k-okada, 2016-10-24:
Solved this issue by installing japanese language pack

```
user@ubuntu-2gb-sfo2-01:~/catkin_ws$ sudo apt-get install language-pack-ja
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following additional packages will be installed:
  language-pack-ja-base
The following NEW packages will be installed:
  language-pack-ja language-pack-ja-base
0 upgraded, 2 newly installed, 0 to remove and 0 not upgraded.
Need to get 2018 kB of archives.
After this operation, 9407 kB of additional disk space will be used.
Do you want to continue? [Y/n] 
Get:1 http://sfo1.mirrors.digitalocean.com/ubuntu xenial-updates/main amd64 language-pack-ja-base all 1:16.04+20160627 [2016 kB]
Get:2 http://sfo1.mirrors.digitalocean.com/ubuntu xenial-updates/main amd64 language-pack-ja all 1:16.04+20160627 [1832 B]
Fetched 2018 kB in 0s (3361 kB/s)     
perl: warning: Setting locale failed.
perl: warning: Please check that your locale settings:
	LANGUAGE = (unset),
	LC_ALL = (unset),
	LC_TIME = "ja_JP.UTF-8",
	LC_MONETARY = "ja_JP.UTF-8",
	LC_ADDRESS = "ja_JP.UTF-8",
	LC_TELEPHONE = "ja_JP.UTF-8",
	LC_NAME = "ja_JP.UTF-8",
	LC_MEASUREMENT = "ja_JP.UTF-8",
	LC_IDENTIFICATION = "ja_JP.UTF-8",
	LC_NUMERIC = "ja_JP.UTF-8",
	LC_PAPER = "ja_JP.UTF-8",
	LANG = "en_US.UTF-8"
    are supported and installed on your system.
perl: warning: Falling back to a fallback locale ("en_US.UTF-8").
locale: Cannot set LC_ALL to default locale: No such file or directory
Selecting previously unselected package language-pack-ja-base.
(Reading database ... 64056 files and directories currently installed.)
Preparing to unpack .../language-pack-ja-base_1%3a16.04+20160627_all.deb ...
Unpacking language-pack-ja-base (1:16.04+20160627) ...
Selecting previously unselected package language-pack-ja.
Preparing to unpack .../language-pack-ja_1%3a16.04+20160627_all.deb ...
Unpacking language-pack-ja (1:16.04+20160627) ...
Setting up language-pack-ja (1:16.04+20160627) ...
Setting up language-pack-ja-base (1:16.04+20160627) ...
Generating locales (this might take a while)...
  ja_JP.UTF-8... done
Generation complete.
user@ubuntu-2gb-sfo2-01:~/catkin_ws$ pip show setuptools
---
Metadata-Version: 1.1
Name: setuptools
Version: 20.7.0
Summary: Easily download, build, install, upgrade, and uninstall Python packages
Home-page: https://github.com/pypa/setuptools
Author: Python Packaging Authority
Author-email: distutils-sig@python.org
License: UNKNOWN
```

k-okada, 2016-10-25:
need to install language pack
```
user@ubuntu-2gb-sfo2-01:~/catkin_ws$ sudo apt-get install language-pack-ja
Reading package lists... Done
Building dependency tree
Reading state information... Done
The following additional packages will be installed:
  language-pack-ja-base
The following NEW packages will be installed:
  language-pack-ja language-pack-ja-base
0 upgraded, 2 newly installed, 0 to remove and 0 not upgraded.
Need to get 2018 kB of archives.
After this operation, 9407 kB of additional disk space will be used.
Do you want to continue? [Y/n]
Get:1 http://sfo1.mirrors.digitalocean.com/ubuntu xenial-updates/main amd64
language-pack-ja-base all 1:16.04+20160627 [2016 kB]
Get:2 http://sfo1.mirrors.digitalocean.com/ubuntu xenial-updates/main amd64
language-pack-ja all 1:16.04+20160627 [1832 B]
Fetched 2018 kB in 0s (3361 kB/s)
perl: warning: Setting locale failed.
perl: warning: Please check that your locale settings:
LANGUAGE = (unset),
LC_ALL = (unset),
LC_TIME = "ja_JP.UTF-8",
LC_MONETARY = "ja_JP.UTF-8",
LC_ADDRESS = "ja_JP.UTF-8",
LC_TELEPHONE = "ja_JP.UTF-8",
LC_NAME = "ja_JP.UTF-8",
LC_MEASUREMENT = "ja_JP.UTF-8",
LC_IDENTIFICATION = "ja_JP.UTF-8",
LC_NUMERIC = "ja_JP.UTF-8",
LC_PAPER = "ja_JP.UTF-8",
LANG = "en_US.UTF-8"
    are supported and installed on your system.
perl: warning: Falling back to a fallback locale ("en_US.UTF-8").
locale: Cannot set LC_ALL to default locale: No such file or directory
Selecting previously unselected package language-pack-ja-base.
(Reading database ... 64056 files and directories currently installed.)
Preparing to unpack .../language-pack-ja-base_1%3a16.04+20160627_all.deb ...
Unpacking language-pack-ja-base (1:16.04+20160627) ...
Selecting previously unselected package language-pack-ja.
Preparing to unpack .../language-pack-ja_1%3a16.04+20160627_all.deb ...
Unpacking language-pack-ja (1:16.04+20160627) ...
Setting up language-pack-ja (1:16.04+20160627) ...
Setting up language-pack-ja-base (1:16.04+20160627) ...
Generating locales (this might take a while)...
  ja_JP.UTF-8... done
Generation complete.
user@ubuntu-2gb-sfo2-01:~/catkin_ws$ pip show setuptools
---
Metadata-Version: 1.1
Name: setuptools
Version: 20.7.0
Summary: Easily download, build, install, upgrade, and uninstall Python
packages
Home-page: https://github.com/pypa/setuptools
```