jsk-ros-pkg/jsk_common #1472: rossetip set ROS_IP and ROS_HOSTNAME to "src" at HRP2018v

mizmizo, 2016-10-27:
HRP2 18号機のvのindigo化の際に，
rosset2018(rossetip)すると，ROS_IPとROS_HOSTNAMEが"src"にセットされてしまう問題が起きたので報告します．
状況としては以下のような感じで，
Could not resolve ip from address. Subnet may be different
というメッセージが出た時だけ正しいROS_IPがセットされて，
通常の場合はsrcにセットされました．

```
hrpuser@hrp2018v:~$ rossethrp2018
set ROS_MASTER_URI to http://hrp2018v:10018
Could not resolve ip from address. Subnet may be different
set ROS_IP and ROS_HOSTNAME to 192.168.96.39
[http://hrp2018v:10018][192.168.96.39] hrpuser@hrp2018v:~$ rossethrp2018
set ROS_MASTER_URI to http://hrp2018v:10018
set ROS_IP and ROS_HOSTNAME to src
[http://hrp2018v:10018][src] hrpuser@hrp2018v:~$
```

この問題が起きたのは
89dc7e501ad8497d30f6f3f3a82916adaed60b16
で，
HRP2 17号機と同じ
cab9af11355775beaf15ecbd4e6923a7b5fa0f95
までバージョンを戻すと
rossetipが正しくできてroscoreが上げられるようになりました．

mizmizo, 2016-10-27:
手元のPCでも2台ほど試してみましたが同じ問題は再現しませんでした．

k-okada, 2016-10-27:
https://github.com/jsk-ros-pkg/jsk_common/compare/cab9af1...89dc7e5

みると、
https://github.com/jsk-ros-pkg/jsk_common/commit/3e95eadd8927d48fe8e30d9a4f58aa5e2598ea23
https://github.com/jsk-ros-pkg/jsk_common/commit/13e0c96b53d85405069564c0a58d736901cd736a
かな．ipというコマンドはあるかな．
apt-get install か rosdep install はしているかな．

furushchev, 2016-10-28:
@mizmizo 詳しく調べたいのでご協力をお願いします。
1. `which ip`としてでてきた表示を教えてください。
2. もし1.でコマンドの表示が出た場合さらに `LANG=C ip -o -4 a`の結果を教えてください。さらに`ifconfig`の結果も知りたいです。
3. 1でコマンドが出ない場合

```bash
cd ~/ros/indigo/
rosdep install --from-paths src --ignore-src -r -n -y
exec -l $SHELL
rossethrp2018
```

として結果が変わるか教えてください。
```

**NOTE** MACアドレスなどが表示されるので、その部分を隠してコメントしていただくか、メール返信のほうがいいかもしれません。

mizmizo, 2016-10-28:
コマンドを試してみました．
MACアドレスは適当に置き換えて有ります．

```
[http://hrp2018v:10018][192.168.96.39] hrpuser@hrp2018v:~$ which ip
/sbin/ip
[http://hrp2018v:10018][192.168.96.39] hrpuser@hrp2018v:~$ LANG=C ip -o -4 a
1: lo    inet 127.0.0.1/8 scope host lo\       valid_lft forever preferred_lft forever
2: eth0    inet 192.168.96.39/22 brd 192.168.99.255 scope global eth0\       valid_lft forever preferred_lft forever
[http://hrp2018v:10018][192.168.96.39] hrpuser@hrp2018v:~$ ifconfig
eth0      Link encap:Ethernet  HWaddr xx:xx:xx:xx:xx:xx  
          inet addr:192.168.96.39  Bcast:192.168.99.255  Mask:255.255.252.0
          inet6 addr: fe80::76d4:35ff:fec7:c63c/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:583816 errors:0 dropped:48730 overruns:0 frame:0
          TX packets:60439 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:276242754 (276.2 MB)  TX bytes:5352817 (5.3 MB)

lo        Link encap:Local Loopback  
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:320 errors:0 dropped:0 overruns:0 frame:0
          TX packets:320 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:0 
          RX bytes:40145 (40.1 KB)  TX bytes:40145 (40.1 KB)
```

furushchev, 2016-10-28:
@mizmizo なるほど。
追加で、
`getent hosts hrp2018v`
と
`ip -o -4 route get $(getent hosts hrp2018v | cut -f 1 -d ' ')`
の結果も教えてもらえるかな？

mizmizo, 2016-10-29:
@furushchev さん
以下のようになりました．

```
[http://hrp2018v:10018][192.168.96.39] hrpuser@hrp2018v:~$ getent hosts hrp2018v
127.0.1.1       hrp2018v
[http://hrp2018v:10018][192.168.96.39] hrpuser@hrp2018v:~$ ip -o -4 route get $(getent hosts hrp2018v | cut -f 1 -d ' ')
local 127.0.1.1 dev lo  src 127.0.0.1 \    cache <local> 

```

furushchev, 2016-10-29:
うーん、このマシンがネットワークにつながっていないようにみえるんだけど、合ってるかな？

mizmizo, 2016-10-29:
いえ，ネットワークにはつながっているはずです．

```
[http://hrp2018v:10018][192.168.96.39] hrpuser@hrp2018v:~$ ping 8.8.8.8
PING 8.8.8.8 (8.8.8.8) 56(84) bytes of data.
64 bytes from 8.8.8.8: icmp_seq=1 ttl=53 time=10.7 ms
^C
--- 8.8.8.8 ping statistics ---
1 packets transmitted, 1 received, 0% packet loss, time 0ms
rtt min/avg/max/mdev = 10.755/10.755/10.755/0.000 ms
[http://hrp2018v:10018][192.168.96.39] hrpuser@hrp2018v:~$ getent hosts hrp2018v
127.0.1.1       hrp2018v
[http://hrp2018v:10018][192.168.96.39] hrpuser@hrp2018v:~$ ip -o -4 route get $(getent hosts hrp2018v | cut -f 1 -d ' ')
local 127.0.1.1 dev lo  src 127.0.0.1 \    cache <local> 
```

kyawawa, 2016-11-01:
横から失礼します．
```
ip -o -4 route get
```
を実行した際，自身のIPアドレスを指定した場合先頭にlocalが付いてしまうように思えます(なので，体内でしか問題が起こらない)．

少し汚いですが，以下のように置き換えるとうまく行く気がします．
```
--- a/jsk_tools/env-hooks/99.jsk_tools.sh
+++ b/jsk_tools/env-hooks/99.jsk_tools.sh
@@ -96,7 +96,7 @@ rossetip_addr() {
         fi
         target_host=$target_host_ip
     fi
-    export ROS_IP=$(ip -o -4 route get $target_host | awk "/$target_host/ "'{print $5}')
+    export ROS_IP=$(ip -o -4 route get $target_host | awk "/$target_host/ "'{print substr($0, index($0, "src"))}' | awk '{print $2}')
     export ROS_HOSTNAME=$ROS_IP
 }
```

furushchev, 2016-11-01:
@kyawawa ありがとう。pull requestにしてくれるかな？

k-okada, 2016-11-01:
2016-11-01 19:31 GMT+09:00 Tatsuya Ishikawa <notifications@github.com>:

> ip -o -4 route get
>
>
で何が帰ってくるのかな．

--
◉ Kei Okada

kyawawa, 2016-11-01:
@k-okada 
IPアドレスが192.168.96.20のマシンの場合，
```
$ ip -o -4 route get 192.168.96.20
local 192.168.96.20 dev lo  src 192.168.96.20 \    cache <local> 
$ ip -o -4 route get 192.168.96.21
192.168.96.21 dev eth0  src 192.168.96.20 \    cache 
```
のような違いがあります

k-okada, 2016-11-01:
133も同じ結果だけど,古田くんが想定しているのは違うのかな．

k-okada@jenkins:~$ ip -o -4 route get 133.11.216.230
local 133.11.216.230 dev lo  src 133.11.216.230 \    cache <local>
k-okada@jenkins:~$ ip -o -4 route get 133.11.216.231
133.11.216.231 dev eth1  src 133.11.216.230 \    cache

kyawawa, 2016-11-22:
@k-okada おそらくですが，外側のPCからロボット体内にrossetipをする想定だったのではないかと思われます

furushchev, 2017-10-30:
@mizmizo 
https://github.com/furushchev/jsk_common/commit/804b2d6881094f40c97d5e3dc2c89c7e04a848fe fixed this issue or still occurs any error?

iory, 2018-07-23:
@furushchev 
This issue seems solved.
I added test code for this PR.
If you have a time, please check the test code.

furushchev, 2018-07-23:
のちにわかったのですが、`ip`コマンドの結果は問い合わせ先によって以下の３パターンになるようです。

- 自分自身に対して→localが追加される

```
$ ip -o -4 route get 192.168.96.20
local 192.168.96.20 dev lo src 192.168.96.20 \ cache 
```

- 同一セグメント内

```
$ ip -o -4 route get 192.168.96.21
192.168.96.21 dev eth0 src 192.168.96.20 \ cache
```

- それ以外→ゲートウェイ情報が追加される

```
$ ip -o -4 route get 8.8.8.8
8.8.8.8 via 133.11.216.254 dev wlp3s0  src 192.168.96.20 \    cache
```

現状のコードでは`src`の次のIPアドレスを取得しているので、いずれも問題ないはずで、
env_hookが動くかの最低限のチェックも
https://github.com/jsk-ros-pkg/jsk_common/pull/1590
でいい気がしますが、
このissueへの解決がなされたかをチェックするならこれもチェックするのかなと思います。