jsk-ros-pkg/jsk_common #1488: jsk_data and jsk_maps do not build on indigo armhf

tfoote, 2016-11-25:
http://build.ros.org/view/Ibin_arm_uThf/job/Ibin_arm_uThf__jsk_data__ubuntu_trusty_armhf__binary/228/console

```
03:45:35 Unknown option: buildsystem
03:45:35 dh_fixperms: warning: ignored unknown options in DH_OPTIONS
03:45:35 	find debian/ros-indigo-jsk-data  -print0 2>/dev/null | xargs -0r chown --no-dereference 0:0
03:45:35 	find debian/ros-indigo-jsk-data ! -type l  -print0 2>/dev/null | xargs -0r chmod go=rX,u+rw,a-s
03:45:35 	find debian/ros-indigo-jsk-data/usr/share/doc -type f  ! -regex 'debian/ros-indigo-jsk-data/usr/share/doc/[^/]*/examples/.*' -print0 2>/dev/null | xargs -0r chmod 644
03:45:35 	find debian/ros-indigo-jsk-data/usr/share/doc -type d  -print0 2>/dev/null | xargs -0r chmod 755
03:45:35 	find debian/ros-indigo-jsk-data/usr/share/man debian/ros-indigo-jsk-data/usr/man/ debian/ros-indigo-jsk-data/usr/X11*/man/ -type f  -print0 2>/dev/null | xargs -0r chmod 644
03:45:36 	find debian/ros-indigo-jsk-data -perm -5 -type f \( -name '*.so.*' -or -name '*.so' -or -name '*.la' -or -name '*.a' \)  -print0 2>/dev/null | xargs -0r chmod 644
03:45:36 	find debian/ros-indigo-jsk-data/usr/include -type f  -print0 2>/dev/null | xargs -0r chmod 644
03:45:36 	find debian/ros-indigo-jsk-data/usr/share/applications -type f  -print0 2>/dev/null | xargs -0r chmod 644
03:45:36 	find debian/ros-indigo-jsk-data -perm -5 -type f \( -name '*.cmxs' \)  -print0 2>/dev/null | xargs -0r chmod 644
03:45:36 	find debian/ros-indigo-jsk-data/usr/lib/perl5 debian/ros-indigo-jsk-data/usr/share/perl5 -type f -perm -5 -name '*.pm'  -print0 2>/dev/null | xargs -0r chmod a-X
03:45:36 	find debian/ros-indigo-jsk-data/usr/lib -type f -name '*.ali'  -print0 2>/dev/null | xargs -0r chmod uga-w
03:45:36    dh_strip
03:45:37 Unknown option: buildsystem
03:45:37 dh_strip: warning: ignored unknown options in DH_OPTIONS
03:45:39 	strip --remove-section=.comment --remove-section=.note debian/ros-indigo-jsk-data/opt/ros/indigo/share/jsk_data/scripts/drive-linux-x64
03:45:39 /usr/bin/strip:debian/ros-indigo-jsk-data/opt/ros/indigo/share/jsk_data/scripts/drive-linux-x64: File format not recognized
03:45:39 dh_strip: strip --remove-section=.comment --remove-section=.note debian/ros-indigo-jsk-data/opt/ros/indigo/share/jsk_data/scripts/drive-linux-x64 returned exit code 1
03:45:39 make: *** [binary] Error 29
03:45:39 dpkg-buildpackage: error: fakeroot debian/rules binary gave error exit status 2
03:45:39 E: Building failed
03:45:39 Traceback (most recent call last):
03:45:39   File "/tmp/ros_buildfarm/ros_buildfarm/binarydeb_job.py", line 133, in build_binarydeb
03:45:39     subprocess.check_call(cmd, cwd=source_dir)
03:45:39   File "/usr/lib/python3.4/subprocess.py", line 561, in check_call
03:45:39     raise CalledProcessError(retcode, cmd)
03:45:39 subprocess.CalledProcessError: Command '['apt-src', 'build', 'ros-indigo-jsk-data']' returned non-zero exit status 1
03:45:39 # END SUBSECTION
03:45:39 
03:45:39 --------------------------------------------------------------------------------------------------
03:45:39 `apt-src build ros-indigo-jsk-data` failed.
03:45:39 This is usually because of an error building the package.
03:45:39 The traceback from this failure (just above) is printed for completeness, but you can ignore it.
03:45:39 You should look above `E: Building failed` in the build log for the actual cause of the failure.
03:45:39 --------------------------------------------------------------------------------------------------
03:45:39 
```

http://build.ros.org/view/Ibin_arm_uThf/job/Ibin_arm_uThf__jsk_maps__ubuntu_trusty_armhf__binary/266/console

```
04:05:45 -- Using empy: /usr/bin/empy
04:05:47 -- Using CATKIN_ENABLE_TESTING: ON
04:05:47 -- Skip enable_testing() when building binary package
04:05:47 -- Using CATKIN_TEST_RESULTS_DIR: /tmp/binarydeb/ros-indigo-jsk-maps-0.0.3/obj-arm-linux-gnueabihf/test_results
04:05:47 -- Looking for include file pthread.h
04:05:48 -- Looking for include file pthread.h - found
04:05:48 -- Looking for pthread_create
04:05:49 -- Looking for pthread_create - not found
04:05:49 -- Looking for pthread_create in pthreads
04:05:50 -- Looking for pthread_create in pthreads - not found
04:05:50 -- Looking for pthread_create in pthread
04:05:52 -- Looking for pthread_create in pthread - found
04:05:52 -- Found Threads: TRUE  
04:05:52 -- Found gtest sources under '/usr/src/gtest': gtests will be built
04:05:52 -- Using Python nosetests: /usr/bin/nosetests-2.7
04:05:52 -- catkin 0.6.18
04:05:54 -- set EUSDIR to /opt/ros/indigo/share/euslisp/jskeus/eus
04:05:54 -- set ARCHDIR to LinuxARM
04:05:54 -- Set GCC_MACHINE to arm-linux-gnueabihf
04:05:54 CMake Error at CMakeLists.txt:36 (message):
04:05:54   cannot find irteusgl
04:05:54 
04:05:54 
04:05:54 -- Configuring incomplete, errors occurred!
04:05:54 See also "/tmp/binarydeb/ros-indigo-jsk-maps-0.0.3/obj-arm-linux-gnueabihf/CMakeFiles/CMakeOutput.log".
04:05:54 See also "/tmp/binarydeb/ros-indigo-jsk-maps-0.0.3/obj-arm-linux-gnueabihf/CMakeFiles/CMakeError.log".
04:05:54 dh_auto_configure: cmake .. -DCMAKE_INSTALL_PREFIX=/usr -DCMAKE_VERBOSE_MAKEFILE=ON -DCMAKE_BUILD_TYPE=None -DCATKIN_BUILD_BINARY_PACKAGE=1 -DCMAKE_INSTALL_PREFIX=/opt/ros/indigo -DCMAKE_PREFIX_PATH=/opt/ros/indigo returned exit code 1
04:05:54 make[1]: *** [override_dh_auto_configure] Error 2
04:05:54 make[1]: Leaving directory `/tmp/binarydeb/ros-indigo-jsk-maps-0.0.3'
04:05:54 make: *** [build] Error 2
04:05:54 dpkg-buildpackage: error: debian/rules build gave error exit status 2
04:05:54 E: Building failed
04:05:54 Traceback (most recent call last):
04:05:54   File "/tmp/ros_buildfarm/ros_buildfarm/binarydeb_job.py", line 133, in build_binarydeb
04:05:54     subprocess.check_call(cmd, cwd=source_dir)
04:05:54   File "/usr/lib/python3.4/subprocess.py", line 561, in check_call
04:05:54     raise CalledProcessError(retcode, cmd)
04:05:54 subprocess.CalledProcessError: Command '['apt-src', 'build', 'ros-indigo-jsk-maps']' returned non-zero exit status 1
04:05:54 # END SUBSECTION
04:05:54 
04:05:54 --------------------------------------------------------------------------------------------------
04:05:54 `apt-src build ros-indigo-jsk-maps` failed.
04:05:54 This is usually because of an error building the package.
04:05:54 The traceback from this failure (just above) is printed for completeness, but you can ignore it.
04:05:54 You should look above `E: Building failed` in the build log for the actual cause of the failure.
04:05:54 --------------------------------------------------------------------------------------------------
04:05:54 
```

wkentaro, 2016-11-25:
@k-okada Should I change something?

k-okada, 2016-11-26:
@wkentaro , yes `scripts/drive-linux-x64` is the cause of trouble
```
strip --remove-section=.comment --remove-section=.note debian/ros-indigo-jsk-data/opt/ros/indigo/share/jsk_data/scripts/drive-linux-x64
/usr/bin/strip:debian/ros-indigo-jsk-data/opt/ros/indigo/share/jsk_data/scripts/drive-linux-x64: File format not recognized
```
Possible solution is
1) do not install drive-linux-x64 only when x84 architecture, maybe something like https://github.com/jsk-ros-pkg/jsk_demos/pull/1208/files#diff-8d261c604db6cc51d62b87871e82af2dR15
2) compile executable from source.

wkentaro, 2016-11-26:
@k-okada Thanks. I sent PR: https://github.com/jsk-ros-pkg/jsk_common/pull/1489