jsk-ros-pkg/jsk_common #1494: terminal is hidden when executing rossetip

yuki-asano, 2016-12-05:
rossetipをすると、(rossetmasterだったかも)
添付のようにターミナルが隠れてしまう状況になったので報告いたします。
その時の、envの結果も添付いたします。

[image]

[env.txt](https://github.com/jsk-ros-pkg/jsk_common/files/631161/env.txt)

furushchev, 2016-12-05:
@yuki-asano どれが隠れてしまった画面でしょうか？

`env.txt`の`PS1=\[\033[00;31m\][http://192.168.1.111:11311][192.168.1.120]\[\033[00m\] `という記述を見ると、表示は画像の左ターミナルのようになるのが正しいように感じます。

またターミナルが隠れるとはどういうことでしょうか？

yuki-asano, 2016-12-05:
隠れているといっているのは、左の画面のことです。
本来、mech-user@みたいにカレントディレクトリが表示されると思いますが、
それが[...][....]で隠れてしまっているという意味でした。

furushchev, 2016-12-05:
なるほど。ちなみにhistoryも保存されていたりしませんか？

furushchev, 2016-12-05:
ソースコードと最近のメールの流れを読んだ上での完全な推測ですが、 `~/.bashrc`に、

```bash
source /opt/ros/kinetic/setup.bash
export ROS_MASTER_URI=http://192.168.1.111:11311
```

と書いて、

```bash
$ rossetip
```

すると再現する気がしました。

yuki-asano, 2016-12-05:
historyは保存していないですね。

> ソースコードと最近のメールの流れを読んだ上での完全な推測ですが、 ~/.bashrcに、
> 
> source /opt/ros/kinetic/setup.bash
> export ROS_MASTER_URI=http://192.168.1.111:11311
> と書いて、
> 
> $ rossetip
> すると再現する気がしました。

再現しないですね。
これ演習中によく起きていた問題で、
自分のPCだと再現しないんですよね。。

wkentaro, 2016-12-05:
@h-kamada これと似た感じに前なっていませんでした？

furushchev, 2016-12-05:
@yuki-asano 

> 再現しないですね。

うーん、僕の手元で、apt (ただしshadow-fixed)から入れると再現するのですが。。これもおかしいですね。

`type rossetmaster`
と
`type rossetip`
と再現しない @yuki-asano のPCでうって、出力を確かめてくれますか？

wkentaro, 2016-12-05:
> Cc: @wkentaro (53c58ac) 直せそうかな?

ん、僕のせいですか？

wkentaro, 2016-12-05:
indigoでshadow-fixed aptだとこうなります。
```
% bash 
wkentaro@hoop:~/Projects/objdepth/src/pascal3d-toolkit/src$ source /opt/ros/indigo/setup.bash
wkentaro@hoop:~/Projects/objdepth/src/pascal3d-toolkit/src$ rossetip
set ROS_IP and ROS_HOSTNAME to 133.11.216.106
wkentaro@hoop:~/Projects/objdepth/src/pascal3d-toolkit/src$ rossetmaster 192.168.1.11
set ROS_MASTER_URI to http://192.168.1.11:11311
[http://192.168.1.11:11311][133.11.216.106] wkentaro@hoop:~/Projects/objdepth/src/pascal3d-toolkit/src$ rossetlocal
set ROS_MASTER_URI to http://localhost:11311
wkentaro@hoop:~/Projects/objdepth/src/pascal3d-toolkit/src$ source /opt/ros/indigo/setup.bash
wkentaro@hoop:~/Projects/objdepth/src/pascal3d-toolkit/src$ export ROS_MASTER_URI=http://192.168.1.111:11311
wkentaro@hoop:~/Projects/objdepth/src/pascal3d-toolkit/src$ rossetip
set ROS_IP and ROS_HOSTNAME to eth1
[http://192.168.1.111:11311][eth1] wkentaro@hoop:~/Projects/objdepth/src/pascal3d-toolkit/src$
```

furushchev, 2016-12-05:
https://github.com/jsk-ros-pkg/jsk_common/commit/53c58ac9b66401df385851f78bd40c9c60989fce#diff-48005ba849f01087afb5a38e0561c822R45

で`ROS_MASTER_URI`をrossetmasterでセットすることを想定している気がするけど、自分で`export`して設定しちゃってから`_update_XX`が呼ばれると、

https://github.com/jsk-ros-pkg/jsk_common/commit/53c58ac9b66401df385851f78bd40c9c60989fce#diff-48005ba849f01087afb5a38e0561c822R30

で空文字列が`PS1`に設定されちゃう気がする。

https://github.com/jsk-ros-pkg/jsk_common/commit/53c58ac9b66401df385851f78bd40c9c60989fce#diff-48005ba849f01087afb5a38e0561c822R30
のコミットは名前が`Merge`になっているけど、Mergeだけじゃなくて、関数の中身が変わっていて、Authorが @wkentaro になっているので、なにかこだわりがあるかと思ったのです。
間にコミットがいくつかあったならごめん。

wkentaro, 2016-12-05:
> 53c58ac#diff-48005ba849f01087afb5a38e0561c822R45

でROS_MASTER_URIをrossetmasterでセットすることを想定している気がするけど、自分でexportして設定しちゃってから_update_XXが呼ばれると、

なるほど。となると僕のせいですね.. 直しておきます。

furushchev, 2016-12-05:
僕の方だと

```
furushchev@ThinkPad-P50:~/ros/indigo/build/darknet/darknet_ext-prefix/src/darknet_ext-stamp$ rossetip
set ROS_IP and ROS_HOSTNAME to 134.102.118.226
furushchev@ThinkPad-P50:~/ros/indigo/build/darknet/darknet_ext-prefix/src/darknet_ext-stamp$ export ROS_MASTER_URI=http://192.168.1.111:11311
furushchev@ThinkPad-P50:~/ros/indigo/build/darknet/darknet_ext-prefix/src/darknet_ext-stamp$ rossetip
set ROS_IP and ROS_HOSTNAME to eth0
[http://192.168.1.111:11311][eth0] 
```

になります。実は最初に`rossetip`をするのもポイントだったかもしれない。

yuki-asano, 2016-12-05:
ちょい遅れましたが、結果は添付のようになりました。
[image]

> になります。実は最初にrossetipをするのもポイントだったかもしれない。

確かに、演習中
rossetmaster
rossetip
の順番だとダメだけど、
rossetip
rossetmaster
だとうまくいくという状況がありました。

他にも、
rossetipを二回実行するとバグるのもこれだったりする？

furushchev, 2016-12-05:
> rossetipを二回実行するとバグるのもこれだったりする？

それは僕の入れた別のバグだと思います（大変失礼しました。）がどうでしょうか？ #1472 
を見てみて、その症状と同じなら、 https://github.com/jsk-ros-pkg/jsk_common/pull/1487 で修正済みですが、まだaptの方には行き渡っていないという状況です。

wkentaro, 2016-12-05:
このissueは https://github.com/jsk-ros-pkg/jsk_common/pull/1495 で直ると思います。

yuki-asano, 2016-12-06:
@furushchev @wkentaro 
早速ありがとう！
対応早いですね．

closed 2016-12-06 by k-okada

robograffitti, 2022-11-04:
.bashrc に rossetmaster; rossetip で書いていたら、このような症状が出ました。

試しに.bashrcからコメントアウトして、起動後にrossetmaster; rossetip にするとこのようにならなかったようなので、ログインシェルでrossetmaster; rossetipの順番で実行されると、この症状が出ました。

一旦、rossetip; rossetmasterの順番で実行するようにして回避してるので特に問題になってませんが、一応、報告しておきます。

[image]

PR #1495 (2016-12-06): Set WITHOUT_ROS_PROMPT at _update_prompt to fix #1494
Fix #1494 

[image]
commit b824a808 (2016-12-06): Merge pull request #1495 from wkentaro/fix-without-ros-prompt

Set WITHOUT_ROS_PROMPT at _update_prompt to fix #1494
commit ba6a06d6 (2016-12-06): Set WITHOUT_ROS_PROMPT at _update_prompt to fix #1494