jsk-ros-pkg/jsk_common #1496: jsk_network_tools cant directly support sensor_msgs/Image type

cretaceous-creature, 2016-12-08:
@furushchev 
If I change the param of '~message' directly to sensor_msgs/Image
it wont work due to the string parse failure.
```
Traceback (most recent call last):
  File "/home/ubuntu/ros/indigo/src/jsk_common/jsk_network_tools/scripts/silverhammer_lowspeed_streamer.py", line 118, in <module>
    st = SilverHammerLowspeedStreamer()
  File "/home/ubuntu/ros/indigo/src/jsk_common/jsk_network_tools/scripts/silverhammer_lowspeed_streamer.py", line 47, in __init__
    self.send_format = msgToStructFormat(self.send_message())
  File "/home/ubuntu/ros/indigo/src/jsk_common/jsk_network_tools/src/jsk_network_tools/silverhammer_util.py", line 53, in msgToStructFormat
    raise Excception("string is not supported!, please use char[static_length]")
NameError: global name 'Excception' is not defined
```
I also tested 
sensor_msgs/Joy (fail)
sensor_msgs/Imu (success)
std_msgs/* (seem to be OK)

my temporary solution is that I create my own msg type under jsk_network_tools package,
in that way it will work for both type of sensor_msgs/Image and sensor_msgs/Joy...

mmurooka, 2016-12-08:
Following is an example of silverhammer use case.
We also generate the original message for lowspeed_streamer.

launch
https://github.com/jsk-ros-pkg/jsk_demos/blob/master/jsk_2015_06_hrp_drc/drc_com_common/launch/field_computer_com.launch#L50-L58

msg
https://github.com/jsk-ros-pkg/jsk_demos/blob/master/jsk_2015_06_hrp_drc/drc_com_common/msg/FC2OCSSmall.msg

> my temporary solution is that I create my own msg type under jsk_network_tools package,
in that way it will work for both type of sensor_msgs/Image and sensor_msgs/Joy...

If the message is general, it should be put under jsk_network_tools.
In DRC, we put message under drc package because it is specific to drc.

In highspeed, we can use any topic.

cretaceous-creature, 2016-12-08:
@mmurooka Thx, 
The reason that I dont want to use the generated msg is that I dont want to 
copy the data and make a new msg due to the weakness CPU of nvidia TK1 and TX1.
But it seems I have to resize the image anyway, so the generated msg is OK for me...

As you said, it seems lowspeed did not support all msgs types..

mmurooka, 2016-12-08:
I see.
Is communication narrow? How about using highspeed?

cretaceous-creature, 2016-12-08:
@mmurooka Thx,
I am using highspeed and it is working now...
Sending an uncompressed mono color image at 10HZ 
seems OK, but it still take 3.2MB network.
I may need to resize the image.

furushchev, 2016-12-08:
Sorry for late response. This package is not for reducing data size but for
communicating in unstable network.
For data size reduction, you can use throttle to drop frame rate, compress
data (please refer image_transport package. But IMO, compressed image is
not so much useful, because we cannot use an image if any part of data is
broken. silverhummer has a feature to try to communicate even if a part of
data is missing, but we cannot take advantage of it if you use compressed
image. ), or
https://github.com/jsk-ros-pkg/jsk_recognition/tree/master/resized_image_transport,
this package resizes image data taking care of other parameters.