jsk-ros-pkg/jsk_common #1516: remove colorma  from ros_console.py

k-okada, 2017-04-20:
https://github.com/jsk-ros-pkg/jsk_common/blob/master/jsk_tools/bin/ros_console.py is very useful but requries `pip colorama`, so we can not use this `out of box`....

wkentaro, 2017-04-21:
Seems rosdep works.
```bash
% rosdep install --from-path . -r -s --reinstall
#[apt] Installation commands:
  sudo -H apt-get install ros-indigo-rosemacs
  sudo -H apt-get install ros-indigo-catkin
  sudo -H apt-get install ros-indigo-rostest
#[pip] Installation commands:
  sudo -H pip install -U tabulate
  sudo -H pip install -U slacker-cli
#[apt] Installation commands:
  sudo -H apt-get install git-core
  sudo -H apt-get install ros-indigo-jsk-network-tools
  sudo -H apt-get install ros-indigo-rqt-reconfigure
  sudo -H apt-get install python-colorama
  sudo -H apt-get install ros-indigo-pr2-computer-monitor
  sudo -H apt-get install ros-indigo-jsk-topic-tools
  sudo -H apt-get install python-texttable
  sudo -H apt-get install ros-indigo-axis-camera
  sudo -H apt-get install ros-indigo-jsk-gui-msgs
  sudo -H apt-get install ros-indigo-roslint
  sudo -H apt-get install ros-indigo-rospy
#[pip] Installation commands:
  sudo -H pip install -U pygithub3
#[apt] Installation commands:
  sudo -H apt-get install ros-indigo-cv-bridge
  sudo -H apt-get install ros-indigo-rosgraph-msgs
  sudo -H apt-get install python-requests
#[pip] Installation commands:
  sudo -H pip install -U percol
#[apt] Installation commands:
  sudo -H apt-get install iproute2
  sudo -H apt-get install ros-indigo-rosbag
```

k-okada, 2017-04-23:
sorry we're testing on hydro, 
https://github.com/tork-a/jsk_common-release/blob/debian/hydro/precise/jsk_tools/package.xml#L22

PR #1517 (2017-04-23): Remove import error message in ros_console.py
Because python-colorama is installed via apt

Close #1516

@k-okada maybe you can uncomment the run_depend of python-colorama when releasing
closed 2017-04-23 by k-okada