jsk-ros-pkg/jsk_common #1562: travis always fails on push brach

k-okada, 2017-10-26:
see https://github.com/jsk-ros-pkg/jsk_common/pull/1550#issuecomment-324017605, 

```
import pkg_resources._vendor.packaging._compat # precompiled from
/usr/local/lib/python2.7/dist-packages/pkg_resources/_vendor/packaging/_compat.pyc
#
/usr/local/lib/python2.7/dist-packages/pkg_resources/_vendor/packaging/requirements.pyc
matches
/usr/local/lib/python2.7/dist-packages/pkg_resources/_vendor/packaging/requirements.py
import pkg_resources._vendor.packaging.requirements # precompiled from
/usr/local/lib/python2.7/dist-packages/pkg_resources/_vendor/packaging/requirements.pyc
#
/usr/local/lib/python2.7/dist-packages/pkg_resources/_vendor/pyparsing.pyc
matches
/usr/local/lib/python2.7/dist-packages/pkg_resources/_vendor/pyparsing.py
import pkg_resources._vendor.pyparsing # precompiled from
/usr/local/lib/python2.7/dist-packages/pkg_resources/_vendor/pyparsing.pyc
# /usr/lib/python2.7/weakref.pyc matches /usr/lib/python2.7/weakref.py
import weakref # precompiled from /usr/lib/python2.7/weakref.pyc
# /usr/lib/python2.7/copy.pyc matches /usr/lib/python2.7/copy.py
import copy # precompiled from /usr/lib/python2.7/copy.pyc
# /usr/lib/python2.7/pprint.pyc matches /usr/lib/python2.7/pprint.py
import pprint # precompiled from /usr/lib/python2.7/pprint.pyc
# /usr/lib/python2.7/threading.pyc matches /usr/lib/python2.7/threading.py
import threading # precompiled from /usr/lib/python2.7/threading.pyc
#
/usr/local/lib/python2.7/dist-packages/pkg_resources/_vendor/packaging/markers.pyc
matches
/usr/local/lib/python2.7/dist-packages/pkg_resources/_vendor/packaging/markers.py
import pkg_resources._vendor.packaging.markers # precompiled from
/usr/local/lib/python2.7/dist-packages/pkg_resources/_vendor/packaging/markers.pyc
No output has been received in the last 10m0s, this potentially indicates a
stalled build or something wrong with the build itself.
Check the details on how to adjust your build configuration on:
https://docs.travis-ci.com/user/common-build-problems/#Build-times-out-because-no-output-was-received

```

k-okada, 2017-11-02:
あれ？　https://github.com/jsk-ros-pkg/jsk_common/pull/1561#issuecomment-339985372　なおっている？　@furushchev

furushchev, 2017-11-02:
推測ですが、 #1561 のPRを出す前に #1560 で間違ってforkじゃなくてjsk-ros-pkgのブランチからPRを送ってしまいました。
（このときのpushテストは失敗しました。）
その後#1561で同じ修正をPRし直したのですが、コミットのhashは同じだったので、テストされ直すことないままXがついたのかなと思っています。
（#1561を出した直後からテストにXがついてました。）
で、その後にこのPRを更新したので、今度はpushもPRのテストもうまく言ったのかなと思っています。

closed 2017-11-06 by k-okada