jsk-ros-pkg/jsk_common #1639: rosbagを再生してもcamera_linkへのtfが出ない

itohdak, 2019-11-08:
最近，[hrp2_record.launch](https://github.com/jsk-ros-pkg/jsk_common/blob/master/jsk_data/launch/hrp2_record.launch) で取ったrosbagをlocalで再生してもpointcloudがrvizに表示されない問題が起こっていました．
BODY（またはground）からcamera_rgb_optical_frameなどへのtfが参照できないのがその原因で，
```
$ rosrun tf view_frames
```
で確認すると確かにHEAD_LINK1からcamera_linkへのtfが出ていませんでした． 

@mmurooka @pazeshun に協力してもらい調べた結果，
先日ロボット体内のPCをindigoからkineticにupdateしたタイミングで，
staticなtfは```/tf_static```として，最初の1回しかpublishされなくなっており，
結果rosbagにも保存されなくなっていたということがわかりました．

現状の解決策としては，
ロボットを起動するときに立ち上げるlaunchファイル内でpublishしている該当する[static tf](https://github.com/start-jsk/rtmros_hrp2/blob/master/jsk_hrp2_ros_bridge/launch/hrp2017_local.launch#L24-L27)を，
localでrosbagを再生するときにも自前でpublishすることで対処しています．

他に良い解決策はあるでしょうか．

iory, 2019-11-08:
tf_staticをrecordする必要があると思います

iory, 2019-11-08:
@itohdak  
https://github.com/jsk-ros-pkg/jsk_common/pull/1641
こちらのPRを試してもらってもいいでしょうか？

itohdak, 2019-11-08:
@iory 
camera_linkへのtfも見られて，pointcloudも見ることができました！

itohdak, 2019-11-08:
ところで，先ほどstatic_transform_publisherを追加すると解決したと述べましたが，
若干間違いがあることが分かって，static_transform_publisherのノードでpublishしたものは```/tf_static```ではなく```/tf```に出るらしいです．

それなら，なぜ元々rosbagでrecordされなかったのでしょうか．．

iory, 2019-11-08:
多分見えていなかったtfは`static_trasnform_publisher`で出しているものではなくて、
`robot_state_publisher`で出ているtf_staticのリンクかと思います。
実は
https://github.com/ros/robot_state_publisher/blob/melodic-devel/src/joint_state_listener.cpp#L60
を見るとdefaultでtf_staticで出されるようになっているため
tfをrecordしても出てこないということになっていると思います。

itohdak, 2019-11-08:
はい，それは承知しているつもりで，[hrp2017_local.launch内](https://github.com/start-jsk/rtmros_hrp2/blob/master/jsk_hrp2_ros_bridge/launch/hrp2017_local.launch#L24-L27)で呼ばれている```static_transform_publisher```のtfが見えないのは不思議だなと思っていましたが，
よく考えたらこれも自分の勘違いで，このlaunchファイルは実機では立ち上がっていませんでした．
お騒がせして申し訳ありません．

mmurooka, 2019-11-11:
> このlaunchファイルは実機では立ち上がっていませんでした．

そうでした，僕が間違った情報を言ってしまった．

整理すると，
1. dashboardは以下のlaunchで上がって，
https://github.com/start-jsk/rtmros_hrp2/blob/master/jsk_hrp2_ros_bridge/launch/hrp2_dashboard.launch#L3-L6

2. ここで以下のrosparamがロードされる．
https://github.com/start-jsk/rtmros_hrp2/blob/master/jsk_hrp2_ros_bridge/config/hrpsys_dashboard.yaml#L7
`hrpsys_ros_bridge_start_command`というrosparamで，以下をセットしていて，
https://github.com/start-jsk/rtmros_hrp2/blob/master/jsk_hrp2_ros_bridge/scripts/hrp2-launch-ros-bridge.sh
このshスクリプトの中からは，例えば`HRP2NO`が17の場合は`hrp2017_ros_bridge.launch`が呼ばれる．

3. dashboardでstart_ros_bridgeのボタンが押されると，2.でセットされている`hrpsys_ros_bridge_start_command`のrosparamのコマンドが実行される，つまり`hrp2017_ros_bridge.launch`が呼ばれる．
https://github.com/start-jsk/rtmros_hrp2/blob/master/jsk_hrp2_ros_bridge/src/jsk_hrp2_ros_bridge/jsk_hrp2_dashboard.py#L148-L149
https://github.com/start-jsk/rtmros_common/blob/master/hrpsys_ros_bridge/src/hrpsys_ros_bridge/hrpsys_dashboard.py#L374-L377
を参照．

なので，dashboardでstart_ros_bridgeのボタンを押すと，`hrp2017_ros_bridge.launch`が呼ばれる．

さらに，robot_descriptionとして，どのモデルが使われるかは，
1. https://github.com/start-jsk/rtmros_hrp2/blob/master/jsk_hrp2_ros_bridge/launch/hrp2017_ros_bridge.launch
2. https://github.com/start-jsk/rtmros_hrp2/blob/master/jsk_hrp2_ros_bridge/launch/jsk_hrp2_ros_bridge.launch
3. https://github.com/start-jsk/rtmros_hrp2/blob/master/jsk_hrp2_ros_bridge/launch/hrp2_set_robot_description.launch

の順にlaunchがincludeされている．

4. https://github.com/start-jsk/rtmros_hrp2/blob/master/jsk_hrp2_ros_bridge/launch/hrp2017_ros_bridge.launch#L13-L17
のrosparamから，3.の`hrp2_set_robot_description`では，
https://github.com/start-jsk/rtmros_hrp2/blob/master/jsk_hrp2_ros_bridge/launch/hrp2_set_robot_description.launch#L14-L15
の行が呼ばれる．

結局，https://github.com/start-jsk/rtmros_hrp2/blob/master/jsk_hrp2_ros_bridge/launch/hrp2017_ros_bridge.launch#L17 のコマンドの結果がrobot_descriptionとしてセットされる．

itohdak, 2019-11-11:
@mmurooka 
詳しくフォローしていただき，ありがとうございます！