jsk-ros-pkg/jsk_common #1666: pose_to_tf.py 的なものが欲しい

ishiguroJSK, 2020-12-02:
tf_to_poseはありますが，PoseStamped (frame_id="camera_optical_frame")の(frame_id="world")に対するposeを知りたい時に，一旦TF解決する必要があるわけですが，
PoseStampedをTFに変換するツールって無くないですか？

なければ作ろうと思いますが
https://answers.ros.org/question/215656/how-to-transform-a-pose/

k-okada, 2020-12-02:
https://github.com/search?q=pose_to_tf.py&type=code
で探すと@mmurooka の
https://github.com/mmurooka/euslib/tree/6080a3e66afc5785111f14561a5d1b290e0f3d0a/demo/murooka/wholebody_manipulation_planner/demo_robot_carry_object/script
が出てきますね．これを入れておきましょう．

ちなみに，どのノードがPoseを出しているのでしょうか．だいたいはそのノードでTFも出すような変更する方向に落ち着く気もします．https://github.com/jsk-ros-pkg/jsk_recognition/search?q=publish_tf

ishiguroJSK, 2020-12-02:
具体的にはPointcloud_screenpointで，実際には対象はPointです(最悪これはtopic_toolsでpoint->poseにできる)．
https://github.com/jsk-ros-pkg/jsk_recognition/blob/2c1400e15d79b4f5f9fc7e6784a58bb277ebd8e3/jsk_pcl_ros/src/pointcloud_screenpoint_nodelet.cpp#L305

ただ一方で，完全にTFに移行すると，クリックしたタイミングに応じたコールバック処理がやりずらくなるので，
確かにPointとTFを両方pubするというのが良いのかもしれません．
https://answers.ros.org/question/63748/subscribe-to-tf-eg-tf-callback/

k-okada, 2020-12-03:
https://github.com/jsk-ros-pkg/jsk_recognition/blob/master/jsk_pcl_ros/sample/sample_pointcloud_screenpoint.launch#L26

を使うようにしたら，tfがでるようになるので，このsample_pointcloud_screenpoint.launchをincludeしたら再利用できるけど．．．．