jsk-ros-pkg/jsk_common #1683: virtual_force_publisherのforceとtorqueがnanになっている

Kanazawanaoaki, 2021-04-17:
https://github.com/jsk-ros-pkg/jsk_robot/blob/acb5000cd23249b38ad69bb5df3ae593fabbb1d5/jsk_pr2_robot/jsk_pr2_startup/pr2.launch#L126-L128
で呼び出されているPR2のvirtual_force_publisherのforceとtorqueがnanになってしまっています．

```
$ rostopic echo /left_endeffector/wrench -n 1
header: 
  seq: 28698129
  stamp: 
    secs: 1618653202
    nsecs: 273494756
  frame_id: "l_gripper_tool_frame"
wrench: 
  force: 
    x: nan
    y: nan
    z: nan
  torque: 
    x: nan
    y: nan
    z: nan
---
```
```
$ rostopic echo /right_endeffector/wrench -n 1
header: 
  seq: 28609328
  stamp: 
    secs: 1618653208
    nsecs:   3465913
  frame_id: "r_gripper_tool_frame"
wrench: 
  force: 
    x: nan
    y: nan
    z: nan
  torque: 
    x: nan
    y: nan
    z: nan
---
```
virtual_force_publisherのnodeがsubscribeしている/joint_statesには問題がなさそうでしょうか．
```
$ rostopic echo /joint_states -n 1
header: 
  seq: 115974509
  stamp: 
    secs: 1618653290
    nsecs: 583341842
  frame_id: ''
name: 
  - fl_caster_rotation_joint
  - fl_caster_l_wheel_joint
  - fl_caster_r_wheel_joint
  - fr_caster_rotation_joint
  - fr_caster_l_wheel_joint
  - fr_caster_r_wheel_joint
  - bl_caster_rotation_joint
  - bl_caster_l_wheel_joint
  - bl_caster_r_wheel_joint
  - br_caster_rotation_joint
  - br_caster_l_wheel_joint
  - br_caster_r_wheel_joint
  - torso_lift_joint
  - torso_lift_motor_screw_joint
  - head_pan_joint
  - head_tilt_joint
  - laser_tilt_mount_joint
  - r_upper_arm_roll_joint
  - r_shoulder_pan_joint
  - r_shoulder_lift_joint
  - r_forearm_roll_joint
  - r_elbow_flex_joint
  - r_wrist_flex_joint
  - r_wrist_roll_joint
  - r_gripper_joint
  - r_gripper_l_finger_joint
  - r_gripper_r_finger_joint
  - r_gripper_r_finger_tip_joint
  - r_gripper_l_finger_tip_joint
  - r_gripper_motor_screw_joint
  - r_gripper_motor_slider_joint
  - l_upper_arm_roll_joint
  - l_shoulder_pan_joint
  - l_shoulder_lift_joint
  - l_forearm_roll_joint
  - l_elbow_flex_joint
  - l_wrist_flex_joint
  - l_wrist_roll_joint
  - l_gripper_joint
  - l_gripper_l_finger_joint
  - l_gripper_r_finger_joint
  - l_gripper_r_finger_tip_joint
  - l_gripper_l_finger_tip_joint
  - l_gripper_motor_screw_joint
  - l_gripper_motor_slider_joint
position: [12.564444878477445, -1919.0729768556473, 353.5531097366099, -3.1421779262590066, -2383.7648448913924, 26.164180470609303, 15.708019907236459, -2293.399924653409, -286.28801442830144, 6.287244456338844, 995.8554187567771, 3167.9468022409856, 0.29954620653977165, 0.0, 0.03970101676069459, -0.4728096943652638, 0.04634639014285266, -1.6237257272959023, -0.07628154593029302, 0.8729522638944561, -1.983201809067611, -2.2929965044419536, -1.535903124405679, 3.6618886042715113, 0.04050919108138637, 0.2274747801083193, 0.2274747801083193, 0.2274747801083193, 0.2274747801083193, 0.0, 0.0, 1.811506757474214, 0.07701445222136138, 1.2223699226485238, -1.7184886157110104, -1.643410322464065, -0.09584312497911762, -0.08215193145566602, 0.08573691768020511, 0.4997805703799245, 0.4997805703799245, 0.4997805703799245, 0.4997805703799245, 0.0, 0.0]
velocity: [-0.0, 0.0, -0.0, -0.0, 0.0, -0.0, -0.0, 0.0, -0.0, -0.0, 0.0, -0.0, -0.0, 0.0, 0.0, 0.0, -0.0, 0.0, 0.0, 0.0, -0.0, -0.0, -0.0, -0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, -0.0, -0.0, -0.0, -0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
effort: [0.0808862504868329, -0.04741607876732521, 0.05299444158645222, 0.0418377159481982, -0.025102630088444357, 0.02231344867888085, 0.0, -0.005578362169720213, 0.0, -0.10877805938721358, 0.06415116722470625, -0.06415116722470625, 83.31074646473967, 0.0, -0.20013600875437262, -2.8893957492370603, -0.24265280415713789, -0.41741666594639887, 0.09139827178929666, 0.6307516231024825, 0.4772275438009743, 0.4209131977717453, 0.14891226057419601, -0.14567503756329483, -0.14421101425614277, -0.02528594821052964, -0.02528594821052964, -0.02528594821052964, -0.02528594821052964, 0.0, 0.0, 0.08958706773226567, -0.2246874317444465, 0.5094531934026001, 0.08521920036283204, -0.1853762772117527, 0.025897787102107973, -0.132726140808911, 31.769688940852316, 4.917575180311551, 4.917575180311551, 4.917575180311551, 4.917575180311551, 0.0, 0.0]
---
```

Kanazawanaoaki, 2021-04-17:
原因は究明出来ませんでしたが，帰宅する前にpr2.launchを立ち上げ直すと数値がpublishするようにはなりました．
```
$ rostopic echo /right_endeffector/wrench -n 1
header: 
  seq: 1227
  stamp: 
    secs: 1618656032
    nsecs: 713500039
  frame_id: "r_gripper_tool_frame"
wrench: 
  force: 
    x: -10.9706741526
    y: 13.7675618368
    z: 21.2767481656
  torque: 
    x: -1.21901630379
    y: -0.152456240429
    z: 2.48089204463
---
```

k-okada, 2021-04-17:
次に再現したら10秒ぐらいrosbag record -a しておいてください．
なんだろうなぁ．．．
あと，おかしいときに一回robot stop して ~/.ros/log/latest
かなぁ，，，にこのノードの出力のログがあるはずなので，それもみたいかも．入力が正しいとなるとロボットモデルがおかしくなっているのかな．．．
--
◉ Kei Okada