jsk-ros-pkg/jsk_common #1792: jsk_data does not run on noetic

sktometometo, 2023-10-16:
jsk_data seems not to run on noetic

```bash
~ $ jsk_data ls
Traceback (most recent call last):
  File "/opt/ros/noetic/bin/jsk_data", line 8, in <module>
    jsk_data.cli.cli(obj={})
  File "/home/sktometometo/.local/lib/python3.8/site-packages/click/core.py", line 1130, in __call__
    return self.main(*args, **kwargs)
  File "/home/sktometometo/.local/lib/python3.8/site-packages/click/core.py", line 1055, in main
    rv = self.invoke(ctx)
  File "/home/sktometometo/.local/lib/python3.8/site-packages/click/core.py", line 1657, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/home/sktometometo/.local/lib/python3.8/site-packages/click/core.py", line 1404, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/sktometometo/.local/lib/python3.8/site-packages/click/core.py", line 760, in invoke
    return __callback(*args, **kwargs)
  File "/opt/ros/noetic/lib/python3/dist-packages/jsk_data/cli.py", line 117, in cmd_ls
    print('\n'.join(_list_aries_files(query, ls_options)))
TypeError: sequence item 0: expected str instance, bytes found
```

k-okada, 2024-12-12:
could not duplicates this
```
~$ jsk_data ls
Traceback (most recent call last):
  File "/opt/ros/noetic/bin/jsk_data", line 8, in <module>
    jsk_data.cli.cli(obj={})
  File "/usr/lib/python3/dist-packages/click/core.py", line 764, in __call__
    return self.main(*args, **kwargs)
  File "/usr/lib/python3/dist-packages/click/core.py", line 717, in main
    rv = self.invoke(ctx)
  File "/usr/lib/python3/dist-packages/click/core.py", line 1137, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/lib/python3/dist-packages/click/core.py", line 956, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/usr/lib/python3/dist-packages/click/core.py", line 555, in invoke
    return callback(*args, **kwargs)
  File "/opt/ros/noetic/lib/python3/dist-packages/jsk_data/cli.py", line 117, in cmd_ls
    print('\n'.join(_list_aries_files(query, ls_options)))
  File "/opt/ros/noetic/lib/python3/dist-packages/jsk_data/cli.py", line 82, in _list_aries_files
    with connect_ssh(HOST, LOGIN_USER) as ssh:
  File "/usr/lib/python3.8/contextlib.py", line 113, in __enter__
    return next(self.gen)
  File "/opt/ros/noetic/lib/python3/dist-packages/jsk_data/ssh.py", line 19, in _connect_ssh_context
    ssh.connect(host, username=username, password=password)
  File "/usr/lib/python3/dist-packages/paramiko/client.py", line 415, in connect
    self._policy.missing_host_key(
  File "/usr/lib/python3/dist-packages/paramiko/client.py", line 823, in missing_host_key
    raise SSHException(
paramiko.ssh_exception.SSHException: Server 'aries.jsk.t.u-tokyo.ac.jp' not found in known_hosts
```

sktometometo, 2024-12-13:
I have got another error

```bash
~ $ jsk_data ls
Traceback (most recent call last):
  File "/opt/ros/noetic/bin/jsk_data", line 8, in <module>
    jsk_data.cli.cli(obj={})
  File "/home/sktometometo/.local/lib/python3.8/site-packages/click/core.py", line 1157, in __call__
    return self.main(*args, **kwargs)
  File "/home/sktometometo/.local/lib/python3.8/site-packages/click/core.py", line 1078, in main
    rv = self.invoke(ctx)
  File "/home/sktometometo/.local/lib/python3.8/site-packages/click/core.py", line 1688, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/home/sktometometo/.local/lib/python3.8/site-packages/click/core.py", line 1434, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/sktometometo/.local/lib/python3.8/site-packages/click/core.py", line 783, in invoke
    return __callback(*args, **kwargs)
  File "/opt/ros/noetic/lib/python3/dist-packages/jsk_data/cli.py", line 117, in cmd_ls
    print('\n'.join(_list_aries_files(query, ls_options)))
  File "/opt/ros/noetic/lib/python3/dist-packages/jsk_data/cli.py", line 82, in _list_aries_files
    with connect_ssh(HOST, LOGIN_USER) as ssh:
  File "/usr/lib/python3.8/contextlib.py", line 113, in __enter__
    return next(self.gen)
  File "/opt/ros/noetic/lib/python3/dist-packages/jsk_data/ssh.py", line 19, in _connect_ssh_context
    ssh.connect(host, username=username, password=password)
  File "/usr/lib/python3/dist-packages/paramiko/client.py", line 435, in connect
    self._auth(
  File "/usr/lib/python3/dist-packages/paramiko/client.py", line 764, in _auth
    raise saved_exception
  File "/usr/lib/python3/dist-packages/paramiko/client.py", line 734, in _auth
    key = self._key_from_filepath(
  File "/usr/lib/python3/dist-packages/paramiko/client.py", line 586, in _key_from_filepath
    key = klass.from_private_key_file(key_path, password)
  File "/usr/lib/python3/dist-packages/paramiko/pkey.py", line 206, in from_private_key_file
    key = cls(filename=filename, password=password)
  File "/usr/lib/python3/dist-packages/paramiko/rsakey.py", line 55, in __init__
    self._from_private_key_file(filename, password)
  File "/usr/lib/python3/dist-packages/paramiko/rsakey.py", line 175, in _from_private_key_file
    data = self._read_private_key_file("RSA", filename, password)
  File "/usr/lib/python3/dist-packages/paramiko/pkey.py", line 279, in _read_private_key_file
    data = self._read_private_key(tag, f, password)
  File "/usr/lib/python3/dist-packages/paramiko/pkey.py", line 289, in _read_private_key
    raise SSHException("not a valid " + tag + " private key file")
paramiko.ssh_exception.SSHException: not a valid RSA private key file
```

This is because my ssh private file is PCSK format.
https://github.com/paramiko/paramiko/issues/1015

This seems solved in latest paramiko. So I have updated it with `pip install paramiko -U`. But another error has occured.

```bash
~ $ jsk_data ls
Traceback (most recent call last):
  File "/opt/ros/noetic/bin/jsk_data", line 8, in <module>
    jsk_data.cli.cli(obj={})
  File "/home/sktometometo/.local/lib/python3.8/site-packages/click/core.py", line 1157, in __call__
    return self.main(*args, **kwargs)
  File "/home/sktometometo/.local/lib/python3.8/site-packages/click/core.py", line 1078, in main
    rv = self.invoke(ctx)
  File "/home/sktometometo/.local/lib/python3.8/site-packages/click/core.py", line 1688, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/home/sktometometo/.local/lib/python3.8/site-packages/click/core.py", line 1434, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/sktometometo/.local/lib/python3.8/site-packages/click/core.py", line 783, in invoke
    return __callback(*args, **kwargs)
  File "/opt/ros/noetic/lib/python3/dist-packages/jsk_data/cli.py", line 117, in cmd_ls
    print('\n'.join(_list_aries_files(query, ls_options)))
  File "/opt/ros/noetic/lib/python3/dist-packages/jsk_data/cli.py", line 82, in _list_aries_files
    with connect_ssh(HOST, LOGIN_USER) as ssh:
  File "/usr/lib/python3.8/contextlib.py", line 113, in __enter__
    return next(self.gen)
  File "/opt/ros/noetic/lib/python3/dist-packages/jsk_data/ssh.py", line 19, in _connect_ssh_context
    ssh.connect(host, username=username, password=password)
  File "/usr/lib/python3/dist-packages/paramiko/client.py", line 435, in connect
    self._auth(
  File "/usr/lib/python3/dist-packages/paramiko/client.py", line 764, in _auth
    raise saved_exception
  File "/usr/lib/python3/dist-packages/paramiko/client.py", line 734, in _auth
    key = self._key_from_filepath(
  File "/usr/lib/python3/dist-packages/paramiko/client.py", line 586, in _key_from_filepath
    key = klass.from_private_key_file(key_path, password)
  File "/usr/lib/python3/dist-packages/paramiko/pkey.py", line 206, in from_private_key_file
    key = cls(filename=filename, password=password)
  File "/usr/lib/python3/dist-packages/paramiko/rsakey.py", line 55, in __init__
    self._from_private_key_file(filename, password)
  File "/usr/lib/python3/dist-packages/paramiko/rsakey.py", line 175, in _from_private_key_file
    data = self._read_private_key_file("RSA", filename, password)
  File "/usr/lib/python3/dist-packages/paramiko/pkey.py", line 279, in _read_private_key_file
    data = self._read_private_key(tag, f, password)
  File "/usr/lib/python3/dist-packages/paramiko/pkey.py", line 289, in _read_private_key
    raise SSHException("not a valid " + tag + " private key file")
paramiko.ssh_exception.SSHException: not a valid RSA private key file
```

This is because my ssh private file is encrypited with passphrase.

PR #1810 (2024-12-13): [jsk_data] Automatically add the host key and Check stdout.read() type
- stdout.read() returns a bytes object in Python 3, whereas in Python 2, it returned a str
-
The error indicates that the new version of Paramiko (2.6.0) is stricter about host key checking, requiring the server to be present in the known_hosts file by default. This is a security feature to prevent man-in-the-middle attacks.

Closes #1792
closed 2024-12-13 by k-okada