jsk-ros-pkg/jsk_common #1811: Noetic armhf jsk_rosbag_tools not building, causing regression of jsk_common

sloretz, 2024-12-17:
https://build.ros.org/view/Nbin_ufhf_uFhf/job/Nbin_ufhf_uFhf__jsk_rosbag_tools__ubuntu_focal_armhf__binary/

This causes a regression of jsk_common in armhf https://repositories.ros.org/status_page/ros_noetic_ufhf.html?q=REGRESSION

Similar to https://github.com/jsk-ros-pkg/jsk_common/issues/1759 ?

```
       Getting requirements to build wheel: finished with status 'done'
23:06:41         Preparing wheel metadata: started
23:06:41         Preparing wheel metadata: finished with status 'error'
23:06:41         ERROR: Command errored out with exit status 1:
23:06:41          command: /tmp/binarydeb/ros-noetic-jsk-rosbag-tools-2.2.15/.obj-arm-linux-gnueabihf/venv/bin/python /tmp/binarydeb/ros-noetic-jsk-rosbag-tools-2.2.15/.obj-arm-linux-gnueabihf/venv/lib/python3.8/site-packages/pip/_vendor/pep517/_in_process.py prepare_metadata_for_build_wheel /tmp/tmp_0h48gjg
23:06:41              cwd: /tmp/pip-install-fxgigebb/numpy
23:06:41         Complete output (54 lines):
23:06:41     
23:06:41         Error compiling Cython file:
23:06:41         ------------------------------------------------------------
23:06:41         ...
23:06:41                     self.rng_state.ctr.v[i] = counter[i]
23:06:41     
23:06:41                 self._reset_state_variables()
23:06:41     
23:06:41                 self._bitgen.state = <void *>&self.rng_state
23:06:41                 self._bitgen.next_uint64 = &philox_uint64
23:06:41                                            ^
23:06:41         ------------------------------------------------------------
23:06:41     
23:06:41         _philox.pyx:195:35: Cannot assign type 'uint64_t (*)(void *) except? -1 nogil' to 'uint64_t (*)(void *) noexcept nogil'. Exception values are incompatible. Suggest adding 'noexcept' to the type of the value being assigned.
23:06:41         Processing numpy/random/_bounded_integers.pxd.in
23:06:41         Processing numpy/random/_philox.pyx
23:06:41         Traceback (most recent call last):
23:06:41           File "/tmp/pip-install-fxgigebb/numpy/tools/cythonize.py", line 235, in <module>
23:06:41             main()
23:06:41           File "/tmp/pip-install-fxgigebb/numpy/tools/cythonize.py", line 231, in main
23:06:41             find_process_files(root_dir)
23:06:41           File "/tmp/pip-install-fxgigebb/numpy/tools/cythonize.py", line 222, in find_process_files
23:06:41             process(root_dir, fromfile, tofile, function, hash_db)
23:06:41           File "/tmp/pip-install-fxgigebb/numpy/tools/cythonize.py", line 188, in process
23:06:41             processor_function(fromfile, tofile)
23:06:41           File "/tmp/pip-install-fxgigebb/numpy/tools/cythonize.py", line 77, in process_pyx
23:06:41             subprocess.check_call(
23:06:41           File "/usr/lib/python3.8/subprocess.py", line 364, in check_call
23:06:41             raise CalledProcessError(retcode, cmd)
23:06:41         subprocess.CalledProcessError: Command '['/tmp/binarydeb/ros-noetic-jsk-rosbag-tools-2.2.15/.obj-arm-linux-gnueabihf/venv/bin/python', '-m', 'cython', '-3', '--fast-fail', '-o', '_philox.c', '_philox.pyx']' returned non-zero exit status 1.
23:06:41         Running from numpy source directory.
23:06:41         setup.py:470: UserWarning: Unrecognized setuptools command, proceeding with generating Cython sources and expanding templates
23:06:41           run_build = parse_setuppy_commands()
23:06:41         Cythonizing sources
23:06:41         Traceback (most recent call last):
23:06:41           File "/tmp/binarydeb/ros-noetic-jsk-rosbag-tools-2.2.15/.obj-arm-linux-gnueabihf/venv/lib/python3.8/site-packages/pip/_vendor/pep517/_in_process.py", line 280, in <module>
23:06:41             main()
23:06:41           File "/tmp/binarydeb/ros-noetic-jsk-rosbag-tools-2.2.15/.obj-arm-linux-gnueabihf/venv/lib/python3.8/site-packages/pip/_vendor/pep517/_in_process.py", line 263, in main
23:06:41             json_out['return_val'] = hook(**hook_input['kwargs'])
23:06:41           File "/tmp/binarydeb/ros-noetic-jsk-rosbag-tools-2.2.15/.obj-arm-linux-gnueabihf/venv/lib/python3.8/site-packages/pip/_vendor/pep517/_in_process.py", line 133, in prepare_metadata_for_build_wheel
23:06:41             return hook(metadata_directory, config_settings)
23:06:41           File "/usr/lib/python3/dist-packages/setuptools/build_meta.py", line 158, in prepare_metadata_for_build_wheel
23:06:41             self.run_setup()
23:06:41           File "/usr/lib/python3/dist-packages/setuptools/build_meta.py", line 249, in run_setup
23:06:41             super(_BuildMetaLegacyBackend,
23:06:41           File "/usr/lib/python3/dist-packages/setuptools/build_meta.py", line 143, in run_setup
23:06:41             exec(compile(code, __file__, 'exec'), locals())
23:06:41           File "setup.py", line 499, in <module>
23:06:41             setup_package()
23:06:41           File "setup.py", line 479, in setup_package
23:06:41             generate_cython()
23:06:41           File "setup.py", line 274, in generate_cython
23:06:41             raise RuntimeError("Running cythonize failed!")
23:06:41         RuntimeError: Running cythonize failed!
23:06:41         ----------------------------------------
23:06:41     ERROR: Command errored out with exit status 1: /tmp/binarydeb/ros-noetic-jsk-rosbag-tools-2.2.15/.obj-arm-linux-gnueabihf/venv/bin/python /tmp/binarydeb/ros-noetic-jsk-rosbag-tools-2.2.15/.obj-arm-linux-gnueabihf/venv/lib/python3.8/site-packages/pip/_vendor/pep517/_in_process.py prepare_metadata_for_build_wheel /tmp/tmp_0h48gjg Check the logs for full command output.
```

sloretz, 2024-12-17:
Looks like https://github.com/ros/rosdistro/pull/43936 still isn't building https://build.ros.org/view/Nbin_ufhf_uFhf/job/Nbin_ufhf_uFhf__jsk_rosbag_tools__ubuntu_focal_armhf__binary/8/console

```
09:55:55     The headers or library files could not be found for jpeg,
09:55:55     a required dependency when compiling Pillow from source.
09:55:55     
09:55:55     Please see the install instructions at:
09:55:55        https://pillow.readthedocs.io/en/latest/installation.html
09:55:55     
09:55:55     Traceback (most recent call last):
09:55:55       File "/tmp/pip-install-gb1zmzab/pillow/setup.py", line 978, in <module>
09:55:55         setup(
09:55:55       File "/tmp/binarydeb/ros-noetic-jsk-rosbag-tools-2.2.15/.obj-arm-linux-gnueabihf/venv/lib/python3.8/site-packages/setuptools/__init__.py", line 145, in setup
09:55:55         return distutils.core.setup(**attrs)
09:55:55       File "/usr/lib/python3.8/distutils/core.py", line 148, in setup
09:55:55         dist.run_commands()
09:55:55       File "/usr/lib/python3.8/distutils/dist.py", line 966, in run_commands
09:55:55         self.run_command(cmd)
09:55:55       File "/usr/lib/python3.8/distutils/dist.py", line 985, in run_command
09:55:55         cmd_obj.run()
09:55:55       File "/tmp/binarydeb/ros-noetic-jsk-rosbag-tools-2.2.15/.obj-arm-linux-gnueabihf/venv/lib/python3.8/site-packages/setuptools/command/install.py", line 61, in run
09:55:55         return orig.install.run(self)
09:55:55       File "/usr/lib/python3.8/distutils/command/install.py", line 589, in run
09:55:55         self.run_command('build')
09:55:55       File "/usr/lib/python3.8/distutils/cmd.py", line 313, in run_command
09:55:55         self.distribution.run_command(command)
09:55:55       File "/usr/lib/python3.8/distutils/dist.py", line 985, in run_command
09:55:55         cmd_obj.run()
09:55:55       File "/usr/lib/python3.8/distutils/command/build.py", line 135, in run
09:55:55         self.run_command(cmd_name)
09:55:55       File "/usr/lib/python3.8/distutils/cmd.py", line 313, in run_command
09:55:55         self.distribution.run_command(command)
09:55:55       File "/usr/lib/python3.8/distutils/dist.py", line 985, in run_command
09:55:55         cmd_obj.run()
09:55:55       File "/tmp/binarydeb/ros-noetic-jsk-rosbag-tools-2.2.15/.obj-arm-linux-gnueabihf/venv/lib/python3.8/site-packages/setuptools/command/build_ext.py", line 84, in run
09:55:55         _build_ext.run(self)
09:55:55       File "/usr/lib/python3.8/distutils/command/build_ext.py", line 340, in run
09:55:55         self.build_extensions()
09:55:55       File "/tmp/pip-install-gb1zmzab/pillow/setup.py", line 790, in build_extensions
09:55:55         raise RequiredDependencyException(f)
09:55:55     __main__.RequiredDependencyException: jpeg
09:55:55     
09:55:55     During handling of the above exception, another exception occurred:
09:55:55     
09:55:55     Traceback (most recent call last):
09:55:55       File "<string>", line 1, in <module>
09:55:55       File "/tmp/pip-install-gb1zmzab/pillow/setup.py", line 1037, in <module>
09:55:55         raise RequiredDependencyException(msg)
09:55:55     __main__.RequiredDependencyException:
09:55:55     
09:55:55     The headers or library files could not be found for jpeg,
09:55:55     a required dependency when compiling Pillow from source.
09:55:55     
09:55:55     Please see the install instructions at:
09:55:55        https://pillow.readthedocs.io/en/latest/installation.html
```

k-okada, 2024-12-18:
https://github.com/ros/rosdistro/pull/43941 should be ok @sloretz